nickg/nvc #547: Adding NVC support to OSVVM

amb5l, 2022-10-04:
I am working on adding NVC to OSVVM in [a temporary fork of OsvvmLibraries](https://github.com/amb5l/OSVVM) - specifically in [the `nvc-support` branch of OSVVM_Scripts](https://github.com/amb5l/OSVVM-Scripts/tree/nvc-support). There is a build error in the AXI4 example:
```
nvc --std=08 --work=osvvm_axi4:V:/work/OsvvmLibraries_amb5l/VHDL_LIBS/nvc-1.8-devel/OSVVM_AXI4 -L V:/work/OsvvmLibraries_amb5l/VHDL_LIBS/nvc-1.8-devel -a AXI4/AxiStream/src/AxiStreamTransmitterVti.vhd
** Error: unexpected signal while parsing entity declarative item, expecting one of attribute, type, subtype, constant, function, procedure, impure, pure, alias, use, disconnect, group or shared
Error:      > V:\work\OsvvmLibraries_amb5l\AXI4\AxiStream\src\AxiStreamTransmitterVti.vhd:110
Error:      |
Error:  110 |   signal TransRec  : StreamRecType (
Error:      |   ^^^^^^ this token was unexpected
```

To reproduce the error, clone my fork, initialise submodules, check out the `nvc-support` branch of OSVVM_Scripts, then enter the following in a TCL shell:
```
cd ~/work/OsvvmLibraries
source Scripts/StartNVC.tcl
build OsvvmLibraries
```

nickg, 2022-10-04:
We're missing support for signals declared in the entity declarative part. This is actually a -93 feature, but not one I've ever seen used before.

nickg, 2022-10-06:
I've fixed this now but found one more error:

```
nvc --std=08 --work=osvvm_DpRam_PT:/home/nick/src/OsvvmLibraries/VHDL_LIBS/nvc-1.8-devel/OSVVM_DPRAM_PT -L /home/nick/src/OsvvmLibraries/VHDL_LIBS/nvc-1.8-devel -a DpRam/src/DpRam_PT.vhd
** Error: expected concurrent statement
Error:     > /home/nick/src/OsvvmLibraries/DpRam/src/DpRam_PT.vhd:74
Error:     |
Error:  74 |   Mem.MemInit(AddrWidth => ADDR_WIDTH,  DataWidth => DATA_WIDTH) ;
Error:     |   ^^^^^^^^^^^
```

We don't allow concurrent procedure calls on protected objects. That should be simple to add too.

nickg, 2022-10-08:
With the commit above:

```
Build Start time  09:32:00 BST Sat Oct 08 2022
Build Finish time 09:32:06, Elasped time: 0:00:06
Build: OsvvmLibraries PASSED,  Passed: 0,  Failed: 0,  Skipped: 0,  Analyze Errors: 0,  Simulate Errors: 0
```

closed 2022-10-08 by nickg

amb5l, 2022-10-09:
Tested and passed on Windows/MSYS2. Fantastic. Thankyou!