nickg/nvc #551: File opened for write not append?

amb5l, 2022-10-11:
I am seeing an issue running the DPRAM VC regression test in OSVVM. To reproduce, check out [my latest OSVVM_Scripts commit](https://github.com/amb5l/OSVVM-Scripts/commit/bb0a578285d4dc7da06e45318a5ecca0152496f5) and...
```
cd ~/work/OsvvmLibraries
source /work/OsvvmLibraries/Scripts/StartNVC.tcl
build /work/OsvvmLibraries/OsvvmLibraries
build /work/OsvvmLibraries/DpRam/RunAllTests
```
OSVVM builds a YML file to capture test results, partly in OsvvmProjectScripts.tcl, partly in ReportPkg.vhd, partly in AlertLogPkg.vhd. I added dumps of the file to a few places in the TCL. At some point it is wiped clean. So instead of something like this...
```
Version: 2022.09
Build:
  Name: DpRam_RunAllTests
  Date: 2022-10-11T12:14+0100
  Simulator: "QuestaSim -c"
  Version: QuestaSim-2022.01
TestSuites: 
  - Name: DpRam
    TestCases:
      - TestCaseName: TbDpRam_BasicReadWrite
        Name: "TbDpRam_BasicReadWrite"
        Status: PASSED
        Results: {TotalErrors: 0, AlertCount: {Failure: 0, Error: 0, Warning: 0}, PassedCount: 20, AffirmCount: 20, RequirementsPassed: 0, RequirementsGoal: 0, DisabledAlertCount: {Failure: 0, Error: 0, Warning: 0}}
        FunctionalCoverage:  
        TestCaseFileName: TbDpRam_BasicReadWrite
        TestCaseGenerics: ""
        ElapsedTime: 0.819
    ElapsedTime: 1.941
Run:
  Start:    2022-10-11T12:14+0100
  Finish:   2022-10-11T12:14+0100
  Elapsed:  2.072
```
...you get something like this:
```
        FunctionalCoverage:  
        TestCaseFileName: TbDpRam_BasicReadWrite
        TestCaseGenerics: ""
        ElapsedTime: 1.424
    ElapsedTime: 2.219
Run:
  Start:    2022-10-11T12:48+0100
  Finish:   2022-10-11T12:48+0100
  Elapsed:  2.387
```
This might happen in the procedure `WriteCovSummaryYaml` in `ReportPkg.vhd`. Maybe NVC opening the file in write mode rather than append mode?
Or perhaps in procedure `WriteOneAlertYaml` in `AlertLogPkg.vhd` - the line `WriteLine(TestFile, buf) ;`. Apologies for not being able to narrow it down further.

nickg, 2022-10-11:
Can you try replacing `"w+b"` with `"ab"` on line 47 of `src/jit/jit-exits.c`? I think it's a simple mistake but can't test at the moment.

amb5l, 2022-10-11:
Another one bites the dust. Thanks for the instant response. At this point, the complete OSVVM VC regression suite is one test away from perfection. The remaining issue looks related to #544 - I'll post there.

commit 67753a17 (2022-10-11): Append mode truncates file. Fixes #551
commit 5d3a061a (2022-10-13): Append mode truncates file. Fixes #551
closed 2022-10-11 by nickg