nickg/nvc #552: Null array assign to null array

avelure, 2022-10-12:
This passes without complaint on GHDL and mentor.
```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
end entity test;
architecture beh of test is
  signal sig : std_logic_vector(-1 downto 0);
  signal sig2 : std_logic_vector(0 to -1);
begin
process(all)
begin
  sig <= (sig'range => '0');
  sig2 <= (sig2'range => '0');
end process;
end architecture beh;
```
```
nvc --std=08 -a --relaxed test_null_range_assign.vhd -e test -r
** Error: aggregate choice index -1 outside of NATURAL range -1 downto 0
   > test_null_range_assign.vhd:8
   |
 8 |   signal sig : std_logic_vector(-1 downto 0);
   |                                 ^^
** Error: aggregate choice index 0 outside of NATURAL range -1 downto 0
   > test_null_range_assign.vhd:8
   |
 8 |   signal sig : std_logic_vector(-1 downto 0);
   |                                           ^
** Error: aggregate choice index 0 outside of NATURAL range 0 to -1
   > test_null_range_assign.vhd:9
   |
 9 |   signal sig2 : std_logic_vector(0 to -1);
   |                                  ^
** Error: aggregate choice index -1 outside of NATURAL range 0 to -1
   > test_null_range_assign.vhd:9
   |
 9 |   signal sig2 : std_logic_vector(0 to -1);
   |                                       ^^
```
In one of the more convoluted AXI TBs based on #544, this generated a fatal at runtime, but I'm not able to recreate that, but I assume the underlying issue is the same.
```
** Fatal: 0ms+1: index -1 outside of NATURAL range -1 to -2
     > C:\proj\inventas\uvvm\bitvis_vip_axi\tb\maintenance_tb\axi_slave_model.vhd:133
     |
 133 |       wr_port_out.bid     <= (wr_port_out.bid'range => '0');
     |                               ^^^^^^^^^^^^^^^^^^^^^
```

nickg, 2022-10-12:
Yes this is a bug: we shouldn't do the index check when the range is null.

nickg, 2022-10-26:
The first incorrect error should be fixed now but I couldn't reproduce the second issue. Could you test again?

avelure, 2022-10-30:
It solves the first issue, but not the second. However, I managed to recreate it, so I'll add another issue.

nickg, 2022-11-05:
Thanks for the additional test case. I think this should be fixed now along wit #560.

closed 2022-11-05 by nickg