nickg/nvc #554: Default as identifier in 08

avelure, 2022-10-17:
I noticed compiling some Unisim libraries (which were written for 93) that they sometimes use "default" as an identifier for some constants.
This is a reserved word in 2008, but it seems like NVC does not complain.
Both Mentor and GHDL gives a "expecting identifier" error when compiled for 08 instead of 93.
I wouldn't mind if it is still accepted with the relaxed option, as I guess the identifier is unambiguous in this case, but there probably should be an error without if you want to be strict to the LRM.

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
end entity test;
architecture beh of test is
  constant DEFAULT : std_logic_vector(3 downto 0) := "1001";
begin
end architecture beh;
```

nickg, 2022-10-17:
Yes it's actually an identifier from PSL (as in `default clock ...`) which NVC doesn't support yet. I've done some work in another branch but it isn't merged yet. Normally for other reserved names it emits a warning like "foo is a reserved word in VHDL-2008".

nickg, 2023-03-05:
Now gives an error:

```
** Error: unexpected default while parsing constant declaration, expecting identifier
   > test.vhd:8
   |
 8 |   constant DEFAULT : std_logic_vector(3 downto 0) := "1001";
   |            ^^^^^^^ this token was unexpected
```

closed 2023-03-05 by nickg

Blebowski, 2023-03-19:
Actually, should it really throw an error ? Looking at 2008 LRM, it seems it should be valid:

```
Within a PSL declaration, a PSL directive, or a PSL verification unit, PSL keywords are reserved words (see
IEEE Std 1850-2005). A PSL keyword shall not be used as an identifier to declare a PSL declaration or a
PSL verification unit. A PSL keyword that is a legal VHDL identifier may be used as an explicitly declared
identifier other than to declare a PSL declaration or a PSL verification unit, but such a declaration is hidden
within a PSL declaration, a PSL directive, or a PSL verification unit (see 12.3).
```

It seems to me that this clause in the LRM is there particularly for backwards compatibility with the code
like the mentioned Unisim library.

@nickg, feel free to close the issue if I got the LRM wrong.

nickg, 2023-03-19:
I'm not sure, the LRM seems a bit ambiguous here. That paragraph is preceded by "A reserved word shall not be used as an explicitly declared identifier."

At any rate `default` is also a plain VHDL keyword in 2008. I just didn't implement that usage yet as I never saw any real-world code using it:

```
interface_package_generic_map_aspect ::=
    generic_map_aspect
  | generic map ( <> )
  | generic map ( default )
```

Blebowski, 2023-03-19:
OK, I see. I didn't know about the other usage of `default`. Throwing error makes sense then.