nickg/nvc #555: MSYS2 paths

amb5l, 2022-10-18:
My MSYS2 MINGW64 build of NVC does not accept MSYS2 paths:
```
$ nvc --std=08 --work=default:/c/work/MINGW-packages/mingw-w64-nvc/VHDL_LIBS/nvc-1.8-devel/DEFAULT --init
** Fatal: invalid character ';' in library name
```
Not sure where the semicolon is creeping in? A mixed path works fine:
```
$ nvc --std=08 --work=default:C:/work/MINGW-packages/mingw-w64-nvc/VHDL_LIBS/nvc-1.8-devel/DEFAULT --init
```

nickg, 2022-10-18:
Could it be that the MSYS shell does this automatically? See the 
"Automatic Unix ⟶ Windows Path Conversion" section in the link below, it 
seems to show automatic ':' to ';' conversion:

https://www.msys2.org/docs/filesystem-paths/

amb5l, 2022-10-18:
Yep. The root cause was me using a MSYS2 nativce TCL shell rather than a MINGW64 one. Apologies.

closed 2022-10-18 by amb5l