nickg/nvc #558: Verilog cosimulation approach

Blebowski, 2022-10-20:
Hello @nickg,

I was wondering about the approach you would like to take when approaching Mixed-language simulation with Verilog/System Verilog. I remember you mentioned that you would like to add some Verilog / System Verilog support.

I am asking since these days Verilator is reaching release version 5 in which it introduces real timing simulation (not just synthesizable code, but IEEE compliant scheduler). So since I am following this, I am basically wondering which of
the options you think is better to choose to achieve mixed language simulation:
1. Integrate Verilator and its whole FE+BE (including code generation engine). It essentially means to connect generated Verilator code with NVC generated code.
2. Integrate only Verilog/System Verilog parser/front-end, and do the code generation engine for Verilog/System Verilog part of the design as part of NVC, which basically means modifying the simulation kernel to have the stratified scheduler as SV standard requires.

What do you think??

Blebowski, 2022-10-20:
Also, a thing which I am curious about is the potential integration level of such mixed language simulation. AFAIK, all Big 3 simulators support only module/entity instantiation in the other language and mapping of std_logic <-> logic and integer types. I have not seen mixed language mapping of enums / records (which I think would be really handy), mapping of functions in packages (e.g. you could call VHDL a function from System Verilog package), nor instantiation of UDP instances in VHDL designs.

Also, interesting point is external names / Verilog "dot" notation to access internal parts of the design written in other language. Many times in my job we run into issues when we want to drive internals of some VHDL model from a vendor (and we can't really rewrite it into System Verilog since we don't know the exact behavior of analog block which it models) by a SV/UVM testbench via virtual interface.

nickg, 2022-10-21:
Actually my plan for an "MVP" is actually much simpler:

1. Use another fronted to do preprocessing/parsing/checking.
2. Add a new object type to internally represent the Verilog AST (like `tree_t`) and serialise it to disk etc. and then use this during elaboration. I've already partially implemented this in another branch.
3. Translate Verilog statements to NVC's internal "vcode" IR which is based on VHDL and leave the simulation model largely unchanged. 

For 3. it obviously won't implement the Verilog semantics exactly, so it will be more like Verilator where it only really works for synthesisable code. So Verilog `wire`/`reg` will be translated to `std_logic` signals, blocking assignment becomes a VHDL signal assignment followed by a 0 ns delay (so it causes a delta cycle), etc.. Years ago I did a GSoC project to add a `-tvhdl` backend to Iverilog that used this translation scheme and it could pass a surprisingly large fraction of the Iverilog test suite.

If this is successful the long-term plan would be to move more towards your 2. above.

Blebowski, 2022-10-22:
Thanks, that sounds reasonable. I like the idea of having single Vcode backend, I will focus on the coverage and see where it gets :)

Blebowski, 2023-03-15:
Hi, I am closing this and moving the topic to the discussion.

closed 2023-03-15 by Blebowski