nickg/nvc #564: Coverage: Support merging of different hierarchies

Blebowski, 2022-11-09:
Currently if during merging a coverage tag is encountered which does not match the hierarchy of
a tag in first loaded covdb, such tag will be dropped. A better behavior would be searching in the
hierarchy, and place such tag in merged file at hierarchy which has most common hierarchy with
the new tag. This will allow doing coverage reports on "generic" IPs where top level generics
change the hierarchy of the IP (e.g. `if generate`)

Blebowski, 2022-11-24:
Hello @nickg , I am about to start with implementation of this issue. The idea is to walk across
`new` array of tags, and upon encountering TAG_HIER(DOWN), to buffer this tag. Then, walk
across `old` array of tags, and if I find matching TAG_HIER, I know this hierarchy exists in the
`old` array, and thus start processing tags within the hierarchy and merge them. I will do it
recursively, so since there are hierarchies within hierarchies.

If I don't find the hierarchy from the `new` in the `old`, I will copy all the tags within this 
new hierarchy to `old` (including hierarchy tag). I will search for a place which has biggest
common hierarchical path prefix with any existing TAG_HIER. This will be done on the
fly as I will be searching for hierarchy tags within `old`.

The issue I have, is that If I don't find the matching TAG, I need to insert the new TAG
into the hierarchy in the middle of existing array, which is sub-optimal. I am thinking
about simple linked-list implementation instead of array, to hold the tags within
cover_tagging_t. What do you think?

nickg, 2022-11-24:
> 
> The issue I have, is that If I don't find the matching TAG, I need to insert the new TAG into the hierarchy in the middle of existing array, which is sub-optimal. I am thinking about simple linked-list implementation instead of array, to hold the tags within cover_tagging_t. What do you think?

I suggesting trying it and see what it looks like. But bearing in mind that a linked list will be massively slower than an array if you have to iterate over it.

closed 2023-11-01 by nickg