nickg/nvc #567: Coverage: Performance enhancements

Blebowski, 2022-11-09:
1. [ ] Add option to exclude some parts of hierarchy from coverage collection (e.g. to collect only on DUT, not DUT + TB).
2. [ ] Encode tag hierarchies differentially to reduce size of covdb file
3. [ ] Optimize merging of tags (several options possible, rethink)

nickg, 2022-11-15:
There's a standard UCIS for coverage database interoperability. Is that widely used and/or useful? It might be worth making sure the database contents is similar to what that provides in case we want to implement a subset of its API in the future.

nickg, 2022-11-15:
https://www.accellera.org/downloads/standards/ucis

Blebowski, 2022-11-15:
Thats a good point, definitely worth checking out. I have not heard about it before though.

nickg, 2022-11-15:
I think it's basically a standardisation of the UCDB format used by Modelsim/Questa. But I don't really know much about it myself either TBH.

There are some open source tools that support it e.g. https://github.com/edaa-org/pyEDAA.UCIS

Blebowski, 2022-12-16:
Benchmarking on basic test from NEORV32 shows following data:

```

-------------------------------------------------------------------------------
-------------------------------------------------------------------------------
NEORV32:
-------------------------------------------------------------------------------
-------------------------------------------------------------------------------

No Coverage:

** Note: initialising [15ms +23464kB]
** Note: loading top-level unit [3ms +1756kB]
** Note: elaborating design [147ms +20872kB]
** Note: generating intermediate code [169ms +24812kB]
** Note: saving library [11ms +0kB]
** Note: code generation for 1665 units [15312ms +61272kB]
** Note: linking shared library [8ms +0kB]

** Note: setup:130ms run:20559ms maxrss:58076kB static:1901kB
37.27user 0.80system 0:25.31elapsed 150%CPU (0avgtext+0avgdata 134112maxresident)k
0inputs+0outputs (139major+178346minor)pagefaults 0swaps

-------------------------------------------------------------------------------
Full Coverage:
-------------------------------------------------------------------------------

-- Compile + Run example test

** Note: initialising [16ms +23396kB]
** Note: loading top-level unit [6ms +1820kB]
** Note: elaborating design [146ms +20900kB]
** Note: generating intermediate code [5017ms +78816kB]
** Note: dumping coverage data [115ms +264kB]
** Note: saving library [13ms +0kB]
** Note: code generation for 1665 units [22837ms +76204kB]
** Note: linking shared library [9ms +0kB]

** Note: setup:138ms run:28961ms maxrss:135792kB static:1905kB
58.08user 0.96system 0:40.31elapsed 146%CPU (0avgtext+0avgdata 202772maxresident)k
0inputs+0outputs (147major+218745minor)pagefaults 0swaps

-- Generate report

** Note: code coverage results for: NEORV32.NEORV32_TB_SIMPLE
** Note:      statement:     65.8 % (4103/6236)
** Note:      branch:        47.5 % (1962/4134)
** Note:      toggle:        3.5 % (18673/534172)
** Note:      expression:    43.7 % (1762/4031)
107.93user 1.24system 1:54.96elapsed 94%CPU (0avgtext+0avgdata 233764maxresident)k
0inputs+0outputs (0major+63109minor)pagefaults 0swaps

-------------------------------------------------------------------------------
Full coverage, but exclude Toggle coverage on mems
-------------------------------------------------------------------------------

-- Compile + Run example test

** Note: initialising [9ms +23200kB]
** Note: loading top-level unit [3ms +1816kB]
** Note: elaborating design [152ms +20852kB]
** Note: generating intermediate code [264ms +34028kB]
** Note: dumping coverage data [15ms +264kB]
** Note: saving library [15ms +0kB]
** Note: code generation for 1665 units [22002ms +76080kB]
** Note: linking shared library [17ms +0kB]

** Note: setup:132ms run:25513ms maxrss:64024kB static:1904kB
49.10user 0.79system 0:32.30elapsed 154%CPU (0avgtext+0avgdata 157716maxresident)k
0inputs+0outputs (149major+188008minor)pagefaults 0swaps

-- Generate report

** Note: code coverage results for: NEORV32.NEORV32_TB_SIMPLE
** Note:      statement:     65.8 % (4103/6236)
** Note:      branch:        47.5 % (1962/4134)
** Note:      toggle:        35.5 % (14261/40160)
** Note:      expression:    43.7 % (1762/4031)
0.57user 0.15system 0:01.04elapsed 69%CPU (0avgtext+0avgdata 79832maxresident)k
0inputs+0outputs (1major+16561minor)pagefaults 0swaps

-------------------------------------------------------------------------------
Only statement + branch coverage
-------------------------------------------------------------------------------

-- Compile + Run example test

** Note: initialising [13ms +23268kB]
** Note: loading top-level unit [4ms +1756kB]
** Note: elaborating design [163ms +20884kB]
** Note: generating intermediate code [211ms +27932kB]
** Note: dumping coverage data [7ms +264kB]
** Note: saving library [12ms +0kB]
** Note: code generation for 1665 units [20379ms +71584kB]
** Note: linking shared library [8ms +516kB]

** Note: setup:134ms run:21735ms maxrss:59732kB static:1901kB
43.53user 0.88system 0:27.68elapsed 160%CPU (0avgtext+0avgdata 147564maxresident)k
0inputs+0outputs (141major+183027minor)pagefaults 0swaps

-- Generate report

** Note: code coverage results for: NEORV32.NEORV32_TB_SIMPLE
** Note:      statement:     65.8 % (4103/6236)
** Note:      branch:        47.5 % (1962/4134)
** Note:      toggle:        N.A.
** Note:      expression:    N.A.
0.05user 0.08system 0:00.25elapsed 53%CPU (0avgtext+0avgdata 66756maxresident)k
0inputs+0outputs (1major+12524minor)pagefaults 0swaps
```

nickg, 2022-12-18:
Do we really need toggle coverage on memories at all? I've never used the commercial tools to do serious verification so not sure if it's an expected feature, but it doesn't seem very useful to me.

Blebowski, 2022-12-18:
Hi,
no, not really, that is why I added the `ignore_mems`, since I have seen performance hits in commercial tools. Nice to see that it has similar effect also in this implementation, proves validity of `ignore_mems`

nickg, 2022-12-28:
Perhaps `ignore-mems` should be on by default then?

Blebowski, 2023-01-01:
Yup, then I can rename it to something like `include-mems` ?

nickg, 2023-01-01:
> Yup, then I can rename it to something like `include-mems` ?

Yes, that sounds sensible.

Blebowski, 2023-01-28:
Current profiling with `gprof` shows that the bottleneck when generating report for NEORV32 with full coverage (including memories) is when determining name of the signal to be printed in the report. Repetitive iteration on `ident_from` for `ident_level` times vastes most of the time:

```
Flat profile:

Each sample counts as 0.01 seconds.
  %   cumulative   self              self     total           
 time   seconds   seconds    calls   s/call   s/call  name    
 99.59    102.74   102.74  1573769     0.00     0.00  ident_from_bytes
  0.12    102.86     0.12  1292067     0.00     0.00  ident_from
  0.07    102.93     0.07      868     0.00     0.10  cover_print_chain
  0.05    102.98     0.05   279483     0.00     0.00  ident_read
  0.03    103.01     0.03  1480203     0.00     0.00  ident_install
  0.03    103.04     0.03   279318     0.00     0.00  ident_new
  0.02    103.06     0.02     1503     0.00     0.00  ident_prefix
  0.02    103.08     0.02      627     0.00     0.00  fastlz_decompress
  0.02    103.10     0.02        1     0.02    18.36  cover_read_tags
  0.01    103.11     0.01 30344504     0.00     0.00  read_u8
  0.01    103.12     0.01   557678     0.00     0.00  cover_print_bin
  0.01    103.13     0.01   491703     0.00     0.00  loc_file_str
  0.01    103.14     0.01   279069     0.00     0.00  cover_append_to_chain
  0.01    103.15     0.01   276366     0.00     0.00  cover_print_bins
  0.01    103.16     0.01        1     0.01     0.01  read_u16
  0.00    103.16     0.00  1482056     0.00     0.00  xmalloc
  0.00    103.16     0.00  1480244     0.00     0.00  xmalloc_flex
  0.00    103.16     0.00  1397422     0.00     0.00  read_u32
  0.00    103.16     0.00   798019     0.00     0.00  istr
  0.00    103.16     0.00   545427     0.00     0.00  cover_tag_to_chain
  0.00    103.16     0.00   524895     0.00     0.00  cover_print_get_exclude_button
  0.00    103.16     0.00   279536     0.00     0.00  fbuf_get_uint
  0.00    103.16     0.00   279484     0.00     0.00  cover_read_one_tag
  0.00    103.16     0.00   279483     0.00     0.00  loc_read
  0.00    103.16     0.00   279483     0.00     0.00  read_u64
  0.00    103.16     0.00   279277     0.00     0.00  cover_file
  0.00    103.16     0.00   279277     0.00     0.00  loc_invalid_p
  0.00    103.16     0.00   276366     0.00     0.00  cover_print_bin_header
  0.00    103.16     0.00    18077     0.00     0.00  xstrdup
  0.00    103.16     0.00    13752     0.00     0.00  cover_read_header
  0.00    103.16     0.00    11502     0.00     0.00  tb_vprintf
  0.00    103.16     0.00    11495     0.00     0.00  tb_printf
  0.00    103.16     0.00     2501     0.00     0.00  xcalloc
  0.00    103.16     0.00     2486     0.00     0.00  xcalloc_array
  0.00    103.16     0.00     2486     0.00     0.00  xcalloc_flex
  0.00    103.16     0.00     2070     0.00     0.00  cover_print_percents_cell
  0.00    103.16     0.00     1503     0.00     0.00  ident_walk_selected
  0.00    103.16     0.00     1498     0.00     0.00  checked_sprintf
  0.00    103.16     0.00      859     0.00     0.00  tb_get
  0.00    103.16     0.00      859     0.00     0.00  tb_new
  0.00    103.16     0.00      852     0.00     0.00  _tb_cleanup
  0.00    103.16     0.00      414     0.00     0.00  cover_print_hierarchy_summary
  0.00    103.16     0.00      217     0.00     0.00  xvasprintf
  0.00    103.16     0.00      213     0.00     0.00  _local_free
  0.00    103.16     0.00      210     0.00     0.00  xasprintf
  0.00    103.16     0.00      208     0.00     0.02  cover_print_html_header.constprop.0
  0.00    103.16     0.00      208     0.00     0.00  cover_print_timestamp
  0.00    103.16     0.00      206     0.00     0.00  ident_rfrom
  0.00    103.16     0.00       74     0.00     0.00  xrealloc
```

nickg, 2023-01-28:
It's probably a regression caused by ef19e3996. I meant to go back and improve it but never got round to it. Can you test again with the commit above?

Blebowski, 2023-01-29:
That helps a lot, from original 2 minutes to generate the report, it is just a couple of seconds:

```
** Note: Code coverage report folder: rpt_dir.
** Note: Code coverage report contains: covered, uncovered, excluded coverage details.
** Note: code coverage results for: NEORV32.NEORV32_TB_SIMPLE
** Note:      statement:     63.2 % (4218/6674)
** Note:      branch:        45.8 % (1975/4316)
** Note:      toggle:        3.5 % (18673/534172)
** Note:      expression:    33.7 % (2340/6939)
2.31user 0.72system 0:05.91elapsed 51%CPU (0avgtext+0avgdata 315912maxresident)k
0inputs+0outputs (0major+100717minor)pagefaults 0swaps
```

Still, majority of the time is spent in the lookup of actual signal name from the full tag hierarchy:

```
Flat profile:

Each sample counts as 0.01 seconds.
  %   cumulative   self              self     total           
 time   seconds   seconds    calls  ms/call  ms/call  name    
 29.17      0.28     0.28       12    23.33    23.33  grow_table
 27.08      0.54     0.26  1573769     0.00     0.00  ident_from_bytes
 19.79      0.73     0.19  1292067     0.00     0.00  ident_from
  4.17      0.77     0.04   279318     0.00     0.00  ident_new
  3.13      0.80     0.03   524895     0.00     0.00  cover_print_get_exclude_button
  3.13      0.83     0.03   279483     0.00     0.00  ident_read
  3.13      0.86     0.03      627     0.05     0.05  fastlz_decompress
  2.08      0.88     0.02 30344504     0.00     0.00  read_u8
  2.08      0.90     0.02      868     0.02     0.81  cover_print_chain
  1.04      0.91     0.01   557678     0.00     0.00  cover_print_bin
  1.04      0.92     0.01   279536     0.00     0.00  fbuf_get_uint
  1.04      0.93     0.01   279069     0.00     0.00  cover_append_to_chain
  1.04      0.94     0.01   276366     0.00     0.00  cover_print_bins
  1.04      0.95     0.01        1    10.00   215.82  cover_read_tags
  1.04      0.96     0.01        1    10.00    10.00  read_u16
  0.00      0.96     0.00  1574870     0.00     0.00  get_table
  0.00      0.96     0.00  1482056     0.00     0.00  xmalloc
  0.00      0.96     0.00  1480244     0.00     0.00  xmalloc_flex
  0.00      0.96     0.00  1397422     0.00     0.00  read_u32
  0.00      0.96     0.00   798019     0.00     0.00  istr
  0.00      0.96     0.00   545427     0.00     0.00  cover_tag_to_chain
  0.00      0.96     0.00   491703     0.00     0.00  loc_file_str
  0.00      0.96     0.00   279484     0.00     0.00  cover_read_one_tag
  0.00      0.96     0.00   279483     0.00     0.00  loc_read
  0.00      0.96     0.00   279483     0.00     0.00  read_u64
  0.00      0.96     0.00   279277     0.00     0.00  cover_file
  0.00      0.96     0.00   279277     0.00     0.00  loc_invalid_p
  0.00      0.96     0.00   276366     0.00     0.00  cover_print_bin_header
  0.00      0.96     0.00    18077     0.00     0.00  xstrdup
  0.00      0.96     0.00    13752     0.00     0.00  cover_read_header
  0.00      0.96     0.00    11502     0.00     0.00  tb_vprintf
  0.00      0.96     0.00    11495     0.00     0.00  tb_printf
  0.00      0.96     0.00     2514     0.00     0.00  xcalloc
  0.00      0.96     0.00     2499     0.00     0.00  xcalloc_flex
  0.00      0.96     0.00     2486     0.00     0.00  xcalloc_array
  0.00      0.96     0.00     2070     0.00     0.00  cover_print_percents_cell
  0.00      0.96     0.00     1503     0.00     0.00  ident_prefix
  0.00      0.96     0.00     1503     0.00     0.00  ident_walk_selected
  0.00      0.96     0.00     1498     0.00     0.00  checked_sprintf
  0.00      0.96     0.00     1089     0.00     0.00  ident_from_byte_vec
  0.00      0.96     0.00      859     0.00     0.00  tb_get
  0.00      0.96     0.00      859     0.00     0.00  tb_new
  0.00      0.96     0.00      852     0.00     0.00  _tb_cleanup
  0.00      0.96     0.00      414     0.00     0.00  cover_print_hierarchy_summary
  0.00      0.96     0.00      217     0.00     0.00  xvasprintf
  0.00      0.96     0.00      213     0.00     0.00  _local_free
  0.00      0.96     0.00      210     0.00     0.00  xasprintf
  0.00      0.96     0.00      208     0.00     0.15  cover_print_html_header.constprop.0
  0.00      0.96     0.00      208     0.00     0.00  cover_print_timestamp
```

I am thinking to keep a separate scope for the signal name of a toggle tag during "lower"
phase, add a "sig_name" attribute to a toggle tag, and store that one in the covdb. Then,
iterating accross identifier would not need to be done during report generation. Only
printing directly the "sig_name". Also, the per-tag attribute `levels` would dissapear, since
that one is there just to determine the depth of iteration...

Blebowski, 2023-01-29:
Profiling for #602  :

```
///////////////////////////////////////////////////////////////////////////////
///////////////////////////////////////////////////////////////////////////////
// Without the optimization - Iterate the Identifier to find name
///////////////////////////////////////////////////////////////////////////////
///////////////////////////////////////////////////////////////////////////////

///////////////////////////////////////////////////////////////////////////////
// Generation of Intermediate code (5 runs)
///////////////////////////////////////////////////////////////////////////////

** Note: generating intermediate code [865ms +99552kB]
** Note: generating intermediate code [871ms +99532kB]
** Note: generating intermediate code [858ms +99268kB]
** Note: generating intermediate code [875ms +100344kB]
** Note: generating intermediate code [892ms +100068kB]

AVG = 872 ms

///////////////////////////////////////////////////////////////////////////////
// Generation of Coverage report (5 runs)
///////////////////////////////////////////////////////////////////////////////

2.40user 0.76system 0:06.38elapsed 49%CPU (0avgtext+0avgdata 315928maxresident)k
0inputs+0outputs (1major+100713minor)pagefaults 0swaps

2.43user 0.88system 0:06.55elapsed 50%CPU (0avgtext+0avgdata 316144maxresident)k
0inputs+0outputs (0major+100719minor)pagefaults 0swaps

2.33user 0.71system 0:06.07elapsed 50%CPU (0avgtext+0avgdata 315928maxresident)k
0inputs+0outputs (0major+100713minor)pagefaults 0swaps

2.30user 0.78system 0:06.16elapsed 49%CPU (0avgtext+0avgdata 316128maxresident)k
0inputs+0outputs (0major+100718minor)pagefaults 0swaps

2.33user 0.80system 0:06.27elapsed 50%CPU (0avgtext+0avgdata 316084maxresident)k
0inputs+0outputs (0major+100720minor)pagefaults 0swaps

AVG:
user = 2.358 s
elapsed = 6.376 s

///////////////////////////////////////////////////////////////////////////////
// GPROF result (single run):
///////////////////////////////////////////////////////////////////////////////

Each sample counts as 0.01 seconds.
  %   cumulative   self              self     total           
 time   seconds   seconds    calls  ms/call  ms/call  name    
 36.91      0.31     0.31       12    25.83    25.83  grow_table
 25.00      0.52     0.21  1573769     0.00     0.00  ident_from_bytes
 15.48      0.65     0.13  1292067     0.00     0.00  ident_from
  4.76      0.69     0.04   279483     0.00     0.00  ident_read
  3.57      0.72     0.03 30344504     0.00     0.00  read_u8
  3.57      0.75     0.03   279318     0.00     0.00  ident_new
  2.38      0.77     0.02      627     0.03     0.03  fastlz_decompress
  2.38      0.79     0.02        1    20.00    20.00  read_u16
  1.19      0.80     0.01  1574870     0.00     0.00  get_table
  1.19      0.81     0.01   557678     0.00     0.00  cover_print_bin
  1.19      0.82     0.01   524895     0.00     0.00  cover_print_get_exclude_button
  1.19      0.83     0.01    13752     0.00     0.00  cover_read_header
  1.19      0.84     0.01      868     0.01     0.70  cover_print_chain
  0.00      0.84     0.00  1482056     0.00     0.00  xmalloc

///////////////////////////////////////////////////////////////////////////////
// COVDB size:
///////////////////////////////////////////////////////////////////////////////
  
[... 110 lines cut ...]

Each sample counts as 0.01 seconds.
  %   cumulative   self              self     total           
 time   seconds   seconds    calls  ms/call  ms/call  name    
 19.36      0.06     0.06   279483     0.00     0.00  ident_read
 19.36      0.12     0.06       10     6.00     6.00  grow_table
 12.90      0.16     0.04 30344504     0.00     0.00  read_u8
  9.68      0.19     0.03   279318     0.00     0.00  ident_new
  6.45      0.21     0.02  1397422     0.00     0.00  read_u32
  6.45      0.23     0.02   524895     0.00     0.00  cover_print_get_exclude_button
  6.45      0.25     0.02   281702     0.00     0.00  ident_from_bytes
  6.45      0.27     0.02      627     0.03     0.03  fastlz_decompress
  3.23      0.28     0.01   798019     0.00     0.00  istr
  3.23      0.29     0.01   557678     0.00     0.00  cover_print_bin
  3.23      0.30     0.01   279277     0.00     0.00  cover_file
  3.23      0.31     0.01      868     0.01     0.06  cover_print_chain
  0.00      0.31     0.00   545427     0.00     0.00  cover_tag_to_chain

///////////////////////////////////////////////////////////////////////////////
// COVDB size:
///////////////////////////////////////////////////////////////////////////////

2,3M	.nvc/neorv32/_NEORV32.NEORV32_TB_SIMPLE.elab.covdb

```

Blebowski, 2023-03-06:
All points implemented, closing

closed 2023-03-06 by Blebowski