nickg/nvc #570: Elaboration seg fault

amb5l, 2022-11-11:
I am seeing a seg fault on Windows/MSYS2 during elaboration of [this design](https://github.com/amb5l/tyto2/tree/main/build/np6532_functest):
```
C:\work\tyto2\build\np6532_functest>make nvc
cd .nvc && nvc -M 32m --std=2008 --work=work -e np6532_functest -V -gvector_init=64512 -gstart_address=1024 -gref_file=C:/work/tyto2/build/np6532_functest/reftrace/6502_functional_test.txt -gprogress_interval=1000 && nvc -M 32m --std=2008 --work=work -r np6532_functest --ieee-warnings=off
** Note: initialising [15ms +12708kB]
** Note: loading top-level unit [0ms +3228kB]
** Note: elaborating design [437ms +131488kB]
** Note: generating intermediate code [484ms +21124kB]
** Note: saving library [437ms +176kB]
** Note: code generation for 2506 units [51344ms +140672kB]
** Note: linking shared library [452ms +0kB]
make: *** [C:/work/tyto2/build/build.mak:137: sim] Segmentation fault

C:\work\tyto2\build\np6532_functest>nvc --version
nvc 1.8-devel (1.7.0.r236.gfca87fda) (Using LLVM 15.0.3)
```

nickg, 2022-11-11:
The segfault is actually caused by a stack overflow in the VHDL code but we don't handle that well at the moment. The underlying problem should be fixed by the commit above.

closed 2022-11-11 by nickg

amb5l, 2022-11-12:
The seg fault is resolved - thankyou. Unfortunately the simulation is not working correctly. I will investigate and report back.

amb5l, 2022-11-12:
There may be an issue with initialisation of a shared variable that holds RAM contents.

The relevant sources are `src/common/retro/np65/np6532_ram.vhd` (top level of the RAM block), `src/common/basic/ram_tdp_s.vhd` (a dual port RAM, this contains the shared variable) and `build/np6532_functest/vhdl/np6532_ram_init_pkg.vhd` (constant data for initialising RAM contents, built by a Python script from binaries during the make process).

`np6532_ram` instantiates 4x `ram_tdp_s`, and passes each instance initial data via the `rambank2sl2d` function.

The latest commit demonstrates this problem. Run `make nvc` as before - the simulation stops rapidly because the soft CPU is seeing zeroes instead of valid instructions.

nickg, 2022-11-13:
Thanks, I managed to reproduce this. Could you open a separate issue to track it?

amb5l, 2022-11-13:
Will do.