nickg/nvc #571: Shared variable initial value

amb5l, 2022-11-14:
I am seeing an issue with initialisation of a shared variable that holds RAM contents. The variables seems to end up zeroed rather than holding the intended constant data.

The design is contained in [this commit](https://github.com/amb5l/tyto2/commit/21615515a900c64051cb53ca12df43607f3433b8). To demonstrate the issue, run `make nvc` in `build/np6532_functest` - the simulation stops rapidly because the soft CPU is seeing zeroes instead of valid instructions.

The relevant sources are `src/common/retro/np65/np6532_ram.vhd` (top level of the RAM block), `src/common/basic/ram_tdp_s.vhd` (a dual port RAM) and `build/np6532_functest/vhdl/np6532_ram_init_pkg.vhd` (constant data for initialising RAM contents, this VHDL file is built by a Python script from binaries during the make process).

`np6532_ram` instantiates 4x `ram_tdp_s`, and passes each instance initial data (as a generic) via the `rambank2sl2d` function. `ram_tdp_s` declares the `ram` shared variable, assigned to an initial value via the `ram_init` function.

I speculate that the issue is connected with the use a generic to pass the initial value, or its manipulation by the 2 functions.

nickg, 2022-11-14:
The initialisation error is fixed now but after running the simulation for a while I get this error:

```
** Note: 3975270ns+1: Report Note: PC = 22CC/22CC  S = FF/FF  P = 30/31  A = 04/04  X = 03/03
          Y = 04/04
         src/designs/np6532_functest/np6532_functest.vhd:181
** Note: 3975270ns+1: Report Note: *** MISMATCH ***
         src/designs/np6532_functest/np6532_functest.vhd:187
```

Is that expected?

amb5l, 2022-11-14:
I have the latest commit - thankyou - and have seen this error. It's not right. I'll look into it and get back to you.

amb5l, 2022-11-14:
OK I've narrowed it down. In `/src/common/retro/np65/np6532_core.vhd`, signal `s2_sel_flag_c` is wrongly assigned in this sequential assignment at line 656:
```
          s2_sel_flag_c <= SEL_FLAG_C_NOP;
          if    s1_id_flag_c = ID_FLAG_C_CLR then                             s2_flag_c_next <= '0';                   -- CLC
          elsif s1_id_flag_c = ID_FLAG_C_SET then                             s2_flag_c_next <= '1';                   -- SEC
          elsif s1_id_flag_c = ID_FLAG_C_SHF and s1_id_rmw /= ID_RMW_SHF then s2_sel_flag_c <= SEL_FLAG_C_SHF_A;
          elsif s1_id_flag_c = ID_FLAG_C_SHF and s1_id_rmw =  ID_RMW_SHF then s2_sel_flag_c <= SEL_FLAG_C_SHF_M;
          elsif s1_id_flag_c = ID_FLAG_C_ADD then                             s2_sel_flag_c <= SEL_FLAG_C_ADD;
          end if;
```
Note the relevant constant values:
```
ID_FLAG_C_NOP    = "000"
ID_FLAG_C_CLR    = "001"
ID_FLAG_C_SET    = "010"
ID_FLAG_C_ADD    = "011"
ID_FLAG_C_SHF    = "100"
ID_RMW_SHF       = '1'
SEL_FLAG_C_NOP   = "00"
SEL_FLAG_C_ADD   = "01"
SEL_FLAG_C_SHF_A = "10"
SEL_FLAG_C_SHF_M = "11"
```
I inserted some debug at line 656:
```
          if s1_reg_pc = x"22CB" then
            report "22CB: s1_id_flag_c = " & integer'image(to_integer(unsigned(s1_id_flag_c))) & 
            "  s1_id_rmw = " & bit'image(to_bit(s1_id_rmw));
          end if;
          if s1_reg_pc = x"22CC" then
            report "22CC: s2_sel_flag_c = " & integer'image(to_integer(unsigned(s2_sel_flag_c)));
          end if;
```
This shows that in the cycle before `s2_sel_flag_c` is wrongly assigned to "000", signal `s1_id_flag_c` = "100" and signal `s1_id_rmw` = '0'. Inspecting the `if` statement, it is apparent that these  should cause `s2_sel_flag_c` to be assigned to SEL_FLAG_C_SHF_A = "10".

amb5l, 2022-11-16:
Slightly better debug: insert an else clause in the if statement at line 656 as follows. This else clause should never be reached.
```
          s2_sel_flag_c <= SEL_FLAG_C_NOP;
          if    s1_id_flag_c = ID_FLAG_C_CLR then                             s2_flag_c_next <= '0';                   -- CLC
          elsif s1_id_flag_c = ID_FLAG_C_SET then                             s2_flag_c_next <= '1';                   -- SEC
          elsif s1_id_flag_c = ID_FLAG_C_SHF and s1_id_rmw /= ID_RMW_SHF then s2_sel_flag_c <= SEL_FLAG_C_SHF_A;
          elsif s1_id_flag_c = ID_FLAG_C_SHF and s1_id_rmw =  ID_RMW_SHF then s2_sel_flag_c <= SEL_FLAG_C_SHF_M;
          elsif s1_id_flag_c = ID_FLAG_C_ADD then                             s2_sel_flag_c <= SEL_FLAG_C_ADD;
          else -- DEBUG ISSUE 571
            if s1_id_flag_c /= "000" then
              report "bad else: s1_id_flag_c = " & integer'image(to_integer(unsigned(s1_id_flag_c))) & " s1_id_rmw = " & bit'image(to_bit(s1_id_rmw));
            end if;
          end if;
```
The simulation should end as follows:
```
** Note: 1439070ns+1: Report Note: bad else: s1_id_flag_c = 4 s1_id_rmw = '0'
         C:/work/tyto2/src/common/retro/np65/np6532_core.vhd:664
** Note: 1439490ns+1: Report Note: bad else: s1_id_flag_c = 4 s1_id_rmw = '0'
         C:/work/tyto2/src/common/retro/np65/np6532_core.vhd:664
** Note: 1439910ns+1: Report Note: bad else: s1_id_flag_c = 4 s1_id_rmw = '0'
         C:/work/tyto2/src/common/retro/np65/np6532_core.vhd:664
** Note: 1439940ns+1: Report Note: PC = 22CC/22CC  S = FF/FF  P = 30/31  A = 04/04  X = 03/03  Y = 04/04
         C:/work/tyto2/src/designs/np6532_functest/np6532_functest.vhd:181
** Note: 1439940ns+1: Report Note: *** MISMATCH ***
         C:/work/tyto2/src/designs/np6532_functest/np6532_functest.vhd:187
** Note: 1439940ns+1: Report Note: instruction count: 46408  cycle count: 46410
         C:/work/tyto2/src/designs/np6532_functest/np6532_functest.vhd:188
** Note: 1439940ns+1: FINISH called
   Procedure FINISH []
         ../lib/std.08/env-body.vhd:38
   Process :np6532_functest:do_test
         C:/work/tyto2/src/designs/np6532_functest/np6532_functest.vhd:142
```

nickg, 2022-11-16:
Is there a mistake in the definition of the `/=` operator in `tyto_utils_pkg.vhd`?

```diff
--- a/tyto2/src/common/tyto_utils_pkg.vhd
+++ b/tyto2/src/common/tyto_utils_pkg.vhd
@@ -51,7 +51,7 @@ package body tyto_utils_pkg is

   function "/=" (L: std_logic; R: std_logic) return std_logic is
   begin
-    return L xnor R;
+    return L xor R;
   end function "/=";

   function "=" (L: std_logic_vector; R: std_logic_vector) return std_logic is
```

But if this works in other simulators then it probably means there's another bug in how NVC applies the condition conversion operator. `if s1_id_flag_c = ID_FLAG_C_SHF and s1_id_rmw /= ID_RMW_SHF then` gets translated into `if ?? (s1_id_flag_c = ID_FLAG_C_SHF and s1_id_rmw /= ID_RMW_SHF) then` which is possibly not correct.

nickg, 2022-11-16:
The LRM says:

> The condition operator implicitly applied, if any, is either the predefined operator for type BIT or an overloaded operator, determined as follows. If, without overload resolution (see 12.5), the expression is of type BOOLEAN defined in package STANDARD, or if, assuming a rule requiring the expression to be of type BOOLEAN defined in package STANDARD, overload resolution can determine at least one interpretation of each constituent of the innermost complete context including the expression, then the condition operator is not applied. Otherwise, the condition operator is implicitly applied, and the type of the expression with the implicit application shall be BOOLEAN defined in package STANDARD.

But this can be interpreted as a boolean expression using predefined `=`, `/=` and `and` so it's incorrect to apply the implicit condition conversion.

amb5l, 2022-11-16:
Looks like you've found the root cause - well done.

And you are right about that incorrect overload of "/=". Thanks for spotting that.

nickg, 2022-11-17:
I think it should be fixed now, although I didn't run the simulation to completion. Could you test again?

amb5l, 2022-11-21:
Sorry for the delay. Will get this done and report back. I'm pretty sure that it will work...

amb5l, 2022-11-23:
Success so far. There are 2 ways to run the simulation. Unmodified the core is hammered with DMAs and interrupts - this takes many days. With the DMAs and interrupts disabled it takes half a day and this worked fine. I'll confirm the long test result eventually!

nickg, 2022-11-24:
I just pushed something that makes it run ~5% faster on my machine (I tested with `--stop-time=1ms`). It's not much but maybe it helps a little bit. 

The problem with this simulation (and a lot of other larger simulations that I've looked at) is the working set is too large to fit in cache so the CPU spends most of the time idling waiting for memory:

```
 Performance counter stats for 'nvc --std=2008 --work=.nvc/work -L.nvc -M 32m -r np6532_functest --stats --ieee-warnings=off --stop-time=500us':

    37,409,749,870      cycles
     8,297,898,075      stalled-cycles-backend    #   22.18% backend cycles idle
     2,220,148,111      cache-misses              #   38.457 % of all cache refs
     5,773,041,401      cache-references

       9.030520498 seconds time elapsed

       8.975985000 seconds user
       0.047978000 seconds sys
```

I need to think more carefully about how to shrink some of the internal data structures.

amb5l, 2022-11-27:
Thanks for this; I have been tied up but should be able to run the simulation this week.

Very interesting profiling data there. May I ask: how does NVC represent a std_logic signal (a single bit of RAM for this design)? Is it like  a C enumeration type that would fit in a single byte of memory?

nickg, 2022-11-27:
It takes a single byte if you declare it as a variable but a signal needs to track a lot more state. For example it needs to keep track of the previous value for `'last_value`, the timestamps for `'last_event` and `'last_active`, the set of drivers for the signal and all the scheduled transactions for those drivers, the set of processes that a are sensitive to that signal, and so on. It adds up quite quickly. NVC does various tricks to avoid storing all this state for each sub-element of an array signal if you always assign it as a block, but even still the current value of the signal is a small fraction of the total memory use.

amb5l, 2022-11-29:
Understood. The RAM in my design is a shared variable (not a signal as I stated previously) - so is implemented as one byte of computer memory per RAM bit. 128kbytes of design RAM => 1MBytes of computer memory. That's too big for L1 data cache, and pushing it for L2.

I could have an alternate RAM for long simulations, swapped in with a `configuration` declaration, with RAM locations implemented as `integer range 0 to 255` instead of `std_logic_vector(7 downto 0)` . I'll give it a try.

amb5l, 2022-12-01:
OK after 3 days - and within hours of finishing - my PC crashed (not for NVC related reasons). Since the sim ran fine up to this point I strongly recommend you close this issue! Thanks for the support.

closed 2022-12-01 by nickg