nickg/nvc #572: Coverage: Pragma cover off/on

Blebowski, 2022-11-14:
Add `coverage off` and `coverage on` pragma support.

nickg, 2022-11-14:
Pragmas like this are quite difficult to implement with the current parser structure because they don't have a clearly delimited scope. I'd rather we didn't add it. Can't you achieve the same effect with an exclude file?

nickg, 2022-11-14:
Let me think about this some more, I've got some ideas about how it could be implemented cleanly.

Blebowski, 2022-11-15:
Hello @nickg ,

use-case for this pragma is following:

```
case my_sel is
when "00" => ABC
when "01" => DEF
when "10" => GEH
when "11" => XYZ
-- cover off
when others =>
-- cover on
end case;
```

When such case statement is in generic block used many times across design, it is very convenient to handle
this unreachable choice directly in the source code.

If you exclude this type of statement in the "exclude" file, we would need some sort of "exclude per source file"
mechanism. Otherwise, we need to repeat the exclude for each instance of such generic block.

So, to answer your question: It is doable via exclude file, but its a inconvenient for user.

Implementation-wise, I was thinking about changing `tree_new` to a function pointer, and upon encountering
such pragma, switching this pointer to `tree_new_with_flag`, where each new `tree_t` which has I_FLAG,
would get new "T_FLAG_IGNORE_COVER". Such flag could then be used during lower phase not to emit
the cover tags for such tree element.

Problem with this approach is, that I would need to add I_FLAG to practically all `tree_t` which are marked
as coverable by `cover_is_stmt`, plus T_COND and T_SELECT. I need to think about how to do this more
efficiently.

nickg, 2022-11-15:
> Implementation-wise, I was thinking about changing `tree_new` to a function pointer, and upon encountering such pragma, switching this pointer to `tree_new_with_flag`, where each new `tree_t` which [...]

This proposal is way too hacky and intrusive, I wouldn't accept a PR that does this. I've got some ideas about how to handle this cleanly: I'll take a look this weekend.

Blebowski, 2022-11-15:
Thanks a lot :) Good to know in advance.

avelure, 2022-11-16:
I would prefer an exclude file option in addition. 
Say you want to reuse a module with a different parameter set, you then don't want to edit the code just because the parameter makes part of the code unreachable.

Blebowski, 2022-11-17:
I would opt for having both options:
1. Pragma for parts of code which are always unreachable - Travels with code itself.
2. Exclude file - Per hierarchy exclude in concrete design.
what do you think?

avelure, 2022-11-18:
Yes, sounds good.
I would also recommend having a look at the documentation on `cover exclude` for modlesim regarding which features would be good to include. There is an old copy of the manual here https://wiki.sj.ifsc.edu.br/wiki/images/0/03/Modelsim_pe_ref.pdf but I don't believe much has changed.

nickg, 2022-11-18:
@avelure do you know what format Modelsim uses for its coverage database? Is it plain XML in the .ucdb file?

tmeissner, 2022-11-18:
I'm not sure, but think it's a binary format. But you can export the ucdb content in xml-format. Which is what we do to post process it with homegrown Python tools.

avelure, 2022-11-18:
Yes, it's binary.
Could also be interesting to also be able to export to corbertura xml as it is handled by at least gitlab so that statistics can be displayed in the web interface.
Aldec has a python script to convert between the two here https://github.com/aldec/Riviera-PRO/blob/main/acdb2cobertura/ucdb2cobertura.py so it is not an urgent option.

nickg, 2022-11-19:
@Blebowski can you try the commit above bcd743e73 and see if that does what you want? Basically the idea is store a list of parsed pragmas with each design unit which the coverage code can query later.

Blebowski, 2022-11-20:
@nickg thanks. I have rebased my branch and tested the pragma on all 3 types of coverage. It seems to behave OK for me. Only minor issue on my side is bit of naming convention. In the existing options for coverage, I tried to stick to following so far:
- `Ignore` - Coverage is not even collected on these objects, no tags are emited for them. This is namely to reduce performance impact (e.g. ignoring the arrays).
- `Exclude` - Excluded tag is collected, and can be excluded right away at elaboration time
 (e.g. I would like to implement automatically excluding toggle coverage on instance ports driven by constant in above hierarchy), or further via "Exclude file" passed during report generation. Excluded tags will be listed as new category (Uncovered / Covered / Excluded), but they will be visible in the report.

In the code you wrote, everything is named "exclude" (`cover_exclude_from_pragmas`, or `exclude_lines`). Would you mind if I change it to "ignore" (e.g. `ignore_lines`), since pragma behavior prevents collection of cover tags alltogether...

I think later when I will try to do the exclude file implementation, and there will be a lot of "exclude" accross `cover.c`, it will be confusing, since each of them will mean different thing.

Blebowski, 2022-11-25:
Current format of pragma implementation seems working, closing Issue.

closed 2022-11-25 by Blebowski