nickg/nvc #573: Coverage: Exclude support

Blebowski, 2022-11-14:
1. [ ] Add coverage waiver file to allow classifying tag as "excluded" during report generation.
2. [ ] Add option to automatically exclude signals from toggle coverage when driven by constant value (either ports of instances, or internal signals).
3. [ ] Add option to automatically exclude expression bins which are not coverable due to some of its parts being constant.

Blebowski, 2023-01-22:
Hi @nickg , I am wondering how to implement the "auto-exclude" feature for toggle coverage on ports. Basically, if an instance has a port driven by a constant, then both toggle bins will be excluded. E.g. :

```
entity clk_gate is
    generic (
        G_TECHNOLOGY        :       natural
    );
    port (
        clk_in                : in    std_logic;
        clk_en               : in    std_logic;
        scan_enable    : in    std_logic;
        clk_out             : out  std_logic
    );
end clk_gate;
```

```
i_clk_gate : entity work.clk_gate
port map (
        clk_in              => sys_clk,
        clk_en             => my_clock_gate_signal,
        scan_enable  => '0',
        clk_out           => clk_gated
);
```

Then, I would like to have `scan_enable` toggle tags automatically excluded. The tricky thing is that, one would practically need to replicate everything in `lower_port_map` and `lower_map_signal` (recursively search for the 
value to get the `src_reg`), and then call `lower_have_signal` to find out whether the "driver" of the port is 
constant. Do you have any idea how to figure out port "constantness" this directly inside `lower_toggle_tag_for` ?

nickg, 2023-01-22:
I'll think about this but it might take a week or so as I'm quite busy at the moment.

Blebowski, 2023-01-23:
Thanks a lot, no rush on it.

nickg, 2023-01-29:
The code in `lower_port_map` is already too complicated with many different corner cases, I really don't want to duplicate it. Instead you can detect this quite easily `x_cover_setup_toggle_cb` which gets called after all the port maps have been processed:

```c
static bool is_constant_input(rt_signal_t *s)
{
   if (tree_kind(s->where) != T_PORT_DECL)
      return false;
   else if (tree_subkind(s->where) != PORT_IN)
      return false;
   else {
      rt_nexus_t *n = &(s->nexus);
      for (unsigned i = 0; i < s->n_nexus; i++, n = n->chain) {
         if (n->n_sources > 0)
            return false;
      }

      return true;
   }
}

....

   if (is_constant_input(s)) {
      // Set some flag to say this is always constant
   }

Blebowski, 2023-02-04:
Thanks, I will give it a try.

Blebowski, 2023-02-04:
The proposed code works nicely, however, it fails if the port is not a simple `std_logic`, but e.g. record consisting of multiple `std_logic` (not `T_PORT_DECL`, but `T_FIELD_DECL`). In such case, I would need to recursively search for parent to determine whether it is a port. 

Strictly speaking, it would not even need to be "port" to monitor this. Each unreachable "toggle" could be monitored. Is there a simple way how to figure out that the "signal" , "port" or field from one of those does
not have a driver which can toggle during simulation?

How does it work with the "nexus" ? Does the "nexus" correspond to a driver? I tried searching for "sources"
on the nexuses also on `T_SIGNAL_DECL`, but those never return any sources.

nickg, 2023-02-04:
> The proposed code works nicely, however, it fails if the port is not a simple std_logic, but e.g. record consisting of multiple std_logic (not T_PORT_DECL, but T_FIELD_DECL). In such case, I would need to recursively search for parent to determine whether it is a port.

I think you can just search upwards using the `parent` field of `rt_signal_t` until `s->parent->kind != SCOPE_SIGNAL`. That will find the `rt_scope_t` that corresponds to the root of the composite signal, and its `where` field will point at the signal declaration.

> Strictly speaking, it would not even need to be "port" to monitor this. Each unreachable "toggle" could be monitored. Is there a simple way how to figure out that the "signal" , "port" or field from one of those does not have a driver which can toggle during simulation?

There will be a `rt_source_t` with `source->tag == SOURCE_DRIVER` if the signal is driven by a process. But those won't exist at the time the toggle setup callback is called because drivers are created the first time a process is executed, which happens afterwards. So at that point the only sources will be `SOURCE_PORT`.

> How does it work with the "nexus" ? Does the "nexus" correspond to a driver?

A nexus is just a slice of a signal. It's an optimisation to avoid having to keep track of the full state of every scalar sub-element. E.g.

```vhdl
signal x : std_logic_vector(3 downto 0);
...
x(3 downto 1) <= "111"
x(0) <= "0";
```

Signal `x` has four scalar sub-elements but only two nexuses (one with 3 elements and the other with a single element). `sizeof(rt_nexus_t)` is around 128B so the savings are quite big if there are large signals that are only ever assigned as a block.

Blebowski, 2023-02-04:
OK, thanks, that explains why I get `n_sources == 0` on signals toggling by a process.

Blebowski, 2023-02-04:
It is tricky, to get the details correct here, e.g. for record port:

```
type t_nested_record is record
        a       : natural;
        b       : std_logic;
        c       : std_logic_vector(2 downto 0);
    end record;

    type t_record is record
        x       : std_logic;
        y       : std_logic_vector(3 downto 0);
        z       : t_nested_record;
    end record;
```

when driven like so:

```
record_port          => ('0', "0000", (0, '0', "000")),
```

finding the root scope to be port works just fine. Flagging all tags with offset up to `s->shared.size` is OK, and I manage to exclude tags on all toggle tags on the constant port. However, e.g. following scenario:

```
mixed_port          => (for_sure_toggling_signal, "0000", (0, '0', "000")),
```

I iterate up to `where` which is `T_SIGNAL_DECL`, and I am not able to detect that `y`, `z.b` and `z.c` are driven by constants.

nickg, 2023-02-04:
I don't think you need to iterate over all the fields. The setup function will be called separately for each.

Blebowski, 2023-02-04:
Srry, I don't iterate over the fields, I walk via the hierarchy up to `rt_scope_t` with `s->parent->kind != SCOPE_SIGNAL`. Then, I take `where` and look at it for being `T_PORT_DECL` and do all the processing you proposed. However, If any of fields is actually driven by signal, I get to `T_SIGNAL_DECL` for all callbacks.

closed 2023-03-28 by Blebowski