nickg/nvc #574: Type conversion considered non-locally static in 2008

avelure, 2022-11-18:
I'm always unsure about these localy-static definitions, but I think that the `to_integer` function should not render a constant non-locally static in 2008.
In 2008 LRM section 9.4.2 point J:  A type conversion whose type mark denotes a locally static subtype and whose expression is a
locally static expression

Modelsim and Aldec gives warning when compiling for 93, but nothing for 2008. GHDL giver error, but it possibly is the same bug.

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package test is
  type t_memory_map_array is array (natural range 0 to 10) of natural;
  constant C_VER : unsigned(4 downto 0) := "00001";
  constant C_MEMORY_MAP_ITEM_DEFAULT : natural := 0;
  constant C_MEMORY_MAP_DEFAULT : t_memory_map_array := (
    to_integer(C_VER) => 1,
    others => C_MEMORY_MAP_ITEM_DEFAULT);
end package;
```

```
nvc --std=08 -a --relaxed test_non_static.vhd
** Error: a choice that is not locally static is allowed only if the array aggregate contains a single element association
    > test_non_static.vhd:10
    |
 10 |     to_integer(C_VER) => 1,
    |     ^^^^^^^^^^^^^^^^^
```

nickg, 2022-11-18:
`to_integer` isn't a type conversion, it's actually a call to a regular function defined in the `numeric_std` package. But you're correct this should be allowed in -2008: the standard just makes all calls to IEEE functions locally static by definition. Seems like a bit of a hack but we still ought to implement it.

avelure, 2022-11-18:
Ah, so this is related to the change for the point e) instead
```
A function call whose function name denotes an implicitly defined operation or an operation defined
in one of the packages STD_LOGIC_1164, NUMERIC_BIT, NUMERIC_STD,
NUMERIC_BIT_UNSIGNED, or NUMERIC_STD_UNSIGNED in library IEEE and whose actual
parameters are each locally static expressions
```

which includes all the IEEE libraries, where in 93 it is only the implicit ones.
```
d) A function call whose function name denotes an implicitly defined operator, and whose actual
parameters are each locally static expressions
```

nickg, 2022-11-19:
I've implemented this now - could you try your simulation again?

avelure, 2022-11-19:
Yes, this looks ok, thanks!

closed 2022-11-20 by nickg