nickg/nvc #575: fatal: cannot handle value kind 5 in lvn_value_num

avelure, 2022-11-18:
I noticed this regression in one of my testbenches.

ubuntu llvm12
```
fatal: cannot handle value kind 5 in lvn_value_num
[0x55cc58cf5713] ../src/util.c:542 fatal_trace
[0x55cc58c670e4] ../src/jit/jit-optim.c:339 lvn_value_num.isra.0
[0x55cc58c67137] ../src/jit/jit-optim.c:363 lvn_get_tuple
[0x55cc58c671c6] ../src/jit/jit-optim.c:381 jit_lvn_generic
[0x55cc58c6ff6f] /usr/include/x86_64-linux-gnu/bits/stdlib.h:513 jit_do_lvn
[0x55cc58c6ff6f] ../src/jit/jit-optim.c:3721 jit_irgen
[0x55cc58c71543] ../src/jit/jit-interp.c:845 jit_interp
[0x55cc58c71892] ../src/jit/jit.h:568 interp_loop
[0x55cc58c71892] ../src/jit/jit-interp.c:877 jit_interp
[0x55cc58c71892] ../src/jit/jit.h:568 interp_loop
[0x55cc58c71892] ../src/jit/jit-interp.c:877 jit_interp
[0x55cc58c71892] ../src/jit/jit.h:568 interp_loop
[0x55cc58c71892] ../src/jit/jit-interp.c:877 jit_interp
[0x55cc58c71892] ../src/jit/jit.h:568 interp_loop
[0x55cc58c71892] ../src/jit/jit-interp.c:877 jit_interp
[0x55cc58c71892] ../src/jit/jit.h:568 interp_loop
[0x55cc58c71892] ../src/jit/jit-interp.c:877 jit_interp
[0x55cc58c71892] ../src/jit/jit.h:568 interp_loop
[0x55cc58c71892] ../src/jit/jit-interp.c:877 jit_interp
[0x55cc58c72b69] ../src/jit/jit-core.c:611 jit_try_vcall.isra.0
[0x55cc58c785d0] ../src/jit/jit-core.c:753 jit_call_thunk
[0x55cc58c8a8f7] ../src/eval.c:130 eval_do_fold.isra.0.lto_priv.0
[0x55cc58cc4831] ../src/diag.c:251 simp_fcall
[0x55cc58cc4831] ../src/simp.c:1637 simp_tree.lto_priv.0
[0x55cc58cb1ae9] ../src/object.c:611 object_rewrite_iter
[0x55cc58cb19cd] ../src/object.c:644 object_rewrite
[0x55cc58cb1a20] ../src/object.c:675 object_rewrite
[0x55cc58cb18b8] ../src/object.c:686 object_rewrite
[0x55cc58cb1a20] ../src/object.c:675 object_rewrite
[0x55cc58cb18b8] ../src/object.c:686 object_rewrite
[0x55cc58cb1a20] ../src/object.c:675 object_rewrite
[0x55cc58cb18b8] ../src/object.c:686 object_rewrite
[0x55cc58cd6076] ../src/tree.c:1206 tree_rewrite
[0x55cc58cc329e] ../src/simp.c:1795 simplify_global
[0x55cc58cca8cc] ../src/elab.c:1779 elab_top_level.lto_priv.0
[0x55cc58d000fa] ../src/vhpi/vhpi-model.c:1846 elaborate
[0x55cc58d000fa] ../src/elab.c:1201 process_command
[0x55cc58c45143] ../src/nvc.c:1320 main
```

mingw llvm15
```
fatal: cannot handle value kind 5 in lvn_value_num
[00007FF7841E68DA] 
[00007FF78428F74C] _nvc_get_foreign+0x94c
[00007FF78428F77A] _nvc_get_foreign+0x97a
[00007FF78428FB2B] _nvc_get_foreign+0xd2b
[00007FF784289053] nvc_current_delta+0x15203
[00007FF78428C118] nvc_current_delta+0x182c8
[00007FF78428B7F6] nvc_current_delta+0x179a6
[00007FF78428B7F6] nvc_current_delta+0x179a6
[00007FF78428B7F6] nvc_current_delta+0x179a6
[00007FF78428B7F6] nvc_current_delta+0x179a6
[00007FF78428B7F6] nvc_current_delta+0x179a6
[00007FF78428B7F6] nvc_current_delta+0x179a6
[00007FF78427C38D] nvc_current_delta+0x853d
[00007FF78427CFCD] nvc_current_delta+0x917d
[00007FF78426A841] 
[00007FF784236826] 
[00007FF784236890] 
[00007FF78423675F] 
[00007FF784236890] 
[00007FF78423675F] 
[00007FF784236890] 
[00007FF78423675F] 
```

nickg, 2022-11-18:
I think this should be fixed by the commit above. Could you test again?

avelure, 2022-11-18:
Yes, that solved it, thanks for the quick update.

closed 2022-11-19 by nickg