nickg/nvc #577: Crash when compiling OSVVM 2022.10

wyvernSemi, 2022-11-29:
NVC version 1.7.2. On both Linux and Windows, when building [OSVVM 2022.10](https://github.com/OSVVM/OsvvmLibraries) an exception is generated when analyzing MemoryPkg.vhd. The commands used, from within a tclsh, were:

source <path>/OsvvmLibraries/Scripts/StartNVC.tcl
build <path>/OssvmLibraries

Logs below:

Linux (Ubuntu 2020.04.5 LTS), NVC 1.7.2
--------------------------------------------

    analyze MemoryPkg.vhd
    nvc --std=08 -H 128m --work=osvvm:/home/simon/OSVVM/nvc/VHDL_LIBS/nvc-1.7.2/OSVVM -L 
    /home/simon/OSVVM/nvc/VHDL_LIBS/nvc-1.7.2 -a ../../git/OSVVM/OsvvmLibraries/osvvm/MemoryPkg.vhd

    *** Caught signal 11 (SIGSEGV) [address=(nil), ip=0x562dac76356c] ***

    [0x562dac7ab19f] ../src/util.c:793 signal_handler.lto_priv.0
    [0x7f68bda6608f] (/lib/x86_64-linux-gnu/libc-2.31.so) 
    [0x562dac76356c] ../src/lower.c:697 lower_bounds.lto_priv.0
    [0x562dac763a00] /usr/include/x86_64-linux-gnu/sys/mman.h:422 lower_array_left
    [0x562dac763a00] ../src/lower.c:418 lower_array_left.lto_priv.0
    [0x562dac765f81] ../src/lower.c:815 lower_wrap_with_new_bounds.lto_priv.0
    [0x562dac75c082] lower_new
    [0x562dac75c082] ../src/lower.c:4151 lower_expr.lto_priv.0
    [0x562dac75f35f] lower_var_assign
    [0x562dac75f35f] ../src/lower.c:6119 lower_stmt.lto_priv.0
    [0x562dac75e909] lower_if
    [0x562dac75e909] ../src/lower.c:6131 lower_stmt.lto_priv.0
    [0x562dac755198] ../src/lower.c:8614 lower_decls
    [0x562dac75542e] ../src/lower.c:7370 lower_decls
    [0x562dac7586bb] lower_package
    [0x562dac7586bb] ../src/lower.c:9593 lower_unit
    [0x562dac7b2be1] ../src/nvc.c:166 analyse
    [0x562dac7b345d] ../src/nvc.c:1043 process_command
    [0x562dac7024b2] ../src/nvc.c:1162 main
    AnalyzeError: See messages above in "analyze MemoryPkg.vhd"
    For tcl errorInfo, puts $::osvvm::AnalyzeErrorInfo

Windows 10/MSYS2, NVC 1.7.2
----------------------------------

    analyze MemoryPkg.vhd
    nvc --std=08 -H 128m --work=osvvm:C:/git/OSVVM/nvc/VHDL_LIBS/nvc-1.7.2/OSVVM -L 
    C:/git/OSVVM/nvc/VHDL_LIBS/nvc-1.7.2 -a
    ../OsvvmLibraries/osvvm/MemoryPkg.vhd

    *** Caught exception c0000005 (EXCEPTION_ACCESS_VIOLATION) [address=0000000000000000, ip=00007ff6fd301100] ***

    AnalyzeError: See messages above in "analyze MemoryPkg.vhd"

amb5l, 2022-11-29:
I am seeing similar problems with [the latest snapshot of NVC](https://github.com/nickg/nvc/commit/2214be934e50cbd53192574568ced5b0130cf2a2) tested against the [latest snapshot of OsvvmLibraries](https://github.com/OSVVM/OsvvmLibraries/commit/3659630994a6a2e13214e819d283874bae809530):
```
C:\work\OsvvmLibraries\tmp>tclsh
% source ../Scripts/StartNVC.tcl
OSVVM Script Version:  2022.11
Simulator Version:     nvc-1.8-devel
% build ../OsvvmLibraries.pro
creating directory C:/work/OsvvmLibraries/tmp/logs/nvc-1.8-devel
build ../OsvvmLibraries.pro
Starting Build at time 16:13:55
include ../OsvvmLibraries.pro
creating directory C:/work/OsvvmLibraries/tmp/VHDL_LIBS/nvc-1.8-devel
library default C:/work/OsvvmLibraries/tmp/VHDL_LIBS/nvc-1.8-devel
nvc --std=08 --work=default:C:/work/OsvvmLibraries/tmp/VHDL_LIBS/nvc-1.8-devel/DEFAULT --init
set CurrentWorkingDirectory ..
source ../OsvvmLibraries.pro
include ./osvvm/osvvm.pro
set CurrentWorkingDirectory ../osvvm
source ../osvvm/osvvm.pro
library osvvm C:/work/OsvvmLibraries/tmp/VHDL_LIBS/nvc-1.8-devel
nvc --std=08 --work=osvvm:C:/work/OsvvmLibraries/tmp/VHDL_LIBS/nvc-1.8-devel/OSVVM --init
analyze TextUtilPkg.vhd
nvc --std=08 -H 128m --work=osvvm:C:/work/OsvvmLibraries/tmp/VHDL_LIBS/nvc-1.8-devel/OSVVM -L C:/work/OsvvmLibraries/tmp/VHDL_LIBS/nvc-1.8-devel -a ../osvvm/TextUtilPkg.vhd
...
analyze MemoryPkg.vhd
nvc --std=08 -H 128m --work=osvvm:C:/work/OsvvmLibraries/tmp/VHDL_LIBS/nvc-1.8-devel/OSVVM -L C:/work/OsvvmLibraries/tmp/VHDL_LIBS/nvc-1.8-devel -a ../osvvm/MemoryPkg.vhd

*** Caught exception c0000005 (EXCEPTION_ACCESS_VIOLATION) [address=0000000000000000, ip=00007ff71a7715f0] ***

AnalyzeError: See messages above in "analyze MemoryPkg.vhd"
```
In more detail:
```
C:\work\OsvvmLibraries\tmp>nvc --std=08 -H 128m --work=osvvm:C:/work/OsvvmLibraries/tmp/VHDL_LIBS/nvc-1.8-devel/OSVVM -L C:/work/OsvvmLibraries/tmp/VHDL_LIBS/nvc-1.8-devel -a ../osvvm/MemoryPkg.vhd

*** Caught exception c0000005 (EXCEPTION_ACCESS_VIOLATION) [address=0000000000000000, ip=00007ff71a7715f0] ***

[00007ff71a746ec4]
[00007fffb57f6b1c] UnhandledExceptionFilter+0x1ec
[00007fffb7dd861d] RtlMoveMemory+0x2c9d
[00007fffb7dbf087] _C_specific_handler+0x97
[00007fffb7dd3eaf] _chkstk+0x12f
[00007fffb7d4eae6] RtlFindCharInUnicodeString+0xa96
[00007fffb7dd2e9e] KiUserExceptionDispatcher+0x2e
[00007ff71a7715f0]
[00007ff71a79a2ff]
[00007ff71a79e0fc]
[00007ff71a79ede1]
[00007ff71a79c81b]
[00007ff71a79dee4]
[00007ff71a7aa1ba]
[00007ff71a7a8eea]
[00007ff71a7ac3a7]
[00007ff71a7ac047]
[00007ff71a7af586]
[00007ff71a742cc7]
[00007ff71a82da4a] vhpi_put_data+0x2464a
[00007ff71a7413ae]
[00007ff71a7414e6]
```

amb5l, 2022-11-29:
@wyvernSemi [this binary package for MSYS2](https://www.nickg.me.uk/files/mingw-w64-x86_64-nvc-1.7.0.r209.g4ef679f1-1-any.pkg.tar.zst) is worth trying.

nickg, 2022-11-29:
The crash with the latest OSVVM dev branch is caused by line 591 in MemoryGenericPkg.vhd:

```vhdl
       MemStructPtr(ID).MemArrayPtr(BlockAddr) := new 
            MemBlockType'(InitMemoryBlockType(BlockWidth, MemoryBaseWidth)) ;
```

@wyvernSemi I've previously tested 2022.05d with the 1.7 branch, later versions likely need some fixes that are only on the master branch. Is it possible to use the latest development version for the time being, or try the MSYS2 package @amb5l linked to which is from a month or so ago?

wyvernSemi, 2022-11-29:
@nickg I'll try these in the morning (I'm in the UK)

wyvernSemi, 2022-11-30:
@nickg I tried with the latest development code (1.8-devel) and it does the same thing as 1.7.2. I have temporarily reverted to older OSVVM memory components  and (with some other hacks) compiled OsvvmLibraries successfully. 

I can't simply go back to an older OSVVM release as I am adding co-simulation features to OSVVM and need the latest code to work on. However, on Linux, with my OSVVM hacks, I can get co-simulation tests working on NVC and verify co-sim support on it. I'm having issues with VHPIDIRECT elaboration on MSYS2 (including with the vhpi4 example in NVC that works on Linux) but this needs more investigation. I may try the MSYS package of @ambl, but I don't want to destroy my MSYS2 setup as other tools I use are sensitive to versions.

nickg, 2022-11-30:
@wyvernSemi this should be fixed now, could you test again with the latest master branch?

nickg, 2022-11-30:
@amb5l I tried updating the OSVMM version to 2022.10 in `install-osvvm.sh` but I got this TCL error, any ideas?

```
Working in /home/nick/.cache/nvc/OsvvmLibraries-2022.10/Ethernet:
invalid command name "ChangeWorkingDirectory"
    while executing
"ChangeWorkingDirectory src"
    (file "Ethernet.pro" line 41)
    invoked from within
"source [file tail $PathAndFile]"
    (procedure "include" line 5)
    invoked from within
"include ./Ethernet/Ethernet.pro"
    invoked from within
"if {[DirectoryExists Ethernet]} {
  include ./Ethernet/Ethernet.pro
}"
    (file "OsvvmLibraries.pro" line 52)
    invoked from within
"source "OsvvmLibraries.pro""
    (file "/home/nick/.local/libexec/nvc/install-osvvm.tcl" line 36)
```

wyvernSemi, 2022-12-01:
@nickg I have tried out the latest code and it has fixed the crash from before (thanks). I do get a new issue though when it compiles the Ethernet VC code (a new component in OSVVM 2022.10). I've attached the log
[nvc_log.txt](https://github.com/nickg/nvc/files/10131848/nvc_log.txt). Maybe this should be a new issue with this one closed?

I'm also (unfortunately) compiling a new issue when elaborating code referencing foreign VHPIDIRECT procedures that I only see on Windows/MSYS2 but works on Linux. It may be a few days as I am travelling from tomorrow.

Thanks for being so responsive. I am keen to have the developing OSVVM co-simulation supported on NVC. We're close, as I already have this running on NVC in Linux. (if I don't compile the OSVVM Ethernet component).

amb5l, 2022-12-01:
@nickg I'll look at the `install-osvvm.sh` issue now and report back. @wyvernSemi I'll also try to reproduce the Ethernet VC issue.

amb5l, 2022-12-01:
NVC is supported in OSVVM 2022.10 onwards, so `install-osvvm.tcl` can be simplified:
```
source "Scripts/StartNVC.tcl"
set ::osvvm::VhdlLibraryDirectory "${::env(NVC_INSTALL_DEST)}"
set ::osvvm::VhdlLibrarySubdirectory "."
source "OsvvmLibraries.pro"
```
This resolves the TCL error mentioned above, but we then see the problem with the Ethernet VC described by @wyvernSemi. For example:
```
nvc --std=08 -H 128m --work=osvvm_ethernet:C:/work/.nvc/lib/./OSVVM_ETHERNET -L C:/work/.nvc/lib/. -a Ethernet/src/xMiiPhyRxTransmitter.vhd
** Error: ambiguous call to function NEWID
     > C:\work\.cache\nvc\OsvvmLibraries-2022.10\Ethernet\src\xMiiPhyRxTransmitter.vhd:118
     |
 118 |     DataFifo     <= NewID("DataFifo",  ID, ReportMode => DISABLED, Search => PRIVATE_NAME) ;
     |                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
     |                     use of name NEWID here
     |
     = Note: candidate NEWID [STRING, ALERTLOGIDTYPE, ALERTLOGREPORTMODETYPE, NAMESEARCHTYPE, ALERTLOGPRINTPARENTTYPE return SCOREBOARDIDTYPE] from OSVVM.SCOREBOARDPKG_SLV
     = Note: candidate NEWID [STRING, ALERTLOGIDTYPE, ALERTLOGREPORTMODETYPE, NAMESEARCHTYPE, ALERTLOGPRINTPARENTTYPE return SCOREBOARDIDTYPE] from OSVVM.SCOREBOARDPKG_INT
```
This specific error can be worked around by prepending `osvvm.ScoreboardPkg_slv.` to `NewID`, but I think that NVC should resolve the ambiguity because it knows the type of `DataFifo`:
```
signal DataFifo : osvvm.ScoreboardPkg_slv.ScoreboardIDType ;
```

nickg, 2022-12-02:
The second error from the Ethernet VC is fixed as well now. I ran all the OSVVM tests successfully with 2022.10 and also bumped the version installed by `nvc --install osvvm`. Thanks for reporting these!

closed 2022-12-02 by nickg