nickg/nvc #579: Fails on i386: ld: error: undefined symbol: _GLOBAL_OFFSET_TABLE_

yurivict, 2022-12-04:
```
json.o src/bounds.o src/make.o  src/object.o src/lower.o src/vcode.o  src/names.o src/debug.o src/eval.o  src/opt.o src/diag.o src/scan.o  src/mask.o src/thread.o  src/rt/rtkern.o src/rt/alloc.o  src/rt/heap.o src/rt/pprint.o  src/rt/cover.o src/rt/wave.o  src/rt/jit.o src/rt/ffi.o  src/rt/mspace.o src/jit/jit-core.o  src/jit/jit-irgen.o src/jit/jit-dump.o  src/jit/jit-interp.o 
llvm-ranlib lib/libnvc.a 
--- bin/nvc ---
cc -Wall  -flto=auto -pthread    -O2 -pipe  -fstack-protector-strong -isystem /usr/local/include -fno-strict-aliasing  -lexecinfo -fstack-protector-strong  -L/usr/local/llvm90/lib -lpthread -Wl,--dynamic-list=/wrkdirs/usr/ports/cad/nvc/work/nvc-r1.7.1/src/symbols.txt   -o bin/nvc src/nvc.o lib/libnvc.a  lib/libcgen.a  lib/libfst.a  lib/libfastlz.a  lib/libjson.a  lib/libvhpi.a  lib/libcpustate.a  -lLLVM-9   -ldwarf -lelf -lz -lm -L/usr/local/lib
ld: error: undefined symbol: _GLOBAL_OFFSET_TABLE_
>>> referenced by mulodi4.c:20 (/usr/src/contrib/llvm-project/compiler-rt/lib/builtins/mulodi4.c:20)
>>>               mulodi4.o:(__mulodi4) in archive /usr/lib/libgcc.a
cc: error: linker command failed with exit code 1 (use -v to see invocation)
*** [bin/nvc] Error code 1

```

Version: 1.7.1
clang-14
llvm-90
FreeBSD 13.1

nickg, 2022-12-09:
Is is possible to reproduce this by cross-compiling from an amd64 FreeBSD system?

yurivict, 2022-12-09:
It is easy to install FreeBSD i386 into a VirtualBox VM.

nickg, 2022-12-11:
It seems to be a problem with LTO: it links OK if I disable that. I suggest passing `--disable-lto` to the configure script on i386.

yurivict, 2022-12-11:
Does this look like a toolchain problem?

I will update the port, thank you.

yurivict, 2022-12-11:
I updated the port, and reported this as a bug in the toolchain.

Thank you very much for investigating this problem!

closed 2022-12-11 by yurivict

nickg, 2022-12-11:
> Does this look like a toolchain problem?
> 

I'm not really sure what the problem is but it seems the symbol `__mulodi4` is defined in both `compiler-rt` (the LLVM runtime library) and `libgcc` (the GCC runtime library). Normally I'd expect the executable to be linked against only one of those, I guess compiler-rt if `cc` is Clang.

yurivict, 2022-12-11:
In case you still have this setup, are you able to verify if the problem exists when LLVM_VERSION=15 (the latest LLVM) ?

nickg, 2022-12-11:
That failed with the same error, see the log below:

[nvc.port.log](https://github.com/nickg/nvc/files/10203333/nvc.port.log)