nickg/nvc #580: Compilation error when compiling OSVVM 2022.11

wyvernSemi, 2022-12-05:
Updating to OSVVM 2022.11 an error is reported whilst compiling UART/src/UartTbPkg.vhd (log attached 
[OSVVM_OsvvmLibraries.log](https://github.com/nickg/nvc/files/10153697/OSVVM_OsvvmLibraries.log)). This file has changed over 2022.10, starting at line 86, from:
```
    type UartRecArrayType is array (integer range <>) of UartRecType ; 
```
to
```
    subtype UartRecArrayType is StreamRecArrayType(open)(
       DataToModel   (UartTb_DataType'range), 
       ParamToModel  (UartTb_ErrorModeType'range), 
       DataFromModel (UartTb_DataType'range), 
       ParamFromModel(UartTb_ErrorModeType'range) 
     ) ;
```

nickg, 2022-12-05:
This error should be fixed now. I've run the full set of OSVVM tests on 2022.11 and the "InterruptHandler" test suite fails. I guess that was a new addition with this release as well.

wyvernSemi, 2022-12-06:
@nickg I've successfully tried this, including running the new co-simulation tests (not yet released).

The interrupt handler tests (Common/TbInterrupt/TbInterrupt.pro) fail to compile---hence fail. The Reference to InterruptHandler in Common/TbInterrupt/testbench/TbAxi4.vhd (and TbAxi4Memory.vhd) is not recognised (log: 
[TbInterrupt.log](https://github.com/nickg/nvc/files/10164602/TbInterrupt.log)). I see the relevant files being compiled into the OSVVM_COMMON library, and there are generated files in VHDL_LIBS/nvc-1.8-devel/OSVVM_COMMON/ (e.g. OSVVM_COMMON.INTERRUPTHANDLER). I tried building the TbInterupt.pro script on GHDL and QuestaSim and the tests compile and run.

amb5l, 2022-12-06:
I've checked out and tried the latest main branch of OsvvmLibraries against the latest build of NVC and I am seeing the interrupt handler failure. To reproduce quickly, run `tclsh` in the root of the OsvvmLibraries repo and...
```
source Scripts/StartNVC.tcl
build OsvvmLibraries
include Common/TbInterrupt
```
The first error looks like this:
```
nvc --std=08 -H 128m --work=InterruptHandler:C:/work/OsvvmLibraries/tmp/VHDL_LIBS/nvc-1.8-devel/INTERRUPTHANDLER -L C:/work/OsvvmLibraries/tmp/VHDL_LIBS/nvc-1.8-devel -a ../Common/TbInterrupt/testbench/TbAxi4.vhd
** Error: object INTERRUPTHANDLER is not a component declaration
     > C:\work\OsvvmLibraries\Common\TbInterrupt\testbench\TbAxi4.vhd:172
     |
 172 |   InterruptHandler_1 : InterruptHandler
     |                        ^^^^^^^^^^^^^^^^
```
Looking for component `InterruptHandler`:
`Common/TbInterrupt/testbench/TbAxi4.vhd` includes `Axi4Context`
`AXI4/Axi4/src/Axi4Context.vhd` includes `OsvvmCommonContext`
`Common/src/OsvvmCommonContext.vhd` includes `InterruptHandlerComponentPkg`
`Common/src/InterruptHandlerComponentPkg.vhd` declares component `InterruptHandler`

nickg, 2022-12-06:
The InterruptHandler failure should be fixed by commit above. I ran `test/test-osvvm.tcl` with 2022.11 and it passed without error. Thanks!

closed 2022-12-06 by nickg