nickg/nvc #581: Elaboration issue with foreign procedures on MSYS2

wyvernSemi, 2022-12-07:
When elaborating code in MSYS2 that references VHPIDIRECT foreign procedures/functions I see an elaboration error with references to the foreign procedures not found. This is not seen when building on Linux. I have managed to boil this issue down by using the vhpi4 test files from NVC. The following commands were issued:

```
gcc -shared -fPIC vhpi4.c - o vhpi4.so
nvc -a vhpi4.vhd
nvc -e vhpi4
```

On the elaboration step I get the following error message:
```
C:/Tools/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\simon\Desktop\work\_WORK.VHPI4.elab.0.8628.obj: in function `WORK.VHPI4.MAIN':
./vhpi4.vhd:28: undefined reference to `__vhpi_sum'
C:/Tools/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: ./vhpi4.vhd:29: undefined reference to `__vhpi_sum_array'
C:/Tools/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: ./vhpi4.vhd:30: undefined reference to `__vhpi_my_not'
C:/Tools/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: ./vhpi4.vhd:31: undefined reference to `__vhpi_my_not'
C:/Tools/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: ./vhpi4.vhd:32: undefined reference to `__vhpi_my_not'
C:/Tools/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: ./vhpi4.vhd:34: undefined reference to `__vhpi_test_proc'
collect2.exe: error: ld returned 1 exit status
** Fatal: gcc -shared -o C:\Users\simon\Desktop\work\_WORK.VHPI4.elab.dll C:\Users\simon\Desktop\work\_WORK.VHPI4.elab.0.8628.obj
          -LC:\Tools\msys64\mingw64\lib\nvc -lnvcimp failed with status 1
```
I also discovered that if I take the actual gcc command displayed in the fatal message and tag vhpi4.so on the end it does compile and the run command works afterwards (also adding -L. -l:vhpi4.so worked) . I tried adding the current directory to the LD_LIBRARY_PATH but this made no difference. I would normally expect that dependant libraries (including shared objects) would need to be listed in the compile to link to the undefined symbols, though this does not seem to be needed on Linux.

I was compiling in MSYS2 MINGW64 (which is the environment that NVC was also built). Is this a gcc version thing or an MSYS2 environment thing maybe? I assume that the vhpi4 test works under MSYS2 elsewhere. Have I built NVC correctly on MSYS2? This issue is preventing the successful running of OSVVM co-simulation tests on NVC under MSYS2, which are all functioning under Linux, as I get the same types of errors for my foreign procedures.

nickg, 2022-12-09:
The vhpi4 test is actually disabled on Windows (the `!windows` on the corresponding line in `test/regress/testlist.txt`), I think because of this issue. Linux and most other Unix-like systems are fine with symbols being unresolved when linking shared libraries - as long as they can be resolved later when the library is loaded - but Windows needs everything resolved up-front. The traditional way to do this on Windows is with a static import library (`-Wl,--out-implib=libname.a`) but it seems linking directly against the DLL works too now, as you discovered above.

There's an (unfortunately undocumented) environment variable `NVC_FOREIGN_OBJ` which you can use to add extra object files to the link command NVC generates. So I think the following should work:

* Link your code with `-Wl,--out-implib=mylib.a`
* `export NVC_FOREIGN_OBJ=mylib.a`
* Elaborate as usual and then make sure the corresponding `.dll` is in the PATH when you run, because it will be loaded directly by the system rather than the `--load` argument.

Longer term I want to move away from linking directly against the user's code and instead call foreign functions through an indirection layer (libffi). Once that's done the behaviour should be identical on all platforms (i.e. everything resolved dynamically at runtime).

wyvernSemi, 2022-12-10:
Success! 

Though it was slightly different than with your instructions. I generated the import library and set NVC_FOREIGN_OBJ to it but got the same error. I'd updated PATH (and LD_LIBRARY_PATH) but no joy. I tried setting NVC_FOREIGN_OBJ to my own static library (as this is an intermediate step to building the shared object) and got better results, but then it complained about missing functions from system libraries I use (e.g. libdl). So I tried simply setting NVC_FOREIGN_OBJ to my shared object directly, as adding that to the raw gcc command manually had worked before, and it all just functioned correctly.

All my OSVVM co-simulation regression tests now pass on NVC under Windows/MSYS. I will see if I can find a way of setting the NVC_FOREIGN_OBJ automatically so a user doesn't have to do it themselves. For me, though, this issue is closed.

closed 2023-02-07 by nickg