nickg/nvc #591: EXCEPTION_ACCESS_VIOLATION on arrays of large vectors

avelure, 2023-01-16:
```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
end entity test;
architecture beh of test is
begin
  
  process
    variable v_slv_long_min  : std_logic_vector(127 downto 0) := (others => '0');
    variable v_slv_long_max  : std_logic_vector(127 downto 0) := (others => '0');
    type t_unsigned_vector is array (natural range <>) of unsigned;
    type t_range_uns_vec is array (natural range <>) of t_unsigned_vector;

    procedure check_value_long(
      constant range_vec : in t_range_uns_vec) is
    begin
  end procedure;
  begin
    check_value_long((0 => (unsigned(v_slv_long_min), unsigned(v_slv_long_max))));
    wait;
  end process;

end architecture beh;
```
```
*** Caught exception c0000005 (EXCEPTION_ACCESS_VIOLATION) [address=0000000000000000, ip=00007FF6C9E72040] ***

[00007FF6C9E46C0F]
[00007FF6C9E46D9D]
[00007FF902E70327] UnhandledExceptionFilter+0x1e7
[00007FF90283F0B8]
[00007FF90283EEA1]
[00007FF9055F5530] memset+0x13b0
[00007FF9055DC876] _C_specific_handler+0x96
[00007FF9055F241F] _chkstk+0x11f
[00007FF9055A14A4] RtlRaiseException+0x434
[00007FF9055F0F4E] KiUserExceptionDispatcher+0x2e
[00007FF6C9E72040]
[00007FF6C9E9B78F]
[00007FF6C9E9F87C]
[00007FF6C9EA7678]
[00007FF6C9E9D2ED]
[00007FF6C9E9F664]
[00007FF6C9EA8F87]
[00007FF6C9EA9340]
[00007FF6C9EAA639]
[00007FF6C9EB0BA7]
[00007FF6C9EB12FB]
[00007FF6C9E42CC0]
[00007FF6C9F33E7A] vhpi_put_data+0x2411a
[00007FF6C9E412EE]
```

nickg, 2023-01-16:
The problem is actually determining the bounds of the of the array aggregate (one of the bounds comes from the result of the concatenation). Your example works now but the VHDL-2008 unconstrained array changes have proven difficult to get right so there are likely other bugs. Could you test again?

avelure, 2023-01-16:
Works now, thanks!