nickg/nvc #592: result of -2147483648 - 1 cannot be represented as INTEGER

avelure, 2023-01-16:
I think this should be ok, at least the other tools don't complain.
If I replace v_len with a literal value I don't get an error.
```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
end entity test;
architecture beh of test is
begin
  
  process
    variable v_len : natural := 31;
  begin
    if 0 > 2 ** v_len - 1 then
      report "error" severity error;
    end if;
    wait;
  end process;

end architecture beh;
```
```
** Fatal: 0ms+0: result of -2147483648 - 1 cannot be represented as INTEGER
    > test_intmax.vhd:16
    |
 13 |     if 0 > 2 ** v_len - 1 then
    |            ^^^^^^^^^^^^^^
```

nickg, 2023-01-16:
I think it's actually correct to report an error here (see #101), but most tools don't strictly implement the overflow check. Although really the error should be reported on the exponentiation not the subtraction (i.e. "result of 2 ** 31 cannot be represented as INTEGER").

nickg, 2023-01-16:
> But I thought it got converted to universal integer, which I understand has a implementation defined size. 

That's an interesting point. I was about to reply and say it's not correct as `integer` doesn't implicitly convert to `universal_integer` but actually `**` is a special case as the relevant overloads are:

```
"**" [universal_integer, INTEGER return universal_integer]
"**" [INTEGER, INTEGER return INTEGER]
```

(I.e. unlike all the other operators there is no `"**" [universal_integer, universal_integer return universal_integer]"` - I missed this when I copied the predefined operators from the LRM.) So the correct interpretation is the first overload which returns `universal_integer`, currently a 64-bit type.

But this bug is particular to `**`: I believe the the behaviour of other operators wrt. overflow is correct.

> As this is not an assignment it doesn't seem like it is necessary to restrict it to integer size. 

I think the LRM is pretty clear that it's the result of the _operations_ that are checked rather than the final result before assignment (which would be covered by the regular subtype check anyway).

nickg, 2023-01-20:
Your original issue should be fixed now, but there's another bug where the result of `**` on integers is not checked for overflow. I'll leave this ticket open as a reminder.

nickg, 2023-02-16:
The `**` operator now checks for overflow too.

closed 2023-02-16 by nickg