nickg/nvc #593: result of 1545611453 - -2147483648 cannot be represented as INTEGER

avelure, 2023-01-16:
This one is a bit more iffy, I'm not sure what the original designer was trying to do, but I don't get errors in the other tools.
```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
end entity test;
architecture beh of test is
begin
  
  process
    constant C_MIN_RANGE      : integer     := integer'left;
    variable v_ret            : integer := 1545611453;
  begin
    v_ret := v_ret - C_MIN_RANGE;
    wait;
  end process;

end architecture beh;
```
```
** Fatal: 0ms+0: result of 1545611453 - -2147483648 cannot be represented as INTEGER
    > test_intmin.vhd:14
    |
 14 |     v_ret := v_ret - C_MIN_RANGE;
    |              ^^^^^^^^^^^^^^^^^^^
```

nickg, 2023-01-16:
The LRM seems pretty clear on this:

> The same arithmetic operators are predefined for all integer types (see 9.2). It is an error if the execution of such an operation (in particular, an implicit conversion) cannot deliver the correct result (that is, if the value corresponding to the mathematical result is not a value of the integer type).

(Section 5.2.3.1 of the 2008 edition.)

The `integer` type is a signed 32-bit number (until VHDL-2019 anyway) and 3693095101 can't be represented in that.

avelure, 2023-01-17:
Yes, I agree. It is relying on some 32b integer rollover which is not really future proof. I'll try to get the code changed.

closed 2023-01-18 by nickg