nickg/nvc #596: Non sl/slv signals don't get reset in wave output

avelure, 2023-01-19:
I was looking at the fst wave output of some simulations and it looks like non-bit type signals like boolean, integer or custom types don't get reset like sl/slv signals do.
I thought it might be some optimization, but setting it to O0 did not help.
The simulation seems to work fine, so I guess it is just an issue with writing to the fst.

If you generate wave output of this code and open it in gtkwave you can see that all sig_xxx signals are undefined first, then only the sig_sl gets reset when arst_n goes low, the others only get updated in the RUN state.

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
end entity test;
architecture beh of test is
  signal arst_n : std_logic;
  signal clk : std_logic;
  signal sig_sl : std_logic;
  signal sig_b : boolean;
  signal sig_i : natural range 0 to 10;
  type t_states is (IDLE, RUN, STOP);
  signal sig_s : t_states;
  signal clk_stop : std_logic := '0';
begin

  arst_n <= '1', '0' after 1 ns, '1' after 10 ns;

  process
  begin
    clk <= '1';
    wait for 2 ns;
    clk <= '0';
    wait for 2 ns;
    if clk_stop = '1' then
      wait;
    end if;
  end process;

  process(arst_n, clk)
  begin
    if arst_n = '0' then
      sig_sl <= '0';
      sig_b <= false;
      sig_i <= 0;
      sig_s <= IDLE;
    elsif rising_edge(clk) then
      case sig_s is
        when IDLE =>
          sig_s <= RUN;
        when RUN =>
          sig_sl <= '1';
          sig_b <= true;
          sig_i <= 1;
          sig_s <= STOP;
        when STOP =>
          clk_stop <= '1';
      end case;
    end if;
  end process;

end architecture beh;
```

nickg, 2023-01-22:
Isn't that because the reset values are actually the same as the initial values of those signals? The waveform dumping code only runs where there is an event on a particular signal, but since the assignment to e.g. `sig_s` in the reset block doesn't cause `sig_s'event` to go high, there is no event generated in the FST file. If you change `t_states` to have an explicit undefined state as `t_states'left`, then you'll see it get reset:

```
  type t_states is (UNDEF, IDLE, RUN, STOP);
```

avelure, 2023-01-22:
[image]
So the wafeform writing code is only triggered on sig_s'event then? It would explain it.

In the waveform it is already showing an implicit undefined state from start of simulation.
Wouldn't it be better to set these to the initial value from the start of the waveform display? I see that's what ghdl and modelsim does. These implicitly undefined signals makes it very difficult to look for signals that actually are undefined.

It would of course be more useful when looking for non sl/slv signals that have missing resets if these signals had implicitly undefined states, but that they somehow got defined on reset, but I understand this would be complicated to implement and not really according to standard.

nickg, 2023-01-22:
OK, then I misunderstood the original question. For me the output looks like this:

[image]

That's with GtkWave v3.3.104. Could you upload the .fst file here?

avelure, 2023-01-22:
I was using GTKWave v3.4.0, but I checked with v3.3.100 now as well
Remove the .txt ending from the file name.
[test.fst.txt](https://github.com/nickg/nvc/files/10475467/test.fst.txt)

Generated with
```
$ nvc --std=08 -a test_wave.vhd
$ nvc -e test -r -w
** Note: writing FST waveform data to test.fst
```

nickg, 2023-01-22:
OK that's weird: I can reproduce this but only on Windows.

nickg, 2023-01-23:
Finally found the bug in the FST writer code we import from GtkWave: on Windows it emulates the `mmap` Unix function using file I/O but it was forgetting to flush the modified buffer to disk in one place. I put in a hack to fix that but the proper way to do this on Windows is to use the native `MapViewOfFileEx` API. I'll try doing that later and submit a PR to GtkWave.

This should have been caught by the regular regression tests on Windows but unfortunately we don't run the wave dumper tests there at the moment as they're all written in shell script. I should fix that too.

avelure, 2023-01-23:
Great, thanks for the quick fix!

nickg, 2023-01-23:
I rewrote the FST writer memory mapping code to use `MapViewOfFileEx`. Seems to work OK on my Windows VM. Would you mind testing it for a bit? If it's OK I'll submit it to GtkWave.

closed 2023-01-23 by nickg

avelure, 2023-01-31:
It seems to work OK for me on native windows 10.