nickg/nvc #597: Signal lost through port records

avelure, 2023-01-19:
I'm not sure the title is correct, but in the wave output it seems like most signals propagate as X to the FIFO.

```vhdl
library IEEE;
USE IEEE.STD_LOGIC_1164.ALL;
USE IEEE.NUMERIC_STD.ALL;

entity avalon_st_fifo is
  generic(
    GC_DATA_WIDTH    : natural := 8;
    GC_CHANNEL_WIDTH : natural := 1;
    GC_EMPTY_WIDTH   : natural := 1;
    GC_ERROR_WIDTH   : natural := 1;
    GC_FIFO_DEPTH    : natural := 256
  );
  port(
    clk_i            : in  std_logic;
    reset_i          : in  std_logic;
    slave_data_i     : in  std_logic_vector(GC_DATA_WIDTH - 1 downto 0);
    slave_channel_i  : in  std_logic_vector(GC_CHANNEL_WIDTH - 1 downto 0);
    slave_empty_i    : in  std_logic_vector(GC_EMPTY_WIDTH - 1 downto 0);
    slave_error_i    : in  std_logic_vector(GC_ERROR_WIDTH - 1 downto 0);
    slave_valid_i    : in  std_logic;
    slave_sop_i      : in  std_logic;
    slave_eop_i      : in  std_logic;
    slave_ready_o    : out std_logic := '1';
    master_data_o    : out std_logic_vector(GC_DATA_WIDTH - 1 downto 0);
    master_channel_o : out std_logic_vector(GC_CHANNEL_WIDTH - 1 downto 0);
    master_empty_o   : out std_logic_vector(GC_EMPTY_WIDTH - 1 downto 0);
    master_error_o   : out std_logic_vector(GC_ERROR_WIDTH - 1 downto 0);
    master_valid_o   : out std_logic := '0';
    master_sop_o     : out std_logic;
    master_eop_o     : out std_logic;
    master_ready_i   : in  std_logic;
    empty_o          : out std_logic := '1'
  );
end avalon_st_fifo;

architecture Behavioral of avalon_st_fifo is

begin
  p_main : process(clk_i)
    type t_data_mem is array (0 to GC_FIFO_DEPTH - 1) of std_logic_vector(GC_DATA_WIDTH - 1 downto 0);
    variable v_data_mem : t_data_mem := (others => (others => '0'));

    type t_channel_mem is array (0 to GC_FIFO_DEPTH - 1) of std_logic_vector(GC_CHANNEL_WIDTH - 1 downto 0);
    variable v_channel_mem : t_channel_mem := (others => (others => '0'));

    type t_empty_mem is array (0 to GC_FIFO_DEPTH - 1) of std_logic_vector(GC_EMPTY_WIDTH - 1 downto 0);
    variable v_empty_mem : t_empty_mem := (others => (others => '0'));

    type t_error_mem is array (0 to GC_FIFO_DEPTH - 1) of std_logic_vector(GC_ERROR_WIDTH - 1 downto 0);
    variable v_error_mem : t_error_mem := (others => (others => '0'));

    type t_sop_mem is array (0 to GC_FIFO_DEPTH - 1) of std_logic;
    variable v_sop_mem : t_sop_mem := (others => '0');

    type t_eop_mem is array (0 to GC_FIFO_DEPTH - 1) of std_logic;
    variable v_eop_mem : t_eop_mem := (others => '0');

    variable v_WrPtr  : natural range 0 to GC_FIFO_DEPTH - 1 := 0;
    variable v_RdPtr  : natural range 0 to GC_FIFO_DEPTH - 1 := 0;
    variable v_looped : boolean                              := false;
  begin
    if rising_edge(clk_i) then
      if reset_i = '1' then
        v_WrPtr        := 0;
        v_RdPtr        := 0;
        v_looped       := false;
        master_valid_o <= '0';
[... 246 lines cut ...]
UVVM: ID_GEN_PULSE                     100.0 ns  TB seq.                        Pulsed to 1 for 100000000 fs. 'Pulsing reset for 10 clock periods'
UVVM: ID_PACKET_INITIATE               205.0 ns  TB seq.                        avalon_st_transmit(9 words/9 sym, ch:0)=>
UVVM: ID_PACKET_DATA                   205.0 ns  TB seq.                        avalon_st_transmit(9 words/9 sym, ch:0)=> x"47" (symbol# 0).
UVVM: ID_PACKET_DATA                   212.5 ns  TB seq.                        avalon_st_transmit(9 words/9 sym, ch:0)=> x"14" (symbol# 1).
UVVM: ID_PACKET_DATA                   222.5 ns  TB seq.                        avalon_st_transmit(9 words/9 sym, ch:0)=> x"3B" (symbol# 2).
UVVM: ID_PACKET_DATA                   232.5 ns  TB seq.                        avalon_st_transmit(9 words/9 sym, ch:0)=> x"25" (symbol# 3).
UVVM: ID_PACKET_DATA                   242.5 ns  TB seq.                        avalon_st_transmit(9 words/9 sym, ch:0)=> x"93" (symbol# 4).
UVVM: ID_PACKET_DATA                   252.5 ns  TB seq.                        avalon_st_transmit(9 words/9 sym, ch:0)=> x"22" (symbol# 5).
UVVM: ID_PACKET_DATA                   262.5 ns  TB seq.                        avalon_st_transmit(9 words/9 sym, ch:0)=> x"E3" (symbol# 6).
UVVM: ID_PACKET_DATA                   272.5 ns  TB seq.                        avalon_st_transmit(9 words/9 sym, ch:0)=> x"70" (symbol# 7).
UVVM: ID_PACKET_DATA                   282.5 ns  TB seq.                        avalon_st_transmit(9 words/9 sym, ch:0)=> x"99" (symbol# 8).
UVVM: ID_PACKET_COMPLETE               292.5 ns  TB seq.                        avalon_st_transmit(9 words/9 sym, ch:0) DONE.
UVVM: ID_PACKET_INITIATE               295.0 ns  TB seq.                        avalon_st_expect(9 words/9 sym, ch:0) while executing avalon_st_receive=>
UVVM: ID_PACKET_DATA                   300.0 ns  TB seq.                        avalon_st_expect(9 words/9 sym, ch:0) while executing avalon_st_receive=> x"47" (symbol# 0).
UVVM: ID_PACKET_DATA                   310.0 ns  TB seq.                        avalon_st_expect(9 words/9 sym, ch:0) while executing avalon_st_receive=> x"14" (symbol# 1).
UVVM: ID_PACKET_DATA                   320.0 ns  TB seq.                        avalon_st_expect(9 words/9 sym, ch:0) while executing avalon_st_receive=> x"3B" (symbol# 2).
UVVM: ID_PACKET_DATA                   330.0 ns  TB seq.                        avalon_st_expect(9 words/9 sym, ch:0) while executing avalon_st_receive=> x"25" (symbol# 3).
UVVM: ID_PACKET_DATA                   340.0 ns  TB seq.                        avalon_st_expect(9 words/9 sym, ch:0) while executing avalon_st_receive=> x"93" (symbol# 4).
UVVM: ID_PACKET_DATA                   350.0 ns  TB seq.                        avalon_st_expect(9 words/9 sym, ch:0) while executing avalon_st_receive=> x"22" (symbol# 5).
UVVM: ID_PACKET_DATA                   360.0 ns  TB seq.                        avalon_st_expect(9 words/9 sym, ch:0) while executing avalon_st_receive=> x"E3" (symbol# 6).
UVVM: ID_PACKET_DATA                   370.0 ns  TB seq.                        avalon_st_expect(9 words/9 sym, ch:0) while executing avalon_st_receive=> x"70" (symbol# 7).
UVVM: ID_PACKET_DATA                   380.0 ns  TB seq.                        avalon_st_expect(9 words/9 sym, ch:0) while executing avalon_st_receive=> x"99" (symbol# 8).
UVVM: ID_BFM                           382.5 ns  TB seq.                        avalon_st_expect(9 words/9 sym, ch:0)=> OK, received 9 symbols.
UVVM: ID_CLOCK_GEN                     390.0 ns  TB seq.                        Stopping clock Avalon-ST CLK
```

nickg, 2023-01-22:
I believe NVC's behaviour is correct here. At the top level the signals in `avalon_st_master_if` are driven both by the process at the end of the file that calls `avalon_st_transmit`, and by the inout ports in `test_harness`. Note that in VHDL an out/inout port or an procedure call with out/inout signal arguments drives every sub-element of the signal **even if there is no assignment to that signal within the architecture/procedure**. For the master signals, because `test_harness` is always driving `U`, the resolved value is also always `U`. You can fix it by adding something like this in `test_harness`:

```vhdl
  avalon_st_master_if.data <= (others => 'Z');
  avalon_st_master_if.channel <= (others => 'Z');
  avalon_st_master_if.empty <= (others => 'Z');
  avalon_st_master_if.data_error <= (others => 'Z');
  avalon_st_master_if.valid <= 'Z';
  avalon_st_master_if.start_of_packet <= 'Z';
  avalon_st_master_if.end_of_packet <= 'Z';

  avalon_st_slave_if.ready <= 'Z';
```

Or by giving the two inout ports an initial value of all-Z in the entity declaration.

Were you testing with GHDL before? GHDL doesn't implement this correctly, there's an open bug for it: ghdl/ghdl#1842

avelure, 2023-01-22:
Yes, I was testing with ghdl and modelsim where the test passes.
I guess I understand the issue, but it is not very intuitive to the user when the problem occurs.
Adding the code you proposed or setting the default value does not make it work, I'll try to litter some more Zs around to see if it works in the end.

nickg, 2023-01-24:
I've added a warning at the start of the simulation if this situation is detected:

```
** Warning: (init): signal AVALON_ST_MASTER_IF.END_OF_PACKET has 2 sources including port
            AVALON_ST_MASTER_IF.END_OF_PACKET which has initial value 'U' and no driver in
            instance I_AVALON_ST_TEST_HARNESS
     > test.vhd:238
     |
 142 |     signal avalon_st_master_if : inout t_avalon_st_if(channel(GC_CHANNEL_WIDTH - 1 downto 0),
     |            ^^^^^^^^^^^^^^^^^^^
     |            sourced by port AVALON_ST_MASTER_IF.END_OF_PACKET which always contributes 'U'
 ...
 238 |   signal avalon_st_master_if : t_avalon_st_if(channel(GC_CHANNEL_WIDTH - 1 downto 0),
     |          ^^^^^^^^^^^^^^^^^^^ signal AVALON_ST_MASTER_IF.END_OF_PACKET declared here
     |
     = Note: the resolved value will always be 'U' which was almost certainly not intended
```

Does this help? You'll need to re-analyse UVVM as well as your own code.

avelure, 2023-01-31:
This is great!
I see I had the wrong library for i_avalon_st_test_harness in the MWE, so adding the Zs do work, but now nvc is giving some other errors I have to get to the bottom of 😛

nickg, 2023-01-31:
OK, great. I'll close this one now...

closed 2023-01-31 by nickg