nickg/nvc #599: nvm 1.8.0 osx build failure

chenrui333, 2023-01-22:
👋 trying to build the latest release, but run into some build issue. The error log is as below:

<details>
<summary>build issue</summary>

```
	mv -f $depbase.Tpo $depbase.Po
rm -f lib/libcpustate.a
ar cr lib/libcpustate.a thirdparty/cpustate.o
ranlib lib/libcpustate.a
depbase=`echo thirdparty/memmem.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
	clang -DHAVE_CONFIG_H -I. -I..  -I../thirdparty -I../src  -Wall   -pthread -D_THREAD_SAFE -I/opt/homebrew/Cellar/check/0.15.2/include  -I/Library/Developer/CommandLineTools/SDKs/MacOSX13.sdk/usr/include/ffi -g -O2 -MT thirdparty/memmem.o -MD -MP -MF $depbase.Tpo -c -o thirdparty/memmem.o ../thirdparty/memmem.c &&\
	mv -f $depbase.Tpo $depbase.Po
../thirdparty/memmem.c:37:10: fatal error: 'str-two-way.h' file not found
#include "str-two-way.h"
         ^~~~~~~~~~~~~~~
```

</details>

full build log, https://github.com/Homebrew/homebrew-core/actions/runs/3980794777/jobs/6824127333
relates to Homebrew/homebrew-core#121245

nickg, 2023-01-23:
Could you try again with the latest 1.8.1 release? I tested building from the .tar.gz on macOS and it built successfully.

chenrui333, 2023-01-24:
sounds good Thanks!

chenrui333, 2023-01-24:
Works for me in my local as well (ventura arm), closing the issue. Thanks again!

closed 2023-01-24 by chenrui333