nickg/nvc #603: Changes to OSVVM development code giving a fatal error

wyvernSemi, 2023-01-30:
Changes to OSVVM code have cause an error as shown below:

```
** Fatal: tree kind T_ARRAY_REF does not have item I_REF
    > CoSimInterruptHandler.vhd:67
    |
 67 |       '0'  when gVProcReadInterrupts'event and gIntReq(i)'delayed = '1' and EDGE_LEVEL(i) = INTERRUPT_ON_EDGE else
    |                                                ^^^^^^^^^^
[0x55994da66add] ../src/object.c:210 object_lookup_failed
[0x55994da67104] ../src/common.c:894 tree_ref
[0x55994da67104] ../src/tree.c:892 tree_ref.cold
[0x55994daeb09c] ../src/diag.c:379 simp_attr_ref
[0x55994daeb09c] ../src/simp.c:1632 simp_tree.lto_priv.0
[0x55994dad68b9] ../src/object.c:652 object_rewrite_iter
[0x55994dad679d] ../src/object.c:685 object_rewrite
[0x55994dad67f0] ../src/object.c:716 object_rewrite
[0x55994dad6688] ../src/object.c:727 object_rewrite
[0x55994dad67f0] ../src/object.c:716 object_rewrite
[0x55994dad6688] ../src/object.c:727 object_rewrite
[0x55994dad67f0] ../src/object.c:716 object_rewrite
[0x55994dad6688] ../src/object.c:727 object_rewrite
[0x55994dad67f0] ../src/object.c:716 object_rewrite
[0x55994dad6688] ../src/object.c:727 object_rewrite
[0x55994dad6688] ../src/object.c:727 object_rewrite
[0x55994dad6688] ../src/object.c:727 object_rewrite
[0x55994dad6688] ../src/object.c:727 object_rewrite
[0x55994dafb76a] ../src/tree.c:1200 tree_rewrite
[0x55994db26398] ../src/simp.c:1771 analyse
[0x55994db22e4c] ../src/nvc.c:1223 process_command
[0x55994da68203] ../src/nvc.c:1344 main

```
This is on an indexed global signal with a 'delayed reference. I have bundled a minimum set of files and a compile.sh script to reproduce the error. I am using version nvc 1.8-devel (1.7.0.r317.g5e43ce49).

[minFileSet.tar.gz](https://github.com/nickg/nvc/files/10537107/minFileSet.tar.gz)

nickg, 2023-01-30:
The specific issue above with array names as the prefix of `'delayed` is fixed now and your `compile.sh` runs without error. Could you test the full case again? Either with the latest master or the v1.8-branch branch.

wyvernSemi, 2023-01-31:
I built from the latest master (nvc 1.9-devel (1.8.0.r36.gb534a5f5)) and the compile issue goes away for the OSVVM build but I now get what seems to be a related error at run-time:
```
nvc --std=08 -H 128m --work=osvvm_CoSim_TbAxi4:/home/simon/OSVVM/nvc/VHDL_LIBS/nvc-1.9-devel/OSVVM_COSIM_TBAXI4 -L /home/simon/OSVVM/nvc/VHDL_LIBS/nvc-1.9-devel -e TbAb_CoSim

Name       OSVVM_COSIM_TBAXI4.TBADDRESSBUSMEMORY.COSIMINTERRUPTHANDLER_1.GINTREQ$delayed_0_p
Kind       process
Context    OSVVM_COSIM_TBAXI4.TBADDRESSBUSMEMORY.COSIMINTERRUPTHANDLER_1
Blocks     2
Registers  7
Types      7
Variables  0
Begin
   0: r0 := var upref 1, GINTREQ$delayed_0 // @<$<0..8>> => 0..8
      r1 := load indirect r0            // $<0..8> => 0..8
      r2 := const 1                     // # => 1
      drive signal r1 count r2
      r3 := link package OSVVM_COMMON.INTERRUPTGLOBALSIGNALPKG // P<OSVVM_COMMON.INTERRUPTGLOBALSIGNALPKG>
      r4 := link var r3 GINTREQ         // @<$<0..8>> => $<0..8>
      r5 := load indirect r4            // $<0..8>
      r6 := const 0                     // # => 0
   1: Empty basic block

** Fatal: missing variable for generic I in OSVVM_COSIM.COSIMINTERRUPTHANDLER-BEHAVIORAL
[0x496c29] ../src/diag.c:890 diag_femit.part.0
[0x4102ec] ../src/util.c:548 fatal_trace
[0x4678dd] ../src/lower.c:2873 lower_ref
[0x4678dd] ../src/lower.c:4511 lower_expr
[0x4678f1] ../src/lower.c:4547 lower_rvalue
[0x465a31] ../src/lower.c:3139 lower_array_ref
[0x465a31] ../src/lower.c:4517 lower_expr
[0x471eb3] ../src/lower.c:4542 lower_lvalue
[0x471eb3] ../src/lower.c:4930 lower_sched_event
[0x478ddf] ../src/lower.c:9383 lower_concurrent_block
[0x478d63] ../src/lower.c:10122 lower_concurrent_block
[0x479418] ../src/lower.c:10150 lower_elab
[0x479418] ../src/lower.c:10204 lower_unit
[0x40d281] ../src/nvc.c:367 elaborate
[0x40d281] ../src/nvc.c:1381 process_command
[0x40a76c] ../src/nvc.c:1506 main
```
I also tried with the v1.8-branch and get the same result. Some of my tests pass, but only those, it would seem, that don't instantiate the CoSimInterruptHandler component that has gIntReq'delayed.  I will try and construct a simpler case to reproduce the error.

wyvernSemi, 2023-02-01:
I have reproduced the above issue with a smaller test case, running with nvc 1.8.1 (1.8.1.r9.gc2e5ca47) 
[testcase.tar.gz](https://github.com/nickg/nvc/files/10555181/testcase.tar.gz)

nickg, 2023-02-05:
This is fixed now along with some other bugs I found with `'delayed`. Could you test again?

wyvernSemi, 2023-02-06:
I have tried the latest branch of nvc and the code now gets as far as running, so I believe this issue is fixed. However, I am getting new failures at run-time unrelated to the gIntReq'delayed but crashing seemingly around updating foreign procedure outputs, which does not happen on v1.8-branch (for tests that don't use the gIntReq'delayed signal, so run on that version). I will raise a new issue for this problem when I have more useful information.

closed 2023-02-07 by nickg