nickg/nvc #604: VHDL-2019: Conditional Analysis `elsif Fails

bpadalino, 2023-01-30:
When performing conditional analysis in VHDL-2019, the `elsif fails.

```
use std.env.all ;
use std.textio.all ;

entity test is
end entity ;

architecture arch of test is

`if TOOL_NAME = "nvc" then
    constant CHECK1 : string := TOOL_NAME ;
`else
    constant CHECK1 : string := "UNDEFINED" ;
`end

`if TOOL_NAME = "Aldec" then
    constant CHECK2 : string := TOOL_NAME ;
`elsif TOOL_NAME = "nvc" then
    constant CHECK2 : string := TOOL_NAME ;
`else
    constant CHECK2 : string := "UNDEFINED" ;
`end

begin

    tb : process
        variable l : line ;
    begin
        write(l, "CHECK1: " & CHECK1) ;
        writeline(output, l) ;
        write(l, "CHECK2: " & CHECK2) ;
        writeline(output, l) ;
        std.env.stop ;
    end process ;

end architecture ;
```

... produces ...

```
CHECK1: nvc
CHECK2: UNDEFINED
** Note: 0ms+0: STOP called
```

... when they should obviously be the same.

bpadalino, 2023-02-01:
It looks like `tCONDELSIF` is never considered in the `case` statement in [`parse.c`](https://github.com/nickg/nvc/blob/661bdbaa360c6277caa2b8b4e6ec21c344793c9c/src/parse.c#L250).  I'm unsure what the `cond_state` is trying to keep track of, or if things are nested how that works out - but I think the issue, mostly, is that it just isn't parsed.

nickg, 2023-02-02:
Yes it simply wasn't implemented before. I've refactored the code somewhat to fix this and some other issues. It's probably not perfect but the example above works at least.

bpadalino, 2023-02-02:
Still problematic.  Seems like nested ones aren't properly handled.

```
use std.env.all ;
use std.textio.all ;

entity test is
end entity ;

architecture arch of test is

`if TOOL_TYPE = "SYNTHESIS" then
    `if TOOL_NAME = "Vivado" then
        constant CHECK : string := TOOL_NAME ;
    `else
        constant CHECK : string := "UNDEFINED_SYNTH" ;
    `end
`else
    `if TOOL_NAME = "Riviera-PRO" then
        constant CHECK : string := TOOL_NAME ;
    `elsif TOOL_NAME = "nvc" then
        constant CHECK : string := TOOL_NAME ;
    `else
        constant CHECK : string := "UNDEFINED_SIM" ;
    `end
`end

begin

    tb : process
        variable l : line ;
    begin
        write(l, "CHECK: " & CHECK) ;
        writeline(output, l) ;
        std.env.stop ;
    end process ;

end architecture ;
```

... produces:

```
** Error: CHECK already declared in this region
    > test.vhd:19
    |
 13 |         constant CHECK : string := "UNDEFINED_SYNTH" ;
    |                  ^^^^^ previous declaration was here
 ...
 19 |         constant CHECK : string := TOOL_NAME ;
    |                  ^^^^^ duplicate declaration
```

nickg, 2023-02-03:
OK nested conditionals should work now too...

closed 2023-02-03 by nickg