nickg/nvc #606: fatal: timeout waiting for TB.P_MAIN

avelure, 2023-01-30:
I got two tests failing when running about 10 tests in parallel with messages like in the title.
Doesn't seem to be easily reproducible. Running the tests standalone does not give an error.
Is it possibly timeouts due to congestion on the CPU?
```
[00007FF770A8719F]
[00007FF770B0F4B5]
[00007FF770A85FE2]
[00007FF770B2284B] nvc_current_delta+0x923b
[00007FF770B59D08] vhpi_put_data+0x8868
[00007FF770B53CAE] vhpi_put_data+0x280e
[00007FF770B11422]
[00007FF770B11721]
[00007FF770B11D91]
[00007FF770B11BF0]
[00007FFBA80E4D0B] pthread_create_wrapper+0x9b
[00007FFBDE301BB2] configthreadlocale+0x92
[00007FFBE0827614] BaseThreadInitThunk+0x14
[00007FFBE09626A1] RtlUserThreadStart+0x21
```
```
[00007FF770A8719F]
[00007FF770B0F4B5]
[00007FF770A85FE2]
[00007FF770B2284B] nvc_current_delta+0x923b
[00007FF770B27948] nvc_current_delta+0xe338
[00007FF770B2F4CD] nvc_current_delta+0x15ebd
[00007FF770B30145] nvc_current_delta+0x16b35
[00007FF770B59D08] vhpi_put_data+0x8868
[00007FF770B53CAE] vhpi_put_data+0x280e
[00007FF770B11422]
[00007FF770B11721]
[00007FF770B11D91]
[00007FF770B11BF0]
[00007FFBBCAE4D0B] pthread_create_wrapper+0x9b
[00007FFBDE301BB2] configthreadlocale+0x92
[00007FFBE0827614] BaseThreadInitThunk+0x14
[00007FFBE09626A1] RtlUserThreadStart+0x21
```

nickg, 2023-01-31:
It's probably because the system was too busy. I've increased the time out from 1 to 10 seconds, can you see if that fixes it?

avelure, 2023-01-31:
Thanks, looks to be ok now, I'll monitor it for a bit as it is not consistently reproducible.

JimLewis, 2023-02-19:
@nickg   I am periodically getting the same thing when running OSVVM regressions.   It is quite random as in a different simulation run a different failure will happen.   If it matters the report is:   
```
nvc --std=08 -H 128m --work=osvvm_TbAxiStream:C:/tools/sim_temp/nvc-1.9-devel/OSVVM_TBAXISTREAM.08 -L C:/tools/sim_temp/nvc-1.9-devel -e TbStream_ReleaseAcquireReceiver1
** Fatal: timeout waiting for OSVVM.ALERTLOGPKG
[00007ff7d08f261f] 
[00007ff7d09923db] nvc_current_delta+0x54b
[00007ff7d08f21d1] 
[00007ff7d099b8f1] nvc_current_delta+0x9a61
[00007ff7d09a45ad] nvc_current_delta+0x1271d
[00007ff7d09aa5cf] nvc_current_delta+0x1873f
[00007ff7d09b173d] _nvc_get_object+0x25ad
[00007ff7d09c55f9] vhpi_put_value+0xf09
[00007ff7d09c8731] vhpi_put_value+0x4041
[00007ff7d098d1f2] 
[00007ff7d098d3c2] 
[00007ff7d098f8e1] 
[00007ff7d098ec30] 
[00007ffcc1e64d0b] pthread_create_wrapper+0x9b
[00007ffcd0cfaf5a] beginthreadex+0x12a
[00007ffcd0cfb02c] endthreadex+0xac
[00007ffcd1b87614] BaseThreadInitThunk+0x14
[00007ffcd1d226a1] RtlUserThreadStart+0x21
```

JimLewis, 2023-02-19:
Any chance there is a setting to increase the timeout so this does not happen?    When I run regressions, I have been running multiple simulators on the same computer.

nickg, 2023-02-19:
@JimLewis can you post the full output from `nvc --version` which includes the git commit hash? I increased the timeout a few weeks ago from one second to 10 but I think the real cause was the slow compilation that was fixed in #608. Can I reproduce this by running the regular OSVVM test scripts?

JimLewis, 2023-02-19:
@nickg  
nvc 1.9-devel (1.8.0.r69.gc35e8b50) (Using LLVM 15.0.7)
Copyright (C) 2011-2023  Nick Gasson

I need something that has the fixes for:   #618, #612, #603, #581, #577

What I am running is on the OSVVM dev branch.   It is our normal regression run which I run it by doing the following in a tclsh that has already loaded the OSVVM scripting environment.   I am running on MSYS2 on top of Windows.
```
build <path to OsvvmLibraries>/OsvvmLibraries/OsvvmLibraries.pro
build <path to OsvvmLibraries>/OsvvmLibraries/RunAllTests.pro
```

JimLewis, 2023-02-20:
@nickg 
I just tried the one you provided a link for in the other branch:
nvc 1.9-devel (1.8.0.r101.g6ff64d1d) (Using LLVM 15.0.7)

It fails here.
--------------------------
I tried the release from GitHub:
mingw-w64-x86_64-nvc-1.8.2.r1.ga6d2c8fc-1-any.pkg.tar.zst

It passes the main sims, but can not run the CoSim simulations.   
--------------------------

So I guess I am waiting for a release that has both.

nickg, 2023-02-20:
OK I changed it to print a warning after 10 seconds rather than abort.

JimLewis, 2023-02-20:
Was the timeout issue supposed to be fixed in nvc 1.9-devel (1.8.0.r101.g6ff64d1d) (Using LLVM 15.0.7)?

It did not timeout in release mingw-w64-x86_64-nvc-1.8.2.r1.ga6d2c8fc-1-any.pkg.tar.zst, however, that release does not support the co-simulation stuff.

nickg, 2023-02-20:
No, I just increased the timeout but I suppose it could still trigger if the system is very overloaded.

nickg, 2023-02-26:
@JimLewis can you try the latest master with OSVVM scripts modified like this?

```diff
diff --git a/VendorScripts_NVC.tcl b/VendorScripts_NVC.tcl
index 52af5ab2c780..d031b501db41 100644
--- a/VendorScripts_NVC.tcl
+++ b/VendorScripts_NVC.tcl
@@ -205,13 +205,8 @@ proc vendor_simulate {LibraryName LibraryUnit args} {
   set GlobalOptions ${LocalGlobalOptions}
   set ElaborateOptions [concat {*}${LocalElaborateOptions} ${LibraryUnit}]
   set RunOptions [concat {*}${LocalRunOptions} ${LibraryUnit}]
-  puts "nvc ${GlobalOptions} -e ${ElaborateOptions}" 
-  if { [catch {exec nvc {*}${GlobalOptions} -e {*}${ElaborateOptions}} ElaborateErrorMessage]} { 
-    puts $ElaborateErrorMessage
-    error "Failed: simulate $LibraryUnit"
-  }
-  puts "nvc ${GlobalOptions} -r ${RunOptions}" 
-  if { [catch {exec nvc {*}${GlobalOptions} -r {*}${RunOptions} >@ stdout 2>@ stdout}]} {
+  puts "nvc ${GlobalOptions} -e ${ElaborateOptions} --jit --no-save -r ${RunOptions}"
+  if { [catch {exec nvc {*}${GlobalOptions} -e {*}${ElaborateOptions} --jit --no-save -r {*}${RunOptions} >@ stdout 2>@ stdout}] } {
     error "Failed: simulate $LibraryUnit"
   }
   
```

So the idea is to elaborate and run in a single command, and add `--jit --no-save`. See #634 for more details.

On my system timing `test/test-osvvm.tcl` before:

```
real    8m27.854s
user    28m54.337s
sys     3m3.749s
```

And after:

```
real    3m43.801s
user    7m50.857s
sys     1m31.736s
```

So the total CPU usage reduced ~3.5x, which should help alleviate the overloading you're seeing here.

@avelure I'm interested how well this works for your simulations too.

JimLewis, 2023-02-26:
@nickg That is interesting.   I will give it a try.

nickg, 2023-04-23:
This should not occur any more as the timeout has been increased dramatically and is now a warning instead of an error.

closed 2023-04-23 by nickg