nickg/nvc #607: User defined time resolution

avelure, 2023-01-31:
I have some vague notion that it was once possible to set a user defined time resolution from the command line.
Was the option removed or was it never present?

nickg, 2023-01-31:
I'm pretty sure the resolution limit has always been `fs`, the smallest VHDL time unit. Is this a feature you often use in other simulators? I always thought it was more of a Verilog thing.

avelure, 2023-01-31:
No, not really, I think it has just been common to use 1ps on modelsim, possibly there once was some performance gain running at more corse resolution.
I know the free questa that comes with Intel is limited to minimum 1 ps and I came across a testbench today that was designed around expecting the resolution to be 1ps and which failed at 1fs. Though we should probably update the testbench to fetch the resolution from the simulator instead of expecting it to be 1ps.
I see GHDL only supports changing resolution on the mcode backend.

JimLewis, 2023-04-24:
Your maximum run length determined by 1 fs * (2**63-1).   This seems to work out to be 9223 seconds, 153.6 minutes, 2.56 hours.   

Did I do my calculations right as this would indicate that we need to be able to set the simulator resolution - even just another 1000 would give us more run time than we could use (106 days).

imphil, 2024-01-02:
Just a drive-by comment while I'm at it: When you implement this feature, please consider using a format that matches Verilog's timescale format (1ns, 10ns, 1ps, etc.). That's what most other simulators use as well, and would make our life a bit easier on the cocotb side.