nickg/nvc #608: EXCEPTION_STACK_OVERFLOW

avelure, 2023-01-31:
I have a huge testbench that creates a stack overflow when simulating.
If I delete enough code from the end it works, but if I add back some of it and delete from the start it also works, so it seems it might be some memory issue and not specific code.
Changing -H and -M does not seem to affect it

```
Thread 5 received signal SIGSEGV, Segmentation fault.
[Switching to Thread 17348.0x72f8]
0x00007ffa201fd196 in ntdll!RtlAllocateHeap () from /c/Windows/SYSTEM32/ntdll.dll
(gdb) bt
#0  0x00007ffa201fd196 in ntdll!RtlAllocateHeap () from /c/Windows/SYSTEM32/ntdll.dll
#1  0x00007ffa201fb44d in ntdll!RtlAllocateHeap () from /c/Windows/SYSTEM32/ntdll.dll
#2  0x00007ffa202c8c38 in ntdll!RtlRegisterSecureMemoryCacheCallback ()
   from /c/Windows/SYSTEM32/ntdll.dll
#3  0x00007ffa201fd255 in ntdll!RtlAllocateHeap () from /c/Windows/SYSTEM32/ntdll.dll
#4  0x00007ffa201fb44d in ntdll!RtlAllocateHeap () from /c/Windows/SYSTEM32/ntdll.dll
#5  0x00007ffa1dc3fde6 in ucrtbase!_malloc_base () from /c/Windows/System32/ucrtbase.dll
#6  0x00007ff96645a50c in ?? () from /ucrt64/bin/libstdc++-6.dll
#7  0x00007ff926ce84ff in ?? () from /ucrt64/bin/libLLVM-15.dll
#8  0x00007ff9273df25c in ?? () from /ucrt64/bin/libLLVM-15.dll
#9  0x00007ff9273dda36 in ?? () from /ucrt64/bin/libLLVM-15.dll
---- SNIP ----
#4958 0x00007ff924b78550 in ?? () from /ucrt64/bin/libLLVM-15.dll
#4959 0x00007ff6345aa087 in llvm_obj_emit (obj=0x2d122393b90, path=<optimized out>) at ../src/jit/jit-llvm.c:2931
#4960 0x00007ff6345a3cc9 in cgen_async_work (context=0x2d122363f80, arg=0x2d12236c2b0) at ../src/cgen.c:5679
#4961 0x00007ff634561422 in execute_task (task=task@entry=0xd1c01ffaf0) at ../src/thread.c:754
#4962 0x00007ff634561721 in steal_task () at ../src/thread.c:914
#4963 0x00007ff634561d91 in worker_thread (arg=<optimized out>) at ../src/thread.c:945
#4964 0x00007ff634561bf0 in thread_wrapper (arg=0x2d12236ecf0) at ../src/thread.c:445
#4965 0x00007ff9e5f04d0b in ?? () from /ucrt64/bin/libwinpthread-1.dll
#4966 0x00007ffa1dc51bb2 in ucrtbase!_configthreadlocale () from /c/Windows/System32/ucrtbase.dll
#4967 0x00007ffa1fa17614 in KERNEL32!BaseThreadInitThunk () from /c/Windows/System32/KERNEL32.DLL
#4968 0x00007ffa202226a1 in ntdll!RtlUserThreadStart () from /c/Windows/SYSTEM32/ntdll.dll
#4969 0x0000000000000000 in ?? ()
```

nickg, 2023-01-31:
The crash happens inside the LLVM library which is a bit unfortunate. Can you try setting the environment variable `NVC_LLVM_VERBOSE=1` and then re-run the elaboration step? It should print multiple messages like this:

```
** Debug: wrote LLVM IR for WORK.FOO.elab.0 to WORK.FOO.elab.0.final.ll
```

And then try running the standalone `llc` program that comes with LLVM on those files to see if it still crashes or not:

```
llc *.final.ll
```

avelure, 2023-01-31:
All of them exited with code 0 according to gdb.

nickg, 2023-02-03:
Sorry would you mind trying with `llc -O2` as well? The problematic file is probably the last one printed.

avelure, 2023-02-03:
Doesn't seem to make any difference.
I attach the output files if it is possible for you to test, if not I can test other suggestions.
The last one written is 0.final.ll
[test.zip](https://github.com/nickg/nvc/files/10583578/test.zip)
The LLVM version is
```
LLVM (http://llvm.org/):
  LLVM version 15.0.5
  Optimized build.
  Default target: x86_64-w64-windows-gnu
  Host CPU: skylake
```
Another clue might be that I only get the "EXCEPTION_STACK_OVERFLOW" printed when I execute the simulation through my python software. When running from commandline there is nothing printed, it just silently ends. The return code is -1073741819.

nickg, 2023-02-04:
If I run `llc` on that file it seems to take forever: I got bored and ctrl-c-ed it. Is there anything unusual about the process `UVVM_UTIL.RAND_TB.P_MAIN`? Something about it causes NVC to generate MBs of LLVM IR, perhaps some construct is not compiled efficiently.

nickg, 2023-02-04:
Tagged the wrong issue in the commit message.

nickg, 2023-02-04:
The commit above might make it a bit better.

avelure, 2023-02-05:
It works now, thanks!
The llc command ends eventually, but I think it took 10-15min. It also took a while to end running on NVC.
The working test only takes ~10s now.

nickg, 2023-02-05:
Great. Out of curiosity could you upload the `0.final.ll` from the latest version? I'd like to check if there's anything else that can be improved.

avelure, 2023-02-06:
[test.zip](https://github.com/nickg/nvc/files/10670041/test.zip)
Here's the llvm files with todays build.
The test does not pass, but don't worry about it, it is related to #607, I'll fix the code at some point.

nickg, 2023-02-14:
Thanks. It looks ok, although there's a lot of code generated.

commit a671ddf9 (2023-02-04): Lost events after splitting signal. Fixes #608
closed 2023-02-14 by nickg