nickg/nvc #609: Data lost through port records

avelure, 2023-01-31:
This continues on #597.
Some packets go through in in the first part of the tests, so the connections should be working (at least it all passes in modelsim)
In the wave I see the write data is present on the top level master_if, but it is missing in the master_if inside the harness.
The pulsed signals pass through however.
There are no warnings on 'U' from the interfaces.

```vhdl
library IEEE;
USE IEEE.STD_LOGIC_1164.ALL;
USE IEEE.NUMERIC_STD.ALL;

entity avalon_st_fifo is
  generic(
    GC_DATA_WIDTH    : natural := 8;
    GC_CHANNEL_WIDTH : natural := 1;
    GC_EMPTY_WIDTH   : natural := 1;
    GC_ERROR_WIDTH   : natural := 1;
    GC_FIFO_DEPTH    : natural := 256
  );
  port(
    clk_i            : in  std_logic;
    reset_i          : in  std_logic;
    slave_data_i     : in  std_logic_vector(GC_DATA_WIDTH - 1 downto 0);
    slave_channel_i  : in  std_logic_vector(GC_CHANNEL_WIDTH - 1 downto 0);
    slave_empty_i    : in  std_logic_vector(GC_EMPTY_WIDTH - 1 downto 0);
    slave_error_i    : in  std_logic_vector(GC_ERROR_WIDTH - 1 downto 0);
    slave_valid_i    : in  std_logic;
    slave_sop_i      : in  std_logic;
    slave_eop_i      : in  std_logic;
    slave_ready_o    : out std_logic := '1';
    master_data_o    : out std_logic_vector(GC_DATA_WIDTH - 1 downto 0);
    master_channel_o : out std_logic_vector(GC_CHANNEL_WIDTH - 1 downto 0);
    master_empty_o   : out std_logic_vector(GC_EMPTY_WIDTH - 1 downto 0);
    master_error_o   : out std_logic_vector(GC_ERROR_WIDTH - 1 downto 0);
    master_valid_o   : out std_logic := '0';
    master_sop_o     : out std_logic;
    master_eop_o     : out std_logic;
    master_ready_i   : in  std_logic;
    empty_o          : out std_logic := '1'
  );
end avalon_st_fifo;

architecture Behavioral of avalon_st_fifo is

begin
  p_main : process(clk_i)
    type t_data_mem is array (0 to GC_FIFO_DEPTH - 1) of std_logic_vector(GC_DATA_WIDTH - 1 downto 0);
    variable v_data_mem : t_data_mem := (others => (others => '0'));

    type t_channel_mem is array (0 to GC_FIFO_DEPTH - 1) of std_logic_vector(GC_CHANNEL_WIDTH - 1 downto 0);
    variable v_channel_mem : t_channel_mem := (others => (others => '0'));

    type t_empty_mem is array (0 to GC_FIFO_DEPTH - 1) of std_logic_vector(GC_EMPTY_WIDTH - 1 downto 0);
    variable v_empty_mem : t_empty_mem := (others => (others => '0'));

    type t_error_mem is array (0 to GC_FIFO_DEPTH - 1) of std_logic_vector(GC_ERROR_WIDTH - 1 downto 0);
    variable v_error_mem : t_error_mem := (others => (others => '0'));

    type t_sop_mem is array (0 to GC_FIFO_DEPTH - 1) of std_logic;
    variable v_sop_mem : t_sop_mem := (others => '0');

    type t_eop_mem is array (0 to GC_FIFO_DEPTH - 1) of std_logic;
    variable v_eop_mem : t_eop_mem := (others => '0');

    variable v_WrPtr  : natural range 0 to GC_FIFO_DEPTH - 1 := 0;
    variable v_RdPtr  : natural range 0 to GC_FIFO_DEPTH - 1 := 0;
    variable v_looped : boolean                              := false;
  begin
    if rising_edge(clk_i) then
      if reset_i = '1' then
[... 252 lines cut ...]
UVVM: ID_PACKET_DATA                   212.5 ns  TB seq.                        avalon_st_transmit(7 words/14 sym, ch:0)=> x"25" (symbol# 3).
UVVM: ID_PACKET_DATA                   222.5 ns  TB seq.                        avalon_st_transmit(7 words/14 sym, ch:0)=> x"93" (symbol# 4).
UVVM: ID_PACKET_DATA                   222.5 ns  TB seq.                        avalon_st_transmit(7 words/14 sym, ch:0)=> x"22" (symbol# 5).
UVVM: ID_PACKET_DATA                   232.5 ns  TB seq.                        avalon_st_transmit(7 words/14 sym, ch:0)=> x"E3" (symbol# 6).
UVVM: ID_PACKET_COMPLETE               242.5 ns  TB seq.                        avalon_st_transmit(7 words/14 sym, ch:0) DONE.
UVVM: ID_PACKET_INITIATE               245.0 ns  TB seq.                        avalon_st_expect(7 words/14 sym, ch:0) while executing avalon_st_receive=>
UVVM: ID_PACKET_DATA                   250.0 ns  TB seq.                        avalon_st_expect(7 words/14 sym, ch:0) while executing avalon_st_receive=> x"47" (symbol# 0).
UVVM: ID_PACKET_DATA                   250.0 ns  TB seq.                        avalon_st_expect(7 words/14 sym, ch:0) while executing avalon_st_receive=> x"14" (symbol# 1).
UVVM: ID_PACKET_DATA                   260.0 ns  TB seq.                        avalon_st_expect(7 words/14 sym, ch:0) while executing avalon_st_receive=> x"3B" (symbol# 2).
UVVM: ID_PACKET_DATA                   260.0 ns  TB seq.                        avalon_st_expect(7 words/14 sym, ch:0) while executing avalon_st_receive=> x"25" (symbol# 3).
UVVM: ID_PACKET_DATA                   270.0 ns  TB seq.                        avalon_st_expect(7 words/14 sym, ch:0) while executing avalon_st_receive=> x"93" (symbol# 4).
UVVM: ID_PACKET_DATA                   270.0 ns  TB seq.                        avalon_st_expect(7 words/14 sym, ch:0) while executing avalon_st_receive=> x"22" (symbol# 5).
UVVM: ID_PACKET_DATA                   280.0 ns  TB seq.                        avalon_st_expect(7 words/14 sym, ch:0) while executing avalon_st_receive=> x"E3" (symbol# 6).
UVVM: ID_BFM                           282.5 ns  TB seq.                        avalon_st_expect(7 words/14 sym, ch:0)=> OK, received 7 symbols.
UVVM: ID_PACKET_INITIATE               285.0 ns  TB seq.                        avalon_st_transmit(1 words/2 sym, ch:0)=>
UVVM: ID_PACKET_DATA                   285.0 ns  TB seq.                        avalon_st_transmit(1 words/2 sym, ch:0)=> x"47" (symbol# 0).
UVVM: ID_PACKET_COMPLETE               292.5 ns  TB seq.                        avalon_st_transmit(1 words/2 sym, ch:0) DONE.
UVVM: ID_PACKET_INITIATE               295.0 ns  TB seq.                        avalon_st_expect(1 words/2 sym, ch:0) while executing avalon_st_receive=>
UVVM: ID_PACKET_DATA                   310.0 ns  TB seq.                        avalon_st_expect(1 words/2 sym, ch:0) while executing avalon_st_receive=> x"E3" (symbol# 0).
UVVM:
UVVM: =========================================================================================================================================================================
UVVM: ***  ERROR #1  ***
UVVM:         312.5 ns   TB seq.
UVVM:                    avalon_st_expect(1 words/2 sym, ch:0)=> Failed in 3 data bits. First mismatch in symbol# 0. Was x"E3". Expected x"47".
```

nickg, 2023-01-31:
You mean this transition 00E3 -> 0000 is not propagated, right?

[image]

avelure, 2023-01-31:
I mean the next one 0047, you can see it more easily if you delete the transmit/expect before the loop, I just added those to show that it works normally.

[image]

nickg, 2023-02-04:
Thanks for the test case. It exposes a bug in an optimisation to group signal sub-elements together (the key bit is transmitting `v_data_packet(0 to 0)` which only writes to half of `data`).

commit bce92f23 (2023-02-04): Rename test case issue608.vhd to issue609.vhd. Fixes #609
commit 666c009c (2023-02-04): Lost events after splitting signal. Fixes #609
closed 2023-02-04 by nickg