nickg/nvc #612: Updates to OSVVM foreign procedure outputs cause crash

wyvernSemi, 2023-02-07:
On latest master code 1.9-devel (1.8.0.r50.g440ee8a3) updating some outputs on the VTrans foreign procedure cause a SIGV_MAPERR crash on Linux:
```
* Note: loading VHPI plugin ./VProc.so
VInit() called
VTrans() called

*** Caught signal 11 (SEGV_MAPERR) [address=0x7f8000a4, ip=0x7f008570b245] ***

[0x411951] ../src/util.c:825 signal_handler
[0x7f00856f141f] (/lib/x86_64-linux-gnu/libpthread-2.31.so) 
[0x7f008570b245] (/home/simon/OSVVM/nvc/src/VProc.so) /home/simon/OSVVM/nvc/src/code/OsvvmVSched.c:70 VTrans
           
-->      *VPDataOut        = 0; // bad on nvc 1.9-devel (1.8.0.r50.g440ee8a3)
         *VPDataOutHi      = 0; // bad
[0x7f008565cff4] (/usr/lib/x86_64-linux-gnu/libffi.so.7.1.0) 
[0x7f008565c409] (/usr/lib/x86_64-linux-gnu/libffi.so.7.1.0) 
[0x4bd541] ../src/jit/jit-ffi.c:154 jit_ffi_call
[0x4bab8c] ../src/jit/jit-exits.c:1086 __nvc_do_fficall
[0x7f0085722399] (/home/simon/OSVVM/nvc/src/work/_WORK.TEST.elab.so) WORK.TEST.TESTPROC
[0x4aa6fe] ../src/jit/jit-core.c:640 jit_fastcall
[0x4a25e7] ../src/rt/model.c:851 run_process
[0x4a25e7] ../src/rt/model.c:2281 async_run_process
[0x49890d] ../src/thread.c:754 execute_task
[0x49991c] ../src/thread.c:999 workq_start
[0x4a33c6] ../src/rt/model.c:2689 model_cycle
[0x4a4a81] ../src/rt/model.c:2730 model_run
[0x40d3ac] ../src/nvc.c:628 run
[0x40d3ac] ../src/nvc.c:1388 process_command
[0x40a77e] ../src/nvc.c:1511 main
```
A similar crash happens for Windows/MSYS2. This doesn't happen for the 1.8.1-branch (1.8.1.r10.g666c009c) where the code functions correctly (but has another issue, #603, which prevents use of some OSVVM co-simulation features). Only some of the outputs seem to cause an issue, with updating others causing no problems. 

I have condensed, as much as I can, a test bundle that reproduces the error.  A compile.sh script will compile the C/C++ code and the VHDL and run a short test, defined in test.vhd. The test will run if all the output updates of VTrans (in code/OsvvmVSched.c) are commented out. If any of those marked as bad are uncommented then the crash occurs. The ones not marked as bad can be commented out and will not crash the test.

[testcase.tar.gz](https://github.com/nickg/nvc/files/10673880/testcase.tar.gz)

nickg, 2023-02-11:
This is working now:

```
** Note: loading VHPI plugin ./VProc.so
VInit() called
VTrans() called
VTrans() called
** Note: 200ns+0: FINISH called with status 0
   Procedure FINISH [INTEGER]
         ../lib/std.08/env-body.vhd:35
   Process :test:testproc
         test.vhd:62
```

It was a regression with foreign functions with more than seven arguments. Could you try the full test again?

wyvernSemi, 2023-02-11:
The OSVVM co-simulation tests now pass unless they set a generic at elaboration with the -g option. This is a separate issue and I will try and condense a new test case and will raise another issue. This one can be closed from my perspective.

nickg, 2023-02-11:
OK, thanks for testing!

closed 2023-02-11 by nickg