nickg/nvc #614: Axilite VVC

avelure, 2023-02-09:
I'm running a testcase with the Axilite VVC from UVVM and I get a different result on NVC vs Modelsim.

```vhdl
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;
-- Opencores model
entity msec_ipcore_axilite is
  generic(
    C_S_AXI_DATA_WIDTH : integer          := 32;
    C_S_AXI_ADDR_WIDTH : integer          := 32
  );
  port(
    core_clk      : in  std_logic;
    S_AXI_ACLK    : in  std_logic;
    S_AXI_ARESETN : in  std_logic;
    S_AXI_AWADDR  : in  std_logic_vector(C_S_AXI_ADDR_WIDTH - 1 downto 0);
    S_AXI_AWVALID : in  std_logic;
    S_AXI_AWREADY : out std_logic;
    S_AXI_WDATA   : in  std_logic_vector(C_S_AXI_DATA_WIDTH - 1 downto 0);
    S_AXI_WVALID  : in  std_logic;
    S_AXI_WREADY  : out std_logic;
    S_AXI_WSTRB   : in  std_logic_vector((C_S_AXI_DATA_WIDTH / 8) - 1 downto 0);
    S_AXI_BVALID  : out std_logic;
    S_AXI_BREADY  : in  std_logic;
    S_AXI_BRESP   : out std_logic_vector(1 downto 0);
    S_AXI_ARADDR  : in  std_logic_vector(C_S_AXI_ADDR_WIDTH - 1 downto 0);
    S_AXI_ARVALID : in  std_logic;
    S_AXI_ARREADY : out std_logic;
    S_AXI_RDATA   : out std_logic_vector(C_S_AXI_DATA_WIDTH - 1 downto 0);
    S_AXI_RVALID  : out std_logic;
    S_AXI_RREADY  : in  std_logic;
    S_AXI_RRESP   : out std_logic_vector(1 downto 0)
  );
end entity msec_ipcore_axilite;

architecture IMP of msec_ipcore_axilite is
  type axi_states is (addr_wait, read_state, write_state, response_state);
  signal state : axi_states;
  signal address : std_logic_vector(C_S_AXI_ADDR_WIDTH - 1 downto 0);
  signal reset   : std_logic;
  signal S_AXI_BVALID_i : std_logic;
  signal cs_array         : std_logic_vector(6 downto 0);
  signal slv_reg_selected : std_logic;
  signal op_mem_selected  : std_logic;
  signal op_sel           : std_logic_vector(1 downto 0);
  signal MNO_sel          : std_logic;
  signal slv_reg              : std_logic_vector(C_S_AXI_DATA_WIDTH - 1 downto 0);
  signal slv_reg_write_enable : std_logic;
  signal load_flags           : std_logic;
  signal write_enable      : std_logic;
  signal core_write_enable : std_logic;
  signal core_fifo_push    : std_logic;
  signal core_data_out     : std_logic_vector(31 downto 0);
  signal core_rw_address   : std_logic_vector(8 downto 0);
  signal core_interrupt     : std_logic;
  signal core_fifo_full     : std_logic;
  signal core_fifo_nopush   : std_logic;
  signal core_ready         : std_logic;
  signal core_mem_collision : std_logic;
  signal core_start          : std_logic;
  signal core_start_bit      : std_logic;
  signal core_start_bit_d    : std_logic;
  signal core_exp_m          : std_logic;
  signal core_p_sel          : std_logic_vector(1 downto 0);
  signal core_dest_op_single : std_logic_vector(1 downto 0);
  signal core_x_sel_single   : std_logic_vector(1 downto 0);
  signal core_y_sel_single   : std_logic_vector(1 downto 0);
  signal core_flags          : std_logic_vector(15 downto 0);
  signal core_modulus_sel    : std_logic;
[... 400 lines cut ...]
# Therefore, simulation behavior may occur that is not in compliance with
# the VHDL standard as the initial values come from the base signal /test/axilite_if_1.write_response_channel.bvalid.
# ** Warning: (vsim-8684) No drivers exist on inout port /test/i1_axilite_vvc/axilite_vvc_master_if.write_response_channel.bresp(1), and its initial value is not used.
# Therefore, simulation behavior may occur that is not in compliance with
# the VHDL standard as the initial values come from the base signal /test/axilite_if_1.write_response_channel.bresp(1).
# ** Warning: (vsim-8684) No drivers exist on inout port /test/i1_axilite_vvc/axilite_vvc_master_if.write_response_channel.bresp(0), and its initial value is not used.
# Therefore, simulation behavior may occur that is not in compliance with
# the VHDL standard as the initial values come from the base signal /test/axilite_if_1.write_response_channel.bresp(0).
# ** Warning: (vsim-8684) No drivers exist on inout port /test/i1_axilite_vvc/axilite_vvc_master_if.read_address_channel.arready, and its initial value is not used.
# Therefore, simulation behavior may occur that is not in compliance with
# the VHDL standard as the initial values come from the base signal /test/axilite_if_1.read_address_channel.arready.
# ** Warning: (vsim-8684) No drivers exist on inout port /test/i1_axilite_vvc/axilite_vvc_master_if.read_data_channel.rvalid, and its initial value is not used.
# Therefore, simulation behavior may occur that is not in compliance with
# the VHDL standard as the initial values come from the base signal /test/axilite_if_1.read_data_channel.rvalid.
# ** Warning: (vsim-8684) No drivers exist on inout port /test/i1_axilite_vvc/axilite_vvc_master_if.read_data_channel.rdata(31), and its initial value is not used.
# Therefore, simulation behavior may occur that is not in compliance with
# the VHDL standard as the initial values come from the base signal /test/axilite_if_1.read_data_channel.rdata(31).
---snip---
# ** Warning: (vsim-8684) No drivers exist on inout port /test/i1_axilite_vvc/axilite_vvc_master_if.read_data_channel.rresp(1), and its initial value is not used.
# Therefore, simulation behavior may occur that is not in compliance with
# the VHDL standard as the initial values come from the base signal /test/axilite_if_1.read_data_channel.rresp(1).
# ** Warning: (vsim-8684) No drivers exist on inout port /test/i1_axilite_vvc/axilite_vvc_master_if.read_data_channel.rresp(0), and its initial value is not used.
# Therefore, simulation behavior may occur that is not in compliance with
# the VHDL standard as the initial values come from the base signal /test/axilite_if_1.read_data_channel.rresp(0).
```

Blebowski, 2023-02-10:
Hello @avelure , unrelated to this topic, I see that you are using NVC a lot and publishing a lot of
bug reports. Could you try to test with the new code coverage implementation please ? It might
reveal some bugs.

avelure, 2023-02-11:
Hi, @Blebowski. 
It is on my agenda, I just haven't gotten around to it yet.
I might have some work for you within a month or so :)

JimLewis, 2023-02-12:
@avelure    UVVM uses shared variables of an ordinary type for communication which was deprecated and removed from VHDL in the 2002 revision.   

Using variables of this sort introduces to VHDL the same race conditions that exist in Verilog and the result is that two compliant simulators can produce different simulation results due to it.   Especially if you are expecting something to happen at an exact clock cycle.    Let me explain.    During a given simulation cycle, perhaps due to clock rising, if both the process that sets a communication variable runs and the process that reads the communication variable runs, then whether the process that reads the variable sees what was set during that clock cycle or the next clock cycle will depend on which process runs first.   

The LRM does not dictate the order in which simulators run a particular process and you will find that different simulators will indeed run the processes in different orders.   It is even possible that due to optimization that a given simulator will run processes in different orders after recompiling a design.  

Your warnings from ModelSim,
# ** Warning: (vsim-8684) No drivers exist on inout port /test/i1_axilite_vvc/axilite_vvc_master_if.write_address_channel.awready, and its initial value is not used.
# Therefore, simulation behavior may occur that is not in compliance with
are due to ModelSim doing port collapsing optimizations.   What they are not clearly saying is that the result of this optimization modelsim may do something that is not in compliance with the standard.    What it seams is happening in this sort of optimization is that the ports are removed from the design and the code is migrated up one level (sort of like unrolling a subprogram).   As a result, the ports and any initializations (typically to Z) are gone.   It should be noted that this sort of issue only potentially causes issues in simulators that do port collapsing optimizations (ie: modelsim).   A long time ago, OSVVM interfaces initialized ports to all Z.   I never saw an issue with it.

nickg, 2023-02-14:
I looked at this a bit but I can't see anything obviously wrong. @avelure is there any way to spot the difference in behaviour in a waveform output? Does GHDL or another simulator give the same result as ModelSim?

nickg, 2023-02-16:
> Hello @avelure , unrelated to this topic, ...

@Blebowski BTW I just enabled GitHub Discussions for this project. That can be used in future for feedback, Q&A, etc.

Blebowski, 2023-02-19:
Hi @nickg , thanks, I will use those for further questions.

nickg, 2023-03-01:
@bpadalino tested with RiveraPRO and got the same result as Modelsim so there is probably a bug here somewhere.

bpadalino, 2023-03-01:
Note that some of the transactions did happen in a different order.

Attaching all logs here for posterity.

[modelsim.log](https://github.com/nickg/nvc/files/10865009/modelsim.log)
[nvc.log](https://github.com/nickg/nvc/files/10865010/nvc.log)
[riviera.log](https://github.com/nickg/nvc/files/10865011/riviera.log)

bpadalino, 2023-03-03:
I figured out how to fix the simulation.  By adding another `wait 0 ns` [here](https://github.com/UVVM/UVVM/blob/ba91436b3489da74b475a5e418d20fb728014063/uvvm_vvc_framework/src_target_dependent/td_vvc_entity_support_pkg.vhd#L905), the simulation completes just fine.

Unsure if this is a bug in `nvc` and the calculation of the delta cycles, or if it's a race condition that UVVM is not considering and they need to change to wait an extra delta cycle in some other places?

h/t to @JimLewis for suggesting adding an extra delta cycle.

avelure, 2023-03-03:
Yes, I found the same earlier, there is a delta-cycle transition that can be filtered out by adding delta-cycle waits at various places, but I don't think that fixes the root cause.
I think it might be due to handling of the protected variable queues, but I'm still investigating.
`queues_are_empty(VOID)` returns true at one point where it shouldn't.

bpadalino, 2023-03-03:
> I think it might be due to handling of the protected variable queues, but I'm still investigating.
> `queues_are_empty(VOID)` returns true at one point where it shouldn't.

Just curious, what makes you say that it returns `true` when it shouldn't?  Is it possible the explanation @JimLewis gave [above](https://github.com/nickg/nvc/issues/614#issuecomment-1427119380) about the ordering of processes?

Do you know if there is something we can instrument to see if `nvc` and `modelsim` are running the processes in different orders?

Just trying to understand the thought process and to verify that indeed the function should return true when it isn't.

JimLewis, 2023-03-03:
@avelure   This was a curious problem to look at.   I added to my notebook that when setting and reading flags in a shared variable - whether it is a protected type or ordinary type - you can run into process order issues.    

So you need to work though the details of when await_completion reads any activity indication and when axilite_write indicates it is doing activity.

nickg, 2023-03-03:
It's definitely to do with the order the processes are run in. Here's a very simple diff you can apply to NVC which causes the processes to be run in the reverse order to normal:

```diff
diff --git a/src/thread.c b/src/thread.c
index 1ccf27e801a0..1428d1fcdc42 100644
--- a/src/thread.c
+++ b/src/thread.c
@@ -1020,7 +1020,7 @@ void workq_start(workq_t *wq)
             nparallel += wptr.count;
          }
          else {
-            for (int j = 0; j < wptr.count; j++)
+            for (int j = wptr.count - 1; j >= 0; j--)
                execute_task(&(eq->tasks[j]));
             nserial += wptr.count;
          }
```

If I apply that then the test has the same behaviour as Modelsim/RiveraPRO.

The LRM doesn't specify which order the processes are to be run in (they might even be run concurrently) so UVVM shouldn't be relying on that. We do seem to be running them in a different order to every other simulator though, but I think that's just an implementation detail.

bpadalino, 2023-03-03:
I wonder if there is any value in being able to randomize this to help expose issues with verification frameworks?  Maybe run forward, backward and random?

Anyway, I'm glad the little insight @JimLewis had was able to resolve where the bug resides.

nickg, 2023-09-11:
> I wonder if there is any value in being able to randomize this to help expose issues with verification frameworks? Maybe run forward, backward and random?
> 

This is a good idea. I've added a `--shuffle` option to the `-r` command. With that the test case above can pass around half the time.

avelure, 2024-09-20:
I did not find the root cause for this in UVVM, meanwhile there has been a new release of UVVM v2 (2024.09.19) where the affected code has been rewritten to avoid the issue in all VVCs.

closed 2024-09-20 by avelure