nickg/nvc #617: Casting sfixed to std_ulogic_vector does not cause an error

bpadalino, 2023-02-10:
The following test should produce an error, but instead casts the `sfixed` type to a `std_ulogic_vector` with `'left` being 31 and `'right` being -32.

```vhdl
entity test is
end test;

library ieee;
use ieee.std_logic_1164.all;
use ieee.fixed_pkg.all;

architecture rtl of test is

  procedure func(y: in std_ulogic_vector) is
  begin
    report to_string(y'left) & " to " & to_string(y'right);
  end procedure;

begin

  process
    variable example : sfixed(31 downto -32);
  begin
    func(std_ulogic_vector(example));
    wait;
  end process;
end rtl;
```

Questa prints the following error:

```
** Error: test.vhd(20): (vcom-1167) Index value -32 (of type std.STANDARD.NATURAL) is out of range 0 to 2147483647.
```

bpadalino, 2023-02-20:
I think the issue is in [`bounds_check_assignment()`](https://github.com/nickg/nvc/blob/31a7f32a14e3c320910933bcab3ac9a3d6afdbfc/src/bounds.c#L859) because it checks the lengths of the assignment, but it doesn't check the indices of the target versus the value for validity.

I am also trying to understand the LRM08 on page 137:

>If the target subtype is an array type or subtype for which the index ranges are not defined, then, for
each index position, the index range of the result is determined as follows:
>
>—If the index type of the operand and the index type of the target type are not closely related,
then the direction and nominal left bound of the index range of the result are the direction and
left bound, respectively, of the corresponding index subtype of the target type. For a non-null
range, the left bound of the index range is the nominal left bound, and the right bound is deter-
mined by the number of values in the corresponding index range of the operand. For a null
range, if there is a value to the left of the nominal left bound (given by the 'LEFTOF attribute),
then the left bound is the nominal left bound, and the right bound is the value to the left of the
nominal left bound; otherwise, the left bound is the value to the right of the nominal left bound,
and the right bound is the nominal left bound. For either a non-null or a null range, it is an error
if the base type of the corresponding index subtype of the target type does not include sufficient
values for the index range of the result.
>
>—If the index type of the operand and the index type of the target type are closely related, then the
bounds of the index range of the result are obtained by converting the bounds of the index
range of the operand to the index type of the target type, and the direction of the index range of
the result is the direction of the index type of the operand.

There is another paragraph later that describes another failing condition:

>In the case of conversions between array types, if the target subtype is an array type for which the index
ranges are not defined, then, for each index position, a check is made that the bounds of the result belong to
the corresponding index subtype of the target type. If the target subtype is an array subtype for which the
index ranges are defined, a check is made that for each element of the operand there is a matching element of
the target subtype, and vice versa. It is an error if any of these checks fail.

`nvc` definitely still has an issue like with this:

```vhdl
entity test is
end test ;

architecture arch of test is

begin

    tb : process
        type smallstring is array(natural range 1 to 5) of character ;
        constant s : string := "a pretty long string" ;
        constant ss : smallstring := smallstring(s) ;
    begin
        std.textio.write(std.textio.output, to_string(s'left) & " - " & to_string(s'right) & LF) ;
        std.textio.write(std.textio.output, to_string(ss'left) & " - " & to_string(ss'right) & LF) ;
        std.env.stop ;
    end process ;

end architecture ;
```

Is [`bounds_check_assignment()`](https://github.com/nickg/nvc/blob/31a7f32a14e3c320910933bcab3ac9a3d6afdbfc/src/bounds.c#L859) the right place to put the check of the indices and lengths of things?

nickg, 2023-02-20:
> 
> Is [`bounds_check_assignment()`](https://github.com/nickg/nvc/blob/31a7f32a14e3c320910933bcab3ac9a3d6afdbfc/src/bounds.c#L859) the right place to put the check of the indices and lengths of things?

No, because there's no assignment here. The problem is `bounds_check_type_conv` is missing the index checks for arrays. I've added that but there are runtime checks missing too.

nickg, 2023-02-26:
Added the corresponding runtime checks in the commit above.

closed 2023-02-26 by nickg