nickg/nvc #619: Mac, brew & NVC

FredKellerman, 2023-02-11:
Hi Nick,

Sorry to file an issue, was just wondering if you knew of anyone who worked out compiling this on a recent m1 silicon Mac?

The configure doesn't like the sort of incomplete llvm that brew installs in a way to not break Xcode.

Nice work here, no worries if this is cannot be done!

nickg, 2023-02-11:
LLVM from homebrew does have llvm-config but it's not in $PATH so you need to pass the location explicitly with the `--with-llvm` argument to configure. The following works for me on an M1 Mac (my packages might be out of date so you need to check which exact version you have):

```
../configure --with-llvm=/opt/homebrew/Cellar/llvm/15.0.7_1/bin/llvm-config
```

BTW you can also install directly with `brew install nvc` or `brew install --HEAD nvc` as it's packaged.

nickg, 2023-02-11:
Actually the best way to do it is: `--with-llvm=$(brew --prefix llvm)/bin/llvm-config`

FredKellerman, 2023-02-13:
Hi Nick,  Thank you!  Used your very last suggestion, compiled and running!

closed 2023-02-13 by FredKellerman