nickg/nvc #637: fstReaderOpen failed for temporary FST file

avelure, 2023-02-28:
I get 
`** Fatal: 1050ns+3: fstReaderOpen failed for temporary FST file` 
when trying to create a VCD file with the command
`nvc --std=08 -e test-beh -r -w --format=vcd`

I get a temp.fst file and a test-beh.vcd, however the vcd file is empty.

nickg, 2023-03-05:
I couldn't reproduce this (but I only used a simple test case). Can you try running the `fst2vcd` program that comes with GtkWave on the `temp.fst` file and see if that gives any error?

erick166, 2024-08-07:
I've tried generating a VCD file using the "-w --format=vcd" arguments, but the file appears to be empty.
It works however if I use "--format=fst", any ideas why?

nickg, 2024-08-07:
@erick166 are you using Windows as well? The VCD writer is actually just the FST writer with an extra step where it writes the FST to a temporary file and then does the equivalent of `fst2vcd` on it.

erick166, 2024-08-07:
Yes, I'm using Windows.

nickg, 2024-08-07:
I'm still unable to reproduce this (trying MinGW x64 build in a Windows 10 VM). Does it happen reliably with any design you try? @erick166 did you get the same "fstReaderOpen failed for temporary FST file" as in the first message?

erick166, 2024-08-07:
Yes, it happens with different testbenches and I also get the same message. Could it maybe be something in the file path separators for the temporary file?

nickg, 2024-08-07:
I managed to reproduce it: it works fine in MSYS2 MingW x64 but *not* UCRT x64 (and I guess Clang x64 too since it uses the same libc).

nickg, 2024-08-07:
The problem seems to be a difference in behaviour under UCRT in GtkWave's `fstapi.c` library that we import. I'm still not sure I understand what's going on but I pushed a workaround that fixes it for me. Could you test again?

erick166, 2024-08-08:
I tested with the fix and it works for me as well, thanks!

avelure, 2024-08-08:
I can confirm I was likely using either ucrt64 or clang64 at the time.

nickg, 2024-08-09:
Thanks for testing. I submitted a slightly improved fix to GtkWave in gtkwave/gtkwave#376.

closed 2024-08-09 by nickg