nickg/nvc #642: Linux Distribution Packaging

bpadalino, 2023-03-06:
A low priority request is to figure out packaging such that different OS distributions can quickly and easily add `nvc` to their package managers.  I don't have an exhaustive list of what package mangers to support but `dpkg`, `rpm`, and `pkg` seem like good starting points.

nickg, 2023-04-23:
The current packaging state is: https://repology.org/project/nvc/versions

I think the most important missing ones now are MSYS2 and Debian (which would then flow into Ubuntu, etc.). @amb5l made a PR for MSYS2 a while back - msys2/MINGW-packages#13613 - but it seems to have gotten stuck. Perhaps bumping the version to the 1.9.1 release would help. 

I've added a `.deb` package to the GitHub Actions artefacts but it would be best if we could get it into the distribution proper. Do do that I think someone needs to create a "Request For Package" ticket but maybe we could also approach the Debian Electronics Team who package GHDL and other FOSS EDA tools to see if they would be willing to help: https://qa.debian.org/developer.php?login=pkg-electronics-devel%40lists.alioth.debian.org

amb5l, 2023-04-24:
@nickg I had forgotten about that PR... I've closed it, how would you feel about creating another in your own right? The attached may save you some time:

[PKGBUILD.zip](https://github.com/nickg/nvc/files/11309420/PKGBUILD.zip)