nickg/nvc #644: VHDL-2008: Generic Package on Entity

bpadalino, 2023-03-06:
Taken from a [GHDL issue #695](https://github.com/ghdl/ghdl/issues/695) - this doesn't seem to parse and expects the package generic map on the entity to be the box (`<>`), but the map should be valid.

```vhdl
library ieee;
use ieee.std_logic_1164.all;

package mux is
  generic (
    MUX_DATA_SIZE : natural
  );
  subtype mux_data is std_logic_vector(MUX_DATA_SIZE-1 downto 0);
  type mux_data_array is array (natural range <>) of mux_data;
end package;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity multiplexer is
  generic (
    MUX_DATA_SIZE : natural;
    MUX_CTRL_SIZE : natural;
    --package mux_g is new work.mux generic map(MUX_DATA_SIZE => MUX_DATA_SIZE)
    package mux_g is new work.mux generic map(<>)
  );
  port (
    MUX_CTRL : in  std_logic_vector(MUX_CTRL_SIZE-1 downto 0);
    MUX_IN   : in  mux_g.mux_data_array(0 to 2**MUX_CTRL_SIZE-1);
    MUX_OUT  : out mux_g.mux_data
  );
end entity;

architecture multiplexer_arch of multiplexer is
begin
  MUX_OUT <= MUX_IN(to_integer(unsigned(MUX_CTRL)));
end architecture;

library ieee ;
use ieee.std_logic_1164.all;

entity test is
end entity ;

architecture arch of test is

    constant MUX_DATA_SIZE : natural := 2 ;
    constant MUX_CTRL_SIZE : natural := 2 ;
    package mux_g is new work.mux generic map(MUX_DATA_SIZE => MUX_DATA_SIZE) ;

    signal MUX_CTRL : std_logic_vector(MUX_CTRL_SIZE-1 downto 0) ;
    signal MUX_IN   : mux_g.mux_data_array(0 to 2**MUX_CTRL_SIZE-1) ;
    signal MUX_OUT  : mux_g.mux_data ;

begin
    U_multiplexer : entity work.multiplexer
      generic map (
        MUX_DATA_SIZE => MUX_DATA_SIZE,
        MUX_CTRL_SIZE => MUX_CTRL_SIZE,
        mux_g         => mux_g
      ) port map (
        MUX_CTRL => MUX_CTRL,
        MUX_IN   => MUX_IN,
        MUX_OUT  => MUX_OUT
      ) ;
end architecture ;
```

Riviera doesn't mind both `<>` and the map, but Questa only takes the `<>` there as well.

`nvc` will complain when supplied with `<>` that the package isn't used correctly:
```
$ nvc --std=2008 -a test.vhdl -e test -r test
** Error: invalid use of package MUX_G
    > test.vhdl:25
    |
 25 |     MUX_IN   : in  mux_g.mux_data_array(0 to 2**MUX_CTRL_SIZE-1);
    |                    ^^^^^
** Error: invalid use of package MUX_G
    > test.vhdl:26
    |
 26 |     MUX_OUT  : out mux_g.mux_data
    |                    ^^^^^
** Error: design unit depends on WORK.MULTIPLEXER which was analysed with errors
    > test.vhdl:30
    |
 30 | architecture multiplexer_arch of multiplexer is
    | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
** Error: design unit MULTIPLEXER not found in library WORK
    > test.vhdl:52
    |
 52 |     U_multiplexer : entity work.multiplexer
    |                     ^^^^^^^^^^^^^^^^^^^^^^^
** Error: invalid use of package MUX_G
    > test.vhdl:56
    |
 56 |         mux_g         => mux_g
    |         ^^^^^
** Error: invalid use of package MUX_G
    > test.vhdl:56
    |
 56 |         mux_g         => mux_g
    |                          ^^^^^
```

Questa doesn't complain to simulate, but Riviera complains with:

```
# ELBREAD: Error: ELBREAD_0097 test.vhdl (51): Port '~mux_data_array{0~to~2**MUX_CTRL_SIZE-1}~132' not found in riviera.multiplexer(multiplexer_arch) instantiated in unit riviera.test(arch).
```

This is quite a strange corner case, but an interesting use of package generics.  It seems like nobody does it the same.

nickg, 2023-03-15:
I've improved this a bit. Now it fails with:

```
** Error: type of actual MUX_DATA_ARRAY does not match type MUX_DATA_ARRAY of formal port MUX_IN
    > test.vhd:49
    |
 49 |         MUX_IN   => MUX_IN,
    |                     ^^^^^^
```

nickg, 2023-05-11:
The test case works now, but only the version with `generic map (<>)`. I'm not actually sure how the non-box version is supposed to work: I need to check the LRM.

nickg, 2023-05-12:
The full generic map aspect is now parsed and checked, including the `default` variant.

closed 2023-05-12 by nickg