nickg/nvc #646: Vivado 2022.2 unisim library issue

amb5l, 2023-03-07:
The source for the unisim odelaye3 primitive has changed between Vivado 2022.1 and 2022.2. NVC does not like at least one change:

```
D:\devtools\msys64\mingw64\bin\nvc.exe --std=1993 --work=C:/work/.nvc/lib/unisim -LC:/work/.nvc/lib -M 64m -a --relaxed /d/devtools/Xilinx/Vivado/2022.2/data/vhdl/src/unisims/primitive/ODELAYE3.vhd
** Error: cannot reference signal PER_BIT_FINE_DELAY during static elaboration
     > D:\devtools\Xilinx\Vivado\2022.2\data\vhdl\src\unisims\primitive\ODELAYE3.vhd:534
     |
 534 |   variable idelay_count_var : integer := ODELAYE3_InitDelayCount(DELAY_FORMAT,DELAY_VALUE,PER_BIT_FINE_DELAY);
     |                                                                                           ^^^^^^^^^^^^^^^^^^
     |
     = Note: the value of a signal is not defined until after the design hierarchy is elaborated
     = Help: IEEE Std 1076-1993 section 12.3 "Elaboration of a declarative part"
** Fatal: bash D:\devtools\msys64\mingw64\libexec\nvc\install-vivado.sh failed with status 1

```
The PER_BIT_FINE_DELAY signal is declared with an initial value (5), in case this is relevant.

nickg, 2023-03-12:
I think this is a regression. Although it's not strictly valid VHDL it used to work with `--relaxed`.

amb5l, 2023-03-12:
Agreed, Vivado 2022.2 was working (in this respect) back when I raised issue #566.

nickg, 2023-03-12:
Can you try again with the latest master? I'd test it myself but the Vivado download is massive.

amb5l, 2023-03-12:
OK that's improved. The reported error is now a warning, and the Vivado library install process completes. I did, however, see the following in the middle of the build process for NVC itself:

```
  AR       lib/libcpustate.a
  CC       thirdparty/memmem.o
  CC       thirdparty/strcasestr.o
  CC       thirdparty/getline.o
  CC       thirdparty/getdelim.o
  AR       lib/libgnulib.a
make[2]: [D:\devtools\msys64\tmp\ccKpH1us.mk:6: D:\devtools\msys64\tmp\ccANRFLo.ltrans1.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccKpH1us.mk:3: D:\devtools\msys64\tmp\ccANRFLo.ltrans0.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccKpH1us.mk:72: D:\devtools\msys64\tmp\ccANRFLo.ltrans23.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccKpH1us.mk:51: D:\devtools\msys64\tmp\ccANRFLo.ltrans16.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccKpH1us.mk:21: D:\devtools\msys64\tmp\ccANRFLo.ltrans6.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccKpH1us.mk:66: D:\devtools\msys64\tmp\ccANRFLo.ltrans21.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccKpH1us.mk:33: D:\devtools\msys64\tmp\ccANRFLo.ltrans10.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccKpH1us.mk:30: D:\devtools\msys64\tmp\ccANRFLo.ltrans9.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccKpH1us.mk:75: D:\devtools\msys64\tmp\ccANRFLo.ltrans24.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccKpH1us.mk:36: D:\devtools\msys64\tmp\ccANRFLo.ltrans11.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccKpH1us.mk:63: D:\devtools\msys64\tmp\ccANRFLo.ltrans20.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccKpH1us.mk:18: D:\devtools\msys64\tmp\ccANRFLo.ltrans5.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccKpH1us.mk:27: D:\devtools\msys64\tmp\ccANRFLo.ltrans8.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccKpH1us.mk:24: D:\devtools\msys64\tmp\ccANRFLo.ltrans7.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccKpH1us.mk:39: D:\devtools\msys64\tmp\ccANRFLo.ltrans12.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccKpH1us.mk:12: D:\devtools\msys64\tmp\ccANRFLo.ltrans3.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccKpH1us.mk:42: D:\devtools\msys64\tmp\ccANRFLo.ltrans13.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccKpH1us.mk:54: D:\devtools\msys64\tmp\ccANRFLo.ltrans17.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccKpH1us.mk:48: D:\devtools\msys64\tmp\ccANRFLo.ltrans15.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccKpH1us.mk:45: D:\devtools\msys64\tmp\ccANRFLo.ltrans14.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccKpH1us.mk:57: D:\devtools\msys64\tmp\ccANRFLo.ltrans18.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccKpH1us.mk:60: D:\devtools\msys64\tmp\ccANRFLo.ltrans19.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccKpH1us.mk:15: D:\devtools\msys64\tmp\ccANRFLo.ltrans4.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccKpH1us.mk:78: D:\devtools\msys64\tmp\ccANRFLo.ltrans25.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccKpH1us.mk:69: D:\devtools\msys64\tmp\ccANRFLo.ltrans22.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccKpH1us.mk:81: D:\devtools\msys64\tmp\ccANRFLo.ltrans26.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccKpH1us.mk:9: D:\devtools\msys64\tmp\ccANRFLo.ltrans2.ltrans.o] Error 1 (ignored)
  NVC      lib/std/STD.STANDARD
  NVC      lib/std/STD.TEXTIO
  NVC      lib/nvc/NVC.POLYFILL
  NVC      lib/std/STD.TEXTIO-body
  NVC      lib/ieee/IEEE.STD_LOGIC_1164
```

The above happened after checking out the latest master. I then tried a clean clone, and got a tiny change - errors after the CCLD step:

```
  AR       lib/libcpustate.a
  CC       thirdparty/memmem.o
  CC       thirdparty/strcasestr.o
  CC       thirdparty/getline.o
  CC       thirdparty/getdelim.o
  AR       lib/libgnulib.a
  CCLD     bin/nvc.exe
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:6: D:\devtools\msys64\tmp\ccKaDZGw.ltrans1.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:3: D:\devtools\msys64\tmp\ccKaDZGw.ltrans0.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:72: D:\devtools\msys64\tmp\ccKaDZGw.ltrans23.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:51: D:\devtools\msys64\tmp\ccKaDZGw.ltrans16.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:21: D:\devtools\msys64\tmp\ccKaDZGw.ltrans6.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:66: D:\devtools\msys64\tmp\ccKaDZGw.ltrans21.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:33: D:\devtools\msys64\tmp\ccKaDZGw.ltrans10.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:30: D:\devtools\msys64\tmp\ccKaDZGw.ltrans9.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:75: D:\devtools\msys64\tmp\ccKaDZGw.ltrans24.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:36: D:\devtools\msys64\tmp\ccKaDZGw.ltrans11.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:63: D:\devtools\msys64\tmp\ccKaDZGw.ltrans20.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:18: D:\devtools\msys64\tmp\ccKaDZGw.ltrans5.ltrans.o] Error 1 (ignored)
[... cut ...]
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:69: D:\devtools\msys64\tmp\ccKaDZGw.ltrans22.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:81: D:\devtools\msys64\tmp\ccKaDZGw.ltrans26.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:9: D:\devtools\msys64\tmp\ccKaDZGw.ltrans2.ltrans.o] Error 1 (ignored)
  NVC      lib/std/STD.STANDARD
  NVC      lib/std/STD.TEXTIO
  NVC      lib/nvc/NVC.POLYFILL
  NVC      lib/std/STD.TEXTIO-body
  NVC      lib/ieee/IEEE.STD_LOGIC_1164
```

nickg, 2023-03-13:
I've seen that before while building the MSYS2 package. I'm not sure what causes it but it seems to be harmless. Presumably configuring with `--disable-lto` would make it go away at a slight cost in performance. Probably worth creating a separate issue?

amb5l, 2023-03-13:
> Presumably configuring with --disable-lto would make it go away at a slight cost in performance.

Confirmed.

> Probably worth creating a separate issue?

Will do.