nickg/nvc #648: Minor MSYS2 build problem

amb5l, 2023-03-13:
As mentioned in issue #646, MSYS2 builds produce (apparently harmless) errors. These go away if the `--disable-lto` option is used with `configure`:

```
  AR       lib/libgnulib.a
  CCLD     bin/nvc.exe
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:6: D:\devtools\msys64\tmp\ccKaDZGw.ltrans1.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:3: D:\devtools\msys64\tmp\ccKaDZGw.ltrans0.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:72: D:\devtools\msys64\tmp\ccKaDZGw.ltrans23.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:51: D:\devtools\msys64\tmp\ccKaDZGw.ltrans16.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:21: D:\devtools\msys64\tmp\ccKaDZGw.ltrans6.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:66: D:\devtools\msys64\tmp\ccKaDZGw.ltrans21.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:33: D:\devtools\msys64\tmp\ccKaDZGw.ltrans10.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:30: D:\devtools\msys64\tmp\ccKaDZGw.ltrans9.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:75: D:\devtools\msys64\tmp\ccKaDZGw.ltrans24.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:36: D:\devtools\msys64\tmp\ccKaDZGw.ltrans11.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:63: D:\devtools\msys64\tmp\ccKaDZGw.ltrans20.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:18: D:\devtools\msys64\tmp\ccKaDZGw.ltrans5.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:27: D:\devtools\msys64\tmp\ccKaDZGw.ltrans8.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:24: D:\devtools\msys64\tmp\ccKaDZGw.ltrans7.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:39: D:\devtools\msys64\tmp\ccKaDZGw.ltrans12.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:12: D:\devtools\msys64\tmp\ccKaDZGw.ltrans3.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:42: D:\devtools\msys64\tmp\ccKaDZGw.ltrans13.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:54: D:\devtools\msys64\tmp\ccKaDZGw.ltrans17.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:48: D:\devtools\msys64\tmp\ccKaDZGw.ltrans15.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:45: D:\devtools\msys64\tmp\ccKaDZGw.ltrans14.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:57: D:\devtools\msys64\tmp\ccKaDZGw.ltrans18.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:60: D:\devtools\msys64\tmp\ccKaDZGw.ltrans19.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:15: D:\devtools\msys64\tmp\ccKaDZGw.ltrans4.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:78: D:\devtools\msys64\tmp\ccKaDZGw.ltrans25.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:69: D:\devtools\msys64\tmp\ccKaDZGw.ltrans22.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:81: D:\devtools\msys64\tmp\ccKaDZGw.ltrans26.ltrans.o] Error 1 (ignored)
make[2]: [D:\devtools\msys64\tmp\ccgC3XOH.mk:9: D:\devtools\msys64\tmp\ccKaDZGw.ltrans2.ltrans.o] Error 1 (ignored)
  NVC      lib/std/STD.STANDARD
  NVC      lib/std/STD.TEXTIO
```

nickg, 2023-10-31:
This doesn't seem to happen any more with an updated MSYS2. Can anyone confirm?

avelure, 2023-11-01:
I can confirm.

amb5l, 2023-11-02:
Me too. I think you can close this one.

closed 2023-11-02 by nickg