nickg/nvc #652: An option to build nvc without "default" library paths

filmil, 2023-03-19:
When NVC is compiled, it obtains a default setting for standard library paths. The default path is usually `$PREFIX/lib/nvc` and such.

This makes using NVC in hermetic environments like `bazel` harder than necessary. It would be nice if the library search path could be made empty by default, requiring the `-L` switch. 
This works much better with `bazel`, which assumes no implicit environment, to achieve hermetic reproducible builds.

nickg, 2023-03-19:
I added a `--disable-default-paths` configure option. Can you see if that does what you want?

filmil, 2023-03-19:
I will report back once I check, but from the looks of it, seems like that's what is needed.

nickg, 2023-03-31:
Presumably this worked ok.

filmil, 2023-03-31:
Yes, sorry I forgot to report back - was distracted by trying to compile nvc to be a fully static binary.

The fix worked as expected.

I was, however, unable to make `nvc` into a working fully static binary. :)

nickg, 2023-03-31:
> 
> I was, however, unable to make `nvc` into a working fully static binary. :)

That probably won't work because the compiled VHDL code needs to dynamically link to symbols from the main binary.

filmil, 2023-03-31:
> That probably won't work because the compiled VHDL code needs to dynamically link to symbols from the main binary.

Indeed. That is only one of several issues preventing it from happening. This makes nvc more difficult to distribute, than to compile locally.

In the end I fell back to local compilation.