nickg/nvc #653: Nested Packages Not Parsed Correctly

bpadalino, 2023-03-19:
The following code doesn't seem to parse correctly, even though I believe it's correct syntax.

```vhdl
library ieee ;
    use ieee.std_logic_1164.all ;

package axi4s is

    type axis_t is record
        data    :   std_ulogic_vector ;
        dest    :   std_ulogic_vector ;
        id      :   std_ulogic_vector ;
        strb    :   std_ulogic_vector ;
        keep    :   std_ulogic_vector ;
        user    :   std_ulogic_vector ;
        last    :   std_ulogic ;
        valid   :   std_ulogic ;
        ready   :   std_ulogic ;
    end record ;

    type axis_array_t is array(natural range <>) of axis_t ;

    package make is
      generic (
        DATA_BYTES  :   positive    := 4 ;
        DEST_WIDTH  :   natural     := 0 ;
        ID_WIDTH    :   natural     := 0 ;
        USER_WIDTH  :   natural     := 0 ;
      ) ;

        subtype DATA_RANGE is natural range DATA_BYTES*8-1 downto 0 ;
        subtype DEST_RANGE is natural range DEST_WIDTH-1 downto 0 ;
        subtype ID_RANGE   is natural range ID_WIDTH-1 downto 0 ;
        subtype KEEP_RANGE is natural range DATA_BYTES-1 downto 0 ;
        subtype USER_RANGE is natural range USER_WIDTH-1 downto 0 ;

        subtype axis_t is axis_t(
            data(DATA_RANGE),
            dest(DEST_RANGE),
            id(ID_RANGE),
            keep(KEEP_RANGE),
            strb(KEEP_RANGE),
            user(USER_RANGE)
        ) ;

    end package ;

end package ;

package axis32 is new package work.axi4s.make ;

entity test is
  port (
    clock   :   in  bit ;
    reset   :   in  bit ;
    rx      :   inout   work.axis32.axi_t ;
    --rx      :   inout   work.axi4s.axis_t(data(31 downto 0), dest(-1 downto 0), id(-1 downto 0), keep(3 downto 0), strb(-1 downto 0), user(-1 downto 0)) ;
    tx      :   inout   work.axi4s.axis_t(data(31 downto 0), dest(-1 downto 0), id(-1 downto 0), keep(3 downto 0), strb(-1 downto 0), user(-1 downto 0))
  ) ;
end entity ;

architecture arch of test is

begin

end architecture ;
```

The error is:
```
$ nvc --std=2008 -a test.vhdl
** Error: unexpected generic while parsing package declarative item, expecting new
    > test.vhdl:21
    |
 21 |       generic (
    |       ^^^^^^^ this token was unexpected
** Error: AXIS_T already declared in this region
    > test.vhdl:34
    |
  6 |     type axis_t is record
    |     ^ previous declaration was here
 ...
 34 |         subtype axis_t is axis_t(
    |         ^ duplicate declaration
** Error: unexpected end while parsing library unit, expecting one of entity, configuration, architecture, package or context
    > test.vhdl:45
    |
 45 | end package ;
    | ^^^ this token was unexpected
```

I am unsure if I can redeclare `axis_t` in this context, too.  But that might be a different issue.

nickg, 2023-03-20:
The example above works now but nested package bodies don't yet.

bpadalino, 2023-03-20:
Thanks.  If I try to use it like this:

```vhdl
package axis32 is new work.axi4s.make ;

entity test is
  port (
    clock   :   in  bit ;
    reset   :   in  bit ;
    rx      :   inout   work.axis32.axis_t ;
    tx      :   inout   work.axi4s.axis_t(data(31 downto 0), dest(-1 downto 0), id(-1 downto 0), keep(3 downto 0), strb(-1 downto 0), user(-1 downto 0))
  ) ;
end entity ;

architecture arch of test is

begin

    -- do nothing for now

end architecture ;
```

Then I get an elaboration error:
```
$ nvc --std=2008 -a test.vhdl -e test

*** Caught signal 11 (SEGV_MAPERR) [address=0x563318bc76f8, ip=0x561b17a6fcbc] ***

[0x561b179dfe4d] ../src/util.c:837 signal_handler
[0x7fa58d6f951f] (/usr/lib/x86_64-linux-gnu/libc.so.6) 
[0x561b17a6fcbc] ../src/vcode.c:3603 emit_load
[0x561b17a58f82] ../src/lower.c:2945 lower_ref
[0x561b17a58f82] ../src/lower.c:4781 lower_expr.lto_priv.0
[0x561b17a701fd] ../src/lower.c:10657 lower_rvalue
[0x561b17a50ad1] ../src/lower.c:1207 lower_param.lto_priv.0
[0x561b17a75bdb] ../src/lower.c:2246 lower_builtin.part.0.lto_priv.0
[0x561b17a571b1] ../src/lower.c:2239 lower_fcall
[0x561b17a571b1] ../src/lower.c:4775 lower_expr.lto_priv.0
[0x561b17a701fd] ../src/lower.c:10657 lower_rvalue
[0x561b17a50ad1] ../src/lower.c:1207 lower_param.lto_priv.0
[0x561b17a75bdb] ../src/lower.c:2246 lower_builtin.part.0.lto_priv.0
[0x561b17a571b1] ../src/lower.c:2239 lower_fcall
[0x561b17a571b1] ../src/lower.c:4775 lower_expr.lto_priv.0
[0x561b17a701fd] ../src/lower.c:10657 lower_rvalue
[0x561b17acc11d] ../src/lower.c:931 lower_constraints.constprop.0
[0x561b17a62c67] ../src/lower.c:7267 lower_sub_signals.lto_priv.0
[0x561b17a62f42] ../src/lower.c:7452 lower_sub_signals.lto_priv.0
[0x561b17a6bc30] ../src/lower.c:10387 lower_port_signal.lto_priv.0
[0x561b17a31de2] ../src/lower.c:10462 lower_instance
[0x561b17a31de2] ../src/lower.c:1167 elab_lower
[0x561b17a3a4de] ../src/elab.c:1952 elab_top_level.lto_priv.0
[0x561b179e946a] ../src/nvc.c:2000 elaborate
[0x561b179e946a] ../src/elab.c:1431 process_command
[0x561b179e89e9] ../src/nvc.c:448 elaborate
[0x561b179e89e9] ../src/nvc.c:1431 process_command
[0x561b179dc583] ../src/nvc.c:1553 main
```

nickg, 2023-06-23:
This elaborates without error now but I'm not sure how to actually test it.

closed 2023-06-23 by nickg