nickg/nvc #658: Generic = or /= function in package causes duplicate declaration

bpadalino, 2023-03-23:
The following code is legal, but appears to cause `nvc` to think there is a duplicate declaration:

```vhdl
package pack is
  generic (
    type t ;
    function "="(l, r : t) return boolean is <>;
    function "/="(l, r : t) return boolean is <>
  ) ;

    function equal(l, r : t) return boolean ;
    function notequal(l, r : t) return boolean ;

end package ;

package body pack is

    function equal(l, r : t) return boolean is
    begin
        return l = r ;
    end function ;

    function notequal(l, r : t) return boolean is
    begin
        return l /= r ;
    end function ;

end package body ;
```

The error is:

```
$ nvc --std=2008 -a test.vhdl 
** Error: "=" [T, T return BOOLEAN] already declared in this region
   > test.vhdl:4
   |
 3 |     type t ;
   |     ^^^^^^ previous declaration was here
 4 |     function "="(l, r : t) return boolean is <>;
   |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ duplicate declaration
** Error: "/=" [T, T return BOOLEAN] already declared in this region
   > test.vhdl:5
   |
 3 |     type t ;
   |     ^^^^^^ previous declaration was here
 4 |     function "="(l, r : t) return boolean is <>;
 5 |     function "/="(l, r : t) return boolean is <>
   |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ duplicate declaration
** Error: design unit depends on WORK.PACK which was analysed with errors
    > test.vhdl:13
    |
 13 | package body pack is
    | ^^^^^^^^^^^^^^^^^

```

bpadalino, 2023-03-23:
I see - section 6.5.3 of the LRM-2008 states that "=" and "/=" are implicitly declared as formal generic subprograms.

Does that mean I can't override them?

nickg, 2023-03-24:
> 
> Does that mean I can't override them?

I re-read section 12.3 on visibility and it seems to me that you should be able to override them.

nickg, 2023-05-05:
This is related to #686: I think I mis-interpreted how the implicit "=" and "/=" generics are supposed to work.

commit 7819ad8e (2023-05-07): Rework handling of generic type predefined operators

Fixes #686
Fixes #658
closed 2023-05-07 by nickg