nickg/nvc #666: Build of r1.9.0 fails with error "../src/jit/jit-code.c:822:12: error: ‘SHT_X86_64_UNWIND’ undeclared ...".

ikwzm, 2023-04-08:
```console
shell$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.4 LTS
Release:	18.04
Codename:	bionic
```

```console
shell$ uname -a
Linux Jabberwock 4.4.0-19041-Microsoft #2311-Microsoft Tue Nov 08 17:09:00 PST 2022 x86_64 x86_64 x86_64 GNU/Linux
```

```console
shell$ dpkg -l | grep libc6-dev
ii  libc6-dev:amd64      2.27-3ubuntu1   amd64           GNU C Library: Development Libraries and Heade
ii  libc6-dev-arm64-cros 2.27-3ubuntu1cr all             GNU C Library: Development Libraries and Heade
ii  libc6-dev-armhf-cros 2.27-3ubuntu1cr all             GNU C Library: Development Libraries and Heade
```

```console
shell$ git clone --branch r1.9.0 --depth 1 https://github.com/nickg/nvc
Cloning into 'nvc'...
remote: Enumerating objects: 1911, done.        
remote: Counting objects: 100% (1911/1911), done.        
remote: Compressing objects: 100% (1672/1672), done.        
remote: Total 1911 (delta 252), reused 1404 (delta 193), pack-reused 0        
Receiving objects: 100% (1911/1911), 1.66 MiB | 11.90 MiB/s, done.
Resolving deltas: 100% (252/252), done.
Note: checking out 'dd6c14dc359ef9b3fcdd79a41a62d5a3ce55e80f'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

Checking out files: 100% (1866/1866), done.
```

```console
shell$ cd nvc
shell$ ./autogen.sh
configure.ac:38: installing './compile'
configure.ac:7: installing './config.guess'
configure.ac:7: installing './config.sub'
configure.ac:22: installing './install-sh'
configure.ac:22: installing './missing'
Makefile.am: installing './depcomp'
configure.ac: installing './ylwrap'
```

```console
shell$ ../configure --with-llvm=/usr/lib/llvm-8/bin/llvm-config
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking target system type... x86_64-pc-linux-gnu
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking whether to enable maintainer-specific portions of Makefiles... yes
checking for style of include used by make... GNU
[... 172 lines cut ...]
  CC       src/rt/mspace.o
  CC       src/rt/stdenv.o
  CC       src/rt/standard.o
  CC       src/rt/simpkg.o
  CC       src/rt/model.o
  CC       src/rt/fileio.o
  CC       src/jit/jit-core.o
  CC       src/jit/jit-irgen.o
  CC       src/jit/jit-dump.o
  CC       src/jit/jit-interp.o
  CC       src/jit/jit-exits.o
  CC       src/jit/jit-optim.o
  CC       src/jit/jit-ffi.o
  CC       src/jit/jit-code.o
../src/jit/jit-code.c: In function ‘code_load_elf’:
../src/jit/jit-code.c:822:12: error: ‘SHT_X86_64_UNWIND’ undeclared (first use in this function); did you mean ‘SHT_IA_64_UNWIND’?
       case SHT_X86_64_UNWIND:
            ^~~~~~~~~~~~~~~~~
            SHT_IA_64_UNWIND
../src/jit/jit-code.c:822:12: note: each undeclared identifier is reported only once for each function it appears in
Makefile:2660: recipe for target 'src/jit/jit-code.o' failed
make[1]: *** [src/jit/jit-code.o] Error 1
Makefile:1937: recipe for target 'all' failed
make: *** [all] Error 2
```

nickg, 2023-04-08:
It also fails in the same way on Alpine. Can you try again with the latest v1.9-branch?

BTW I think your simulations would benefit from the `--jit` option explained in #634. It makes elaboration much faster.

ikwzm, 2023-04-08:
Thank you for your quick response.
I tried v1.9-branch and it built successfully.

```console
shell$ git clone --branch v1.9-branch --depth 1 https://github.com/nickg/nvc
Cloning into 'nvc'...
remote: Enumerating objects: 1911, done.        
remote: Counting objects: 100% (1911/1911), done.        
remote: Compressing objects: 100% (1670/1670), done.        
remote: Total 1911 (delta 254), reused 1397 (delta 195), pack-reused 0        
Receiving objects: 100% (1911/1911), 1.66 MiB | 11.75 MiB/s, done.
Resolving deltas: 100% (254/254), done.
Checking out files: 100% (1867/1867), done.
```

```console
shell$ cd nvc
shell$ ./autogen.sh 
configure.ac:38: installing './compile'
configure.ac:7: installing './config.guess'
configure.ac:7: installing './config.sub'
configure.ac:22: installing './install-sh'
configure.ac:22: installing './missing'
Makefile.am: installing './depcomp'
configure.ac: installing './ylwrap'
```

```console
shell$ mkdir build
shell$ cd build
shell$ ../configure --with-llvm=/usr/lib/llvm-8/bin/llvm-config
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking target system type... x86_64-pc-linux-gnu
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking whether to enable maintainer-specific portions of Makefiles... yes
checking for style of include used by make... GNU
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking minix/config.h usability... no
checking minix/config.h presence... no
checking for minix/config.h... no
[... 252 lines cut ...]
  NVC      lib/ieee.19/IEEE.NUMERIC_STD
  NVC      lib/ieee.19/IEEE.NUMERIC_STD_UNSIGNED
  NVC      lib/ieee.19/IEEE.STD_LOGIC_1164-body
  NVC      lib/nvc.19/NVC.SIM_PKG
  NVC      lib/ieee.19/IEEE.NUMERIC_STD-body
  NVC      lib/ieee.19/IEEE.NUMERIC_STD_UNSIGNED-body
  NVC      lib/ieee.19/IEEE.NUMERIC_BIT-body
  NVC      lib/ieee.19/IEEE.NUMERIC_BIT_UNSIGNED-body
  NVC      lib/ieee.19/IEEE.STD_LOGIC_TEXTIO
  NVC      lib/ieee.19/IEEE.MATH_REAL
  NVC      lib/ieee.19/IEEE.MATH_REAL-body
  NVC      lib/ieee.19/IEEE.MATH_COMPLEX
  NVC      lib/ieee.19/IEEE.MATH_COMPLEX-body
  NVC      lib/ieee.19/IEEE.FIXED_FLOAT_TYPES
  NVC      lib/ieee.19/IEEE.FIXED_GENERIC_PKG
  NVC      lib/ieee.19/IEEE.FIXED_GENERIC_PKG-body
  NVC      lib/ieee.19/IEEE.FIXED_PKG
  NVC      lib/ieee.19/IEEE.FLOAT_GENERIC_PKG
  NVC      lib/ieee.19/IEEE.FLOAT_GENERIC_PKG-body
  NVC      lib/ieee.19/IEEE.FLOAT_PKG
  NVC      lib/ieee.19/IEEE.IEEE_STD_CONTEXT
  NVC      lib/ieee.19/IEEE.IEEE_BIT_CONTEXT
  NVC      lib/nvc.19/NVC.TEXTBUF-body
  NVC      lib/preload19.so
```

closed 2023-04-09 by nickg