nickg/nvc #667: Build failure with 1.9.0.

carlocab, 2023-04-08:
We're trying to update the version of nvc packaged in Homebrew at Homebrew/homebrew-core#127784. The build fails on all of our CI machines, unfortunately. We run macOS 11, 12, and 13 on Intel and M1 (which makes 6 machines), along with a seventh build on Linux.

The build failures look pretty similar across all seven machines. On [macOS](https://github.com/Homebrew/homebrew-core/actions/runs/4637499850/jobs/8212834087?pr=127784#step:7:409):
```
  ./bin/nvc --work=lib/std -L lib/ --std=1993 --aotgen -O2 -o lib/preload93.so std nvc ieee
  ** Fatal: execv: No such file or directory
  ** Fatal: clang -bundle -flat_namespace -undefined dynamic_lookup -o lib/preload93.so /private/tmp/nvc-rvald8 failed with status 1
  make[1]: *** [lib/preload93.so] Error 1
```
On [Linux](https://github.com/Homebrew/homebrew-core/actions/runs/4637499850/jobs/8212835002?pr=127784#step:8:409):
```
  ./bin/nvc --work=lib/std -L lib/ --std=1993 --aotgen -O2 -o lib/preload93.so std nvc ieee
  ** Fatal: execv: No such file or directory
  ** Fatal: ld --eh-frame-hdr -shared -o lib/preload93.so /tmp/nvc-j44J7E failed with status 1
  make[1]: *** [Makefile:4069: lib/preload93.so] Error 1
```

Complete build logs are available [here](https://github.com/Homebrew/homebrew-core/actions/runs/4637499850?pr=127784). I've linked to the start of the errors on macOS 13-arm64 and Linux above.

#666 also reports a build failure with the same version. However, the error is different, so I've opened a separate issue for this. One important difference might be that we are building with LLVM 16.

Any assistance would be appreciated.

nickg, 2023-04-08:
The argument to `--with-system-cc` should be an absolute path as it's passed to `execv()` which doesn't search `$PATH`, so `/usr/bin/clang` instead of `clang` should work. Similarly for the path to `ld` on Linux. I don't think that code has changed so I'm surprised the patching in that script worked before, but the reason it fails now is that this is called during the build process in 1.9 whereas it wasn't in 1.8.

Another fix might be to call `execvp()` instead of `execv()` which would then find `clang` in the user's `$PATH`. I think I avoided that before because it seems likely to break in subtle ways.

carlocab, 2023-04-08:
Our build-time compiler/linker is not found at the same path during run-time.

Is there a way to specify a specific path during the build but not hardcode it for use at runtime? (Because it won't be there at runtime.)

nickg, 2023-04-08:
OK I see. Would it work if there was an environment variable - say `NVC_CC` - that could override the value configured with `--with-system-cc`? So you'd run `configure --with-system-cc=/usr/bin/clang` but during the build also set `NVC_CC=/path/to/sandbox/clang`.

carlocab, 2023-04-08:
Yes, I suppose that might work. We'd also need the same for `ld`.

nickg, 2023-04-08:
After thinking about it a bit more I decided to just switch to using `execvp()` which searches in the PATH. This matches what we were doing on Windows anyway. Could you try cherry-picking c857e16c3 to see if that fixes your build failure?

carlocab, 2023-04-08:
Will do; thanks.

carlocab, 2023-04-08:
Seems to work! https://github.com/Homebrew/homebrew-core/actions/runs/4646130292/jobs/8222400773?pr=127784

carlocab, 2023-04-08:
Incidentally: I noticed you pass `-undefined dynamic_lookup` to the linker on macOS. Note that this throws a warning on newer versions of macOS `ld`:
```
ld: warning: -undefined dynamic_lookup may not work with chained fixups
```
You may wish to add `-no_fixup_chains` for sufficiently new versions of macOS `ld`. Sufficiently new here should be version `711` or newer.

You can query the version of `ld` with `ld -v` (plain text) or, with Xcode 10.2 or newer (ships `ld` version `450.3`), `ld -version_details` (JSON). `ld` throws errors when given an unknown flag.

See python/cpython#97524 for additional background here.

carlocab, 2023-04-08:
Since this is fixed now on `master`, feel free to close this when you're ready.

nickg, 2023-04-12:
> Incidentally: I noticed you pass `-undefined dynamic_lookup` to the linker on macOS. Note that this throws a warning on newer versions of macOS `ld`:
> 

Thanks for sharing. I'd seen this before but didn't know what caused it. I added  `-no_fixup_chains` with a check to see if the linker accepts it in 6bd234d above. I'll include that and the original fix in a 1.9.1 release some time soon.