nickg/nvc #668: "*** Caught signal 11 (SEGV_MAPERR) [address=(nil), ip=0x7fce50c78852] ***" occurs when elabolating VHDL with nvc 1.9.0.

ikwzm, 2023-04-08:
I'm currently trying to simulate a VHDL project with nvc 1.9.0.r2
nvc 1.9.0.r2 is the version that fixes issue #666.

```console
nvc --version
nvc 1.9.0 (1.9.0.r2.ge550f26a) (Using LLVM 8.0.0)
Copyright (C) 2011-2023  Nick Gasson
This program comes with ABSOLUTELY NO WARRANTY. This is free software, and
you are welcome to redistribute it under certain conditions. See the GNU
General Public Licence for details.
```

The project is included at https://github.com/ikwzm/Merge_Sorter
You can download it as follows:

```console
shell$ git clone --branch nvc-issue-668 --depth 1 --recursive https://github.com/ikwzm/Merge_Sorter.git
Cloning into 'Merge_Sorter'...
remote: Enumerating objects: 467, done.        
remote: Counting objects: 100% (467/467), done.        
remote: Compressing objects: 100% (315/315), done.        
remote: Total 467 (delta 277), reused 271 (delta 143), pack-reused 0        
Receiving objects: 100% (467/467), 4.69 MiB | 13.34 MiB/s, done.
Resolving deltas: 100% (277/277), done.
Checking out files: 100% (445/445), done.
Submodule 'Dummy_Plug' (https://github.com/ikwzm/Dummy_Plug.git) registered for path 'Dummy_Plug'
Submodule 'PipeWork' (https://github.com/ikwzm/PipeWork.git) registered for path 'PipeWork'
Cloning into '/mnt/d/ichiro/tmp/Merge_Sorter/Dummy_Plug'...
remote: Enumerating objects: 5505, done.        
remote: Counting objects: 100% (145/145), done.        
remote: Compressing objects: 100% (88/88), done.        
remote: Total 5505 (delta 67), reused 114 (delta 44), pack-reused 5360        
Receiving objects: 100% (5505/5505), 6.82 MiB | 12.06 MiB/s, done.
Resolving deltas: 100% (4407/4407), done.
Cloning into '/mnt/d/ichiro/tmp/Merge_Sorter/PipeWork'...
remote: Enumerating objects: 2735, done.        
remote: Counting objects: 100% (171/171), done.        
remote: Compressing objects: 100% (117/117), done.        
remote: Total 2735 (delta 104), reused 106 (delta 50), pack-reused 2564        
Receiving objects: 100% (2735/2735), 2.18 MiB | 12.27 MiB/s, done.
Resolving deltas: 100% (1998/1998), done.
Submodule path 'Dummy_Plug': checked out '933f42016fb8c21f92f2f96a5c1ab0a2e591a1fd'
Submodule 'docs' (https://github.com/ikwzm/Dummy_Plug.git) registered for path 'Dummy_Plug/docs'
Cloning into '/mnt/d/ichiro/tmp/Merge_Sorter/Dummy_Plug/docs'...
remote: Enumerating objects: 5505, done.        
remote: Counting objects: 100% (145/145), done.        
remote: Compressing objects: 100% (88/88), done.        
remote: Total 5505 (delta 67), reused 114 (delta 44), pack-reused 5360        
Receiving objects: 100% (5505/5505), 6.82 MiB | 6.08 MiB/s, done.
Resolving deltas: 100% (4407/4407), done.
Submodule path 'Dummy_Plug/docs': checked out 'a801570f3d1d46f52edd64d02ba9960fef2d5728'
Submodule path 'PipeWork': checked out 'fe8a74d327be76f020c67254f74369e444892319'
```

The project is sim/nvc/argsort_axi
You can recreate the error by running:

```console
shell$ cd Merge_Sorter/sim/nvc/argsort_axi
shell$ make analyze_libs
sh analyze_libs.sh
** Warning: `default' is a reserved word in VHDL-2008
      > /mnt/d/ichiro/tmp/Merge_Sorter/Dummy_Plug/src/main/vhdl/axi4/axi4_channel_player.vhd:2958
      |
 2958 |                           default     : in  std_logic;
      |                           ^^^^^^^
      |
      = Note: pass --std=2008 to enable this feature
** Warning: shared variable MEM must have protected type
     > /mnt/d/ichiro/tmp/Merge_Sorter/Dummy_Plug/src/main/vhdl/axi4/axi4_memory_player.vhd:231
     |
 231 |     shared variable  mem        :  MEMORY_TYPE;
     |                      ^^^
cat analyze_libs.sh >analyze_libs 
chmod a+x analyze_libs
```

```console
shell$ make
nvc -L ./ -M 32M --work=WORK -e  ArgSort_AXI_Test_Bench_x04_w1_f0

*** Caught signal 11 (SEGV_MAPERR) [address=(nil), ip=0x7f321b278852] ***

[0x7f321b2127ac] ../src/util.c:837 signal_handler
[0x7f321aa3ef0f] (/lib/x86_64-linux-gnu/libc-2.27.so) 
[0x7f321b278852] ../src/jit/jit-interp.c:517 interp_loop
[0x7f321b278852] ../src/jit/jit-interp.c:1108 jit_interp
[0x7f321b276f7d] ../src/jit/jit-interp.c:666 interp_loop
[0x7f321b276f7d] ../src/jit/jit-interp.c:1108 jit_interp
[0x7f321b2c42b0] ../src/jit/jit-core.c:656 jit_try_vcall.lto_priv.180
[0x7f321b2c4b25] ../src/jit/jit-core.c:714 jit_try_call
[0x7f321b23a98b] ../src/eval.c:326 eval_instance
[0x7f321b23a98b] ../src/eval.c:1161 elab_lower.lto_priv.420
[0x7f321b23bddf] ../src/elab.c:1558 elab_stmts.lto_priv.419
[0x7f321b23ca02] ../src/elab.c:1260 elab_stmts.lto_priv.419
[0x7f321b23ca02] ../src/elab.c:1260 elab_stmts.lto_priv.419
[0x7f321b23ca02] ../src/elab.c:1260 elab_stmts.lto_priv.419
[0x7f321b23ca02] ../src/elab.c:1260 elab_stmts.lto_priv.419
[0x7f321b265a77] ../src/elab.c:1943 elab_top_level.lto_priv.616
[0x7f321b21b862] ../src/nvc.c:1989 elaborate
[0x7f321b21b862] ../src/elab.c:1425 process_command
[0x7f321b20ea48] ../src/nvc.c:1547 main
Makefile:55: recipe for target 'WORK/_WORK.ARGSORT_AXI_TEST_BENCH_X04_W1_F0.elab.so' failed
make: *** [WORK/_WORK.ARGSORT_AXI_TEST_BENCH_X04_W1_F0.elab.so] Error 2
```

nickg, 2023-04-10:
The crash is fixed by the commit above but there is another error that occurs during elaboration which I am still investigating.

nickg, 2023-04-15:
Both issues are fixed now.

closed 2023-04-15 by nickg