nickg/nvc #671: MSYS build issues

avelure, 2023-04-12:
I did the mistake of updating my MSYS, the latest update brings in LLVM16 as the default and it is not so easy to roll back.
Is it in the works to support LLVM16?

I can install LLVM15 as a separate package, but they don't include LLVM15 as a shared library in that package.
Building NVC with `../configure --with-llvm=/ucrt64/opt/llvm-15/bin/llvm-config.exe --enable-static-llvm`  gives a few ignored errors during compile
```
  CXXLD    bin/nvc.exe
make[2]: [C:\msys64\tmp\cclZpxIF.mk:6: C:\msys64\tmp\ccbd2Juv.ltrans1.ltrans.o] Error 1 (ignored)
make[2]: [C:\msys64\tmp\cclZpxIF.mk:78: C:\msys64\tmp\ccbd2Juv.ltrans25.ltrans.o] Error 1 (ignored)
make[2]: [C:\msys64\tmp\cclZpxIF.mk:69: C:\msys64\tmp\ccbd2Juv.ltrans22.ltrans.o] Error 1 (ignored)
make[2]: [C:\msys64\tmp\cclZpxIF.mk:3: C:\msys64\tmp\ccbd2Juv.ltrans0.ltrans.o] Error 1 (ignored)
make[2]: [C:\msys64\tmp\cclZpxIF.mk:36: C:\msys64\tmp\ccbd2Juv.ltrans11.ltrans.o] Error 1 (ignored)
make[2]: [C:\msys64\tmp\cclZpxIF.mk:33: C:\msys64\tmp\ccbd2Juv.ltrans10.ltrans.o] Error 1 (ignored)
make[2]: [C:\msys64\tmp\cclZpxIF.mk:81: C:\msys64\tmp\ccbd2Juv.ltrans26.ltrans.o] Error 1 (ignored)
make[2]: [C:\msys64\tmp\cclZpxIF.mk:66: C:\msys64\tmp\ccbd2Juv.ltrans21.ltrans.o] Error 1 (ignored)
make[2]: [C:\msys64\tmp\cclZpxIF.mk:57: C:\msys64\tmp\ccbd2Juv.ltrans18.ltrans.o] Error 1 (ignored)
make[2]: [C:\msys64\tmp\cclZpxIF.mk:60: C:\msys64\tmp\ccbd2Juv.ltrans19.ltrans.o] Error 1 (ignored)
make[2]: [C:\msys64\tmp\cclZpxIF.mk:24: C:\msys64\tmp\ccbd2Juv.ltrans7.ltrans.o] Error 1 (ignored)
make[2]: [C:\msys64\tmp\cclZpxIF.mk:39: C:\msys64\tmp\ccbd2Juv.ltrans12.ltrans.o] Error 1 (ignored)
make[2]: [C:\msys64\tmp\cclZpxIF.mk:63: C:\msys64\tmp\ccbd2Juv.ltrans20.ltrans.o] Error 1 (ignored)
make[2]: [C:\msys64\tmp\cclZpxIF.mk:42: C:\msys64\tmp\ccbd2Juv.ltrans13.ltrans.o] Error 1 (ignored)
make[2]: [C:\msys64\tmp\cclZpxIF.mk:15: C:\msys64\tmp\ccbd2Juv.ltrans4.ltrans.o] Error 1 (ignored)
make[2]: [C:\msys64\tmp\cclZpxIF.mk:21: C:\msys64\tmp\ccbd2Juv.ltrans6.ltrans.o] Error 1 (ignored)
make[2]: [C:\msys64\tmp\cclZpxIF.mk:51: C:\msys64\tmp\ccbd2Juv.ltrans16.ltrans.o] Error 1 (ignored)
make[2]: [C:\msys64\tmp\cclZpxIF.mk:27: C:\msys64\tmp\ccbd2Juv.ltrans8.ltrans.o] Error 1 (ignored)
make[2]: [C:\msys64\tmp\cclZpxIF.mk:12: C:\msys64\tmp\ccbd2Juv.ltrans3.ltrans.o] Error 1 (ignored)
make[2]: [C:\msys64\tmp\cclZpxIF.mk:54: C:\msys64\tmp\ccbd2Juv.ltrans17.ltrans.o] Error 1 (ignored)
make[2]: [C:\msys64\tmp\cclZpxIF.mk:87: C:\msys64\tmp\ccbd2Juv.ltrans28.ltrans.o] Error 1 (ignored)
make[2]: [C:\msys64\tmp\cclZpxIF.mk:30: C:\msys64\tmp\ccbd2Juv.ltrans9.ltrans.o] Error 1 (ignored)
make[2]: [C:\msys64\tmp\cclZpxIF.mk:75: C:\msys64\tmp\ccbd2Juv.ltrans24.ltrans.o] Error 1 (ignored)
make[2]: [C:\msys64\tmp\cclZpxIF.mk:45: C:\msys64\tmp\ccbd2Juv.ltrans14.ltrans.o] Error 1 (ignored)
make[2]: [C:\msys64\tmp\cclZpxIF.mk:48: C:\msys64\tmp\ccbd2Juv.ltrans15.ltrans.o] Error 1 (ignored)
make[2]: [C:\msys64\tmp\cclZpxIF.mk:84: C:\msys64\tmp\ccbd2Juv.ltrans27.ltrans.o] Error 1 (ignored)
make[2]: [C:\msys64\tmp\cclZpxIF.mk:9: C:\msys64\tmp\ccbd2Juv.ltrans2.ltrans.o] Error 1 (ignored)
make[2]: [C:\msys64\tmp\cclZpxIF.mk:72: C:\msys64\tmp\ccbd2Juv.ltrans23.ltrans.o] Error 1 (ignored)
make[2]: [C:\msys64\tmp\cclZpxIF.mk:18: C:\msys64\tmp\ccbd2Juv.ltrans5.ltrans.o] Error 1 (ignored)
 /usr/bin/mkdir -p '/ucrt64/bin'
 ```
 Running `make check-TESTS` from within MSYS, then all tests except the VHPI ones pass (vhpi[1,2,3,5] and issue612)
 ```
 C:\proj\public\nvc\build\bin\nvc.exe --std=1993 -a C:\proj\public\nvc\test\regress\vhpi3.vhd -e vhpi3 -O0 -r --load=
C:\proj\public\nvc\build\bin/../lib/vhpi3.so.exe vhpi3
** Note: loading VHPI plugin C:\proj\public\nvc\build\bin/../lib/vhpi3.so.exe
** Fatal: failed to load C:\proj\public\nvc\build\bin/../lib/vhpi3.so.exe
```
Running my own tests from native Windows I get
 ```
 fatal: failed to resolve symbol memcpy
[00007FF60F69302F]
[00007FF60F735A95]
[00007FF60F6921D2]
[00007FF60F770164] vhpi_put_data+0x27c4
[00007FF60F73A932]
[00007FF60F73D011]
[00007FF60F73D068]
[00007FF60F732720]
[00007FF8217B4D0B] pthread_create_wrapper+0x9b
[00007FF855E11BB2] configthreadlocale+0x92
[00007FF8568D7614] BaseThreadInitThunk+0x14
[00007FF8580226A1] RtlUserThreadStart+0x21
```
I'm not able to easily test my own tests from within MSYS without changing a lot of build scripts as it seems NVC does not like linux style paths in the library parameter when running under MSYS, which is what my python based build scripts choose to use under MSYS
```
$ nvc -L/c/proj/library --work=libname:/c/proj/library/libname --std=08 -a /c/proj/libraries/libname/lib.vhd
fatal: invalid character ';' in library name
```
I don't know if these are all related issues.

nickg, 2023-04-12:
On 2023-04-12 10:21, Arild Velure wrote:
> I did the mistake of updating my MSYS, the latest update brings in
> LLVM16 as the default and it is not so easy to roll back.
> Is it in the works to support LLVM16?

The master branch as of a few weeks ago as well as the 1.9 release 
should support LLVM 16.

> 
>   CXXLD    bin/nvc.exe
> make[2]: [C:\msys64\tmp\cclZpxIF.mk:6:
> C:\msys64\tmp\ccbd2Juv.ltrans1.ltrans.o] Error 1 (ignored)
> make[2]: [C:\msys64\tmp\cclZpxIF.mk:78:
> C:\msys64\tmp\ccbd2Juv.ltrans25.ltrans.o] Error 1 (ignored)
> make[2]: [C:\msys64\tmp\cclZpxIF.mk:69:

This was also reported in #648. I don't know what causes it yet but it 
seems to be harmless as the program still works correctly. You can avoid 
it by passing --disable-lto to configure but there will be a small 
performance loss.

> Running make check-TESTS from within MSYS, then all tests except the
> VHPI ones pass (vhpi[1,2,3,5] and issue612)
>
> C:\proj\public\nvc\build\bin\nvc.exe --std=1993 -a [...]

Try `make check` instead as that will also build the VHPI libraries 
those tests require.

> Running my own tests from native Windows I get
>
> fatal: failed to resolve symbol memcpy
> [00007FF60F69302F]
> [00007FF60F735A95]
> [00007FF60F6921D2]
> [00007FF60F770164] vhpi_put_data+0x27c4 [...]

This is definitely a bug. I think I know what causes it and I'll try to 
fix it this evening.

> I'm not able to easily test my own tests from within MSYS without
> changing a lot of build scripts as it seems NVC does not like linux
> style paths in the library parameter when running under MSYS, [...]

When you run commands from MSYS2 bash, any ':' characters in program 
arguments get converted to ';'. I guess Python is running that command 
through bash which triggers this behaviour. There's some explanation 
here:

https://www.msys2.org/docs/filesystem-paths/

You might be able to fix it by setting `MSYS2_ARG_CONV_EXCL=--work=`. I 
know @amb5l ran into this before so he may have a better solution.

avelure, 2023-04-12:
I'm sure I pulled master yesterday, but I see it works now.
make check works as well and passes all tests.

I looked at the C:\msys64\tmp\cclZpxIF.mk file and it is failing on the last line shown below
```
C:\msys64\tmp\ccBRQMhS.ltrans1.ltrans.o:
	@C:\msys64\ucrt64\bin\gcc.exe  '-xlto' '-c' '-fno-openmp' '-fno-openacc' '-fpic' '-fcf-protection=none' '-mtune=generic' '-march=nocona' '-O2' '-pthread' '-g' '-O2' '-L' 'C:/msys64/ucrt64/lib' '-L' 'C:/msys64/ucrt64/lib/../lib' '-mtune=generic' '-march=nocona' '-dumpdir' 'bin/nvc.' '-dumpbase' 'bin/nvc.ltrans1.ltrans' '-fltrans' '-o' 'C:\msys64\tmp\ccBRQMhS.ltrans1.ltrans.o' 'C:\msys64\tmp\ccBRQMhS.ltrans1.o'
	@-touch -r C:\msys64\tmp\ccBRQMhS.ltrans1.o C:\msys64\tmp\ccBRQMhS.ltrans1.o.tem > /dev/null 2>&1 && mv C:\msys64\tmp\ccBRQMhS.ltrans1.o.tem C:\msys64\tmp\ccBRQMhS.ltrans1.o

```
Which seems like it's just a way of emptying the file and keeping the timestamp, so I guess it is harmless.

The same happens when running in bash outside the python, so I guess so.
Setting MSYS2_ARG_CONV_EXCL seems to work, but then I get another error
```
fatal: mkdir: /c/proj/library/libname: The system cannot find the path specified.
```
But running `mkdir /c/proj/library/libname` works fine.

Latest master gives same `failed to resolve symbol memcpy` in native widndows

nickg, 2023-04-12:
> 
> The same happens when running in bash outside the python, so I guess so. Setting MSYS2_ARG_CONV_EXCL seems to work, but then I get another error
> 
> ```
> fatal: mkdir: /c/proj/library/libname: The system cannot find the path specified.
> ```

OK so the problem is that `MSYS2_ARG_CONV_EXCL` disables both the ':' to ';' conversion (which we want), and the Unix to Windows path conversion (which we don't want). I think a solution might be to modify the `--work` (and `--map`) arguments so they accept ';' as well as ':' to separate the library name and directory. Then it would work regardless of whether MSYS2 does translation or not.

avelure, 2023-04-12:
I guess that might be a quick hack to fix it yes.

nickg, 2023-04-12:
I've implemented this now so `--work=foo;c:\foo` is allowed and as is equivalent to `--work=foo:/c/foo` before the path translation. I also fixed the "failed to resolve symbol memcpy" error. Could you test again?

avelure, 2023-04-13:
Great, it now it compiles in MSYS, but I get
```
fatal: failed to resolve symbol ___chkstk_ms
[00007FF705E4302F]
[00007FF705EE5BE5]
[00007FF705E421D2]
[00007FF705F2010A] vhpi_put_data+0x271a
[00007FF705EEAA82]
[00007FF705EEAC31]
[00007FF705EED1E8]
[00007FF705EE2870]
[00007FF82D044D0B] pthread_create_wrapper+0x9b
[00007FF855E11BB2] configthreadlocale+0x92
[00007FF8568D7614] BaseThreadInitThunk+0x14
[00007FF8580226A1] RtlUserThreadStart+0x21
```
In native windows it works ok, also in simulation.

nickg, 2023-04-13:
Which MSYS environment is that running in? I can't reproduce that in MinGW x64 or UCRT x64.

nickg, 2023-04-13:
> Which MSYS environment is that running in? I can't reproduce that in MinGW x64 or UCRT x64.

Never mind, needs a function with a stack frame larger than one page. E.g.

```vhdl
entity test is
end entity;

architecture tb of test is

    procedure other(x : bit_vector) is
    begin
    end procedure;

    procedure proc is
        variable b : bit_vector(1 to 10000);
    begin
        other(b);
    end procedure;

begin
   p_main: process
   begin
       proc;
       wait for 1 ns;
  end process;
end architecture;
```

nickg, 2023-04-13:
Should be fixed by the commit above.

closed 2023-04-13 by nickg

avelure, 2023-04-17:
It all works from MSYS now, thanks.