nickg/nvc #677: Analysing file causes segmentation fault (unaffected keyword)

yikkrubz, 2023-04-24:
Analysing the file unaffected_ff.vhd causes a segmentation fault.

nvc --std=08 -a unaffected_ff.vhd 

*** Caught signal 11 (SIGSEGV) [address=0x8, ip=0x108810e42] ***

[0x1087d4535] signal_handler+0x405
[0x7ff80f5b0dfc] (/usr/lib/system/libsystem_platform.dylib) _sigtramp+0x1c
[0x108810e43] set_delay_mechanism+0x73
[0x10880fab4] p_signal_assignment_statement+0x624
[0x10880d8f9] p_sequential_statement+0x35f9
[0x10880c269] p_sequential_statement+0x1f69
[0x10880be09] p_sequential_statement+0x1b09
[0x108814149] p_process_statement+0xa59
[0x1087de4db] p_concurrent_statement+0x66b
[0x1087dc3a8] p_secondary_unit+0xae8
[0x1087d8010] parse+0x3b0
[0x108868fec] analyse_vhdl+0x3c
[0x1087cc751] process_command+0x1df1
[0x1087ca1fa] main+0x4aa

nvc --version
nvc 1.10-devel (1.9.0.r53.g15f4cb8a) (Using LLVM 16.0.2)

[unaffected_ff.txt](https://github.com/nickg/nvc/files/11314342/unaffected_ff.txt)

nickg, 2023-04-24:
Try `nvc -r --exit-severity=failure`.

nickg, 2023-04-24:
Sorry ignore that remark. Posted on the wrong issue.

nickg, 2023-04-24:
Should be fixed now.

closed 2023-04-24 by nickg