nickg/nvc #680: Segmentation faults running vunit-based testbenches

yikkrubz, 2023-04-25:
When I try to run vunit-based test benches nvc is stopped with segmentation faults. This happens with the most recent vunit release and the current git version. It does not happen with the repository nickg/vunit.

For example, when I try to run the example array included in the vunit distribution the simulation causes a segmentation fault. For details see the transcript attached below. I appologize if this is a local issue.  

Starting lib.tb_sobel_x.test_random_data_against_model
Output file: /Users/Peter/Downloads/vunit/examples/vhdl/array/vunit_out/test_output/lib.tb_sobel_x.test_random_data_against_model_9dbb88996d73eddd4839bdd5d1e2a36eaba47e42/output.txt
/usr/local/bin/nvc --work=lib:/Users/Peter/Downloads/vunit/examples/vhdl/array/vunit_out/nvc/libraries/lib --std=2008 --map=vunit_lib:/Users/Peter/Downloads/vunit/examples/vhdl/array/vunit_out/nvc/libraries/vunit_lib --map=osvvm:/Users/Peter/Downloads/vunit/examples/vhdl/array/vunit_out/nvc/libraries/osvvm --map=lib:/Users/Peter/Downloads/vunit/examples/vhdl/array/vunit_out/nvc/libraries/lib -H 64m -e tb_sobel_x-tb -gtb_path=/Users/Peter/Downloads/vunit/examples/vhdl/array/src/test/ -grunner_cfg=active python runner : true,enabled_test_cases : test_random_data_against_model,output path : /Users/Peter/Downloads/vunit/examples/vhdl/array/vunit_out/test_output/lib.tb_sobel_x.test_random_data_against_model_9dbb88996d73eddd4839bdd5d1e2a36eaba47e42/,tb path : /Users/Peter/Downloads/vunit/examples/vhdl/array/src/test/,use_color : true --no-save --jit -r --exit-severity=error

*** Caught signal 11 (SIGSEGV) [address=0x10d20a086, ip=0x10d20a086] ***

[0x108b934f5] signal_handler+0x405
[0x7ff802480dfc] (/usr/lib/system/libsystem_platform.dylib) _sigtramp+0x1c
fail (P=0 S=0 F=1 T=2) lib.tb_sobel_x.test_random_data_against_model (0.2 seconds)

Operating system macos 12.6.5 
nvc --version
nvc 1.10-devel (1.9.0.r54.ge02e7e19) (Using LLVM 16.0.2)

[transcript.txt](https://github.com/nickg/nvc/files/11326375/transcript.txt)

nickg, 2023-04-25:
Is it an Intel or Arm Mac?

nickg, 2023-04-25:
If it's an Intel CPU the `--jit` option that VUnit uses probably doesn't work correctly as I only had an M1 to test on. I'll fix it soon but you could try temporarily commenting out the line that adds `--jit` in `vunit/sim_if/nvc.py`.

yikkrubz, 2023-04-25:
> 
> 
> If it's an Intel CPU the --jit option that VUnit uses probably doesn't work correctly as I only had an M1 to test on. I'll fix it soon but you could try temporarily commenting out the line that adds --jit in vunit/sim_if/nvc.py.

It is indeed an Intel-based mac and commenting the lines 266 and 267 in the file nvc.py does fix the issue. Thank you very much for the fix.

Starting lib.tb_sobel_x.test_random_data_against_model
Output file: /Users/Peter/Downloads/tmp/demo/vunit/examples/vhdl/array/vunit_out/test_output/lib.tb_sobel_x.test_random_data_against_model_9dbb88996d73eddd4839bdd5d1e2a36eaba47e42/output.txt
pass (P=1 S=0 F=0 T=2) lib.tb_sobel_x.test_random_data_against_model (1.1 seconds)

Starting lib.tb_sobel_x.test_input_file_against_output_file
Output file: /Users/Peter/Downloads/tmp/demo/vunit/examples/vhdl/array/vunit_out/test_output/lib.tb_sobel_x.test_input_file_against_output_file_b1de5eca46714be73ca0390ba61a470c08d9bdc0/output.txt
pass (P=2 S=0 F=0 T=2) lib.tb_sobel_x.test_input_file_against_output_file (0.9 seconds)

==== Summary ==============================================================
pass lib.tb_sobel_x.test_random_data_against_model      (1.1 seconds)
pass lib.tb_sobel_x.test_input_file_against_output_file (0.9 seconds)
===========================================================================
pass 2 of 2
===========================================================================
Total time was 2.1 seconds
Elapsed time was 2.1 seconds

nickg, 2023-04-28:
This should be fixed now. Could you test again with the workaround removed?

yikkrubz, 2023-04-28:
I ran some vunit-based testbenches They all work with the workaround removed. Thank you very much for fixing the issue.

closed 2023-04-28 by nickg