nickg/nvc #682: Testbench does not stop

yikkrubz, 2023-04-28:
When the adder is simulated by nvc,  the simulation does not stop. When questa or ghdl are employed for simulating, the simulation stops.

nvc 1.10-devel (1.9.0.r54.ge02e7e19) (Using LLVM 16.0.2)
operating system macos 12.6.5 (intel)

nvc -a adder.vhd
nvc -a adder_tb.vhd
nvc -e adder_tb
nvc -r adder_tb
^C** Fatal: 362802510ns+0: interrupted in process :adder_tb:_p0
    > adder_tb.vhd:23
    |
 23 |   clk <= not clk after 10 ns;
    |          ^

adder.zip](https://github.com/nickg/nvc/files/11356356/adder.zip)

nickg, 2023-04-28:
How long did you run it for with Questa/GHDL? The clock generation process always generates new transactions every 10 ns so the simulation will only stop when the time reaches `TIME'HIGH`. In NVC `TIME` is a signed 64-bit value so that will be `(2 ** 63) FS`.

yikkrubz, 2023-04-28:
Disregard my message. It was my mistake. I appologize for the noise.

closed 2023-04-28 by yikkrubz