nickg/nvc #686: Package references only resolved when explicitly `use`d

m42uko, 2023-05-02:
When referencing a function from a package, it is only visible when it is explicitly used, just using the package and calling the function is not enough, as in:

```
use work.my_pkg;
[...]
my_pkg.myfunc()
```

Adding a `use work.my_pkg.myfunc()` makes the code compile.

**Expected behavior**: It calls the function.
**Actual behavior**: NVC complains that there is no visible subprogram with that signature.
**Version:** nvc 1.10-devel (1.9.0.r65.g9299dc6d) (Using LLVM 15.0.7)

See attachement [dot_pkg.tar.gz](https://github.com/nickg/nvc/files/11373352/dot_pkg.tar.gz) for full example.

nickg, 2023-05-02:
> 
> Adding a `use work.my_pkg.myfunc()` makes the code compile.
> 

You mean `use work.my_pkg.all` right? I think this is only a problem with the implicit `"="` and `"/="` generics that follow a generic type. Sections 14.3.3.1 of the 2008 LRM says:

> The generic association list also contains implicit association elements for the predefined equality (=) operator and inequality (/=) operators of each generic type; the actual part of such an implicit association element is the name of the predefined equality operator or inequality operator for the base type of the subtype indication in the actual part of the association element corresponding to the generic type.

But this is implemented incorrectly at the moment as it just looks up `"="` and `"/="` using the normal visibility rules.

m42uko, 2023-05-02:
> You mean `use work.my_pkg.all` right? I think this is only a problem with the implicit "=" and "/=" generics that follow a generic type. [...]

Oops, yes, I meant `.all`.  I thought I had seen a similar issue in another part of code as well, but I'll check tomorrow and provide an example if I find something.

Thanks for the super quick reply (and fix on my other ticket) though!

nickg, 2023-05-02:
> I thought I had seen a similar issue in another part of code as well, but I'll check tomorrow and provide an example if I find something.

Possibly something like `use work.my_pkg.my_type` which in VHDL-2008 should also import all the pre-defined operators for `my_type` but doesn't currently in NVC (I'll try to fix it along with this).

m42uko, 2023-05-04:
I think the other behavior I saw is something else. I created #688 for that.

closed 2023-05-07 by nickg

nickg, 2024-07-31:
I ended up reverting the changes here after the discussion in ghdl/ghdl#2696 so now it requires the operators to be visible, as the 2019 LRM would. But I did improve the error message which should help any other users who hit this problem:

```
** Error: no visible subprogram "=" matches signature [EXPECTEDTYPE, EXPECTEDTYPE return BOOLEAN]
    > ScoreboardPkg_tlg.vhd:13
    |
 13 | package ScoreboardPkg_tlg is new osvvm.ScoreboardGenericPkg
    | ^
    |
    = Note: while resolving interface subprogram default for "="
        > osvvm/ScoreboardGenericPkg.vhd:95
    = Note: add a use clause that makes the operators for WORK.TLG_PKG.TLG_T visible
    = Note: the 2008 LRM is unclear on the visibility requirements for the implicit operators 
            associated with a generic type, and behaviour varies between implementations; the 
            interpretation here is consistent with the 2019 LRM
    = Help: IEEE Std 1076-2008 section 6.5.3 "Interface type declarations"

```

commit 7819ad8e (2023-05-07): Rework handling of generic type predefined operators

Fixes #686
Fixes #658